Ysurac/openmptcprouter #1128: Bug détecté pour l'IPv6 + suppression port forwarding

AdrienDssy, 2020-08-11:
## Current Behavior
Bon pas le temps de le rédiger en Anglais mais depuis que j'ai fait la mise à jour du routeur j'ai constaté quelques "bugs", je m'explique, quand j'active l'IPv6 est activé tout est ok sauf que je l'ai pas et en plus ça retire mon port forwarding mais quand y'a juste l'IPv4 le port forwarding reste. J'ai constaté que l'IPv6 ne fonctionne plus sur aucun des VPN que j'ai testés

## Specifications

  - OpenMPTCProuter version:  (0.55) 
  - OpenMPTCProuter VPS version:  (0.1017)
  - OpenMPTCProuter platform: (86_64)

Ysurac, 2020-08-11:
Il y a des erreurs dans la page status ?

AdrienDssy, 2020-08-11:
Sur une page en particulier ?

Ysurac, 2020-08-11:
Oui, donc la page status. Dans Système->OpenMPTCProuter

AdrienDssy, 2020-08-11:
Sur la page Status (Overview) j'ai rien de particulier

Ysurac, 2020-08-11:
Donc l'IPv6 fonctionne sur le routeur.
Pour le port forwarding, dans /etc/shorewall/rules sur le VPS, les ports forwardés sont bien présents ?

AdrienDssy, 2020-08-11:
Oui mais ce qui est bizarre c'est que j'ai l'impression que ça ce retire dès que je met la v6

Ysurac, 2020-08-11:
Active l'IPv6 pour vérifier justement.

AdrienDssy, 2020-08-11:
Côté VPS ça reste mais pas côté routeur

AdrienDssy, 2020-08-11:
et en plus quand la v6 est active la connexion est refusée (sur la v4)

Ysurac, 2020-08-11:
Comment ça ? Les entrées sont toujours présentes dans la partie firewall via l'interface web ?

Ysurac, 2020-08-11:
La connexion est refusée où ? Un peu plus de précision si possible.

AdrienDssy, 2020-08-11:
Y'a les entrée qui ce retirent quand j'active la v6 et la connexion est refusée en même temps, côté VPS il y est encore

Ysurac, 2020-08-11:
Etrange, je n'ai rien pour supprimer des régles.
La connexion est refusée dans quel sens ? plus possible d'aller sur internet, ou pas possible d'accèder au routeur via le VPS (logique si les régles sont effacées).

Il y a quoi dans les logs dans Etat->Journal système ?

AdrienDssy, 2020-08-11:
Plus possible à ma machine d'être accédée de l’extérieure

AdrienDssy, 2020-08-11:
`Tue Aug 11 10:04:50 2020 user.info : using tcp fast open
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2166]: enable TCP no-delay
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2166]: resolving hostname to IPv6 address first
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2166]: initializing ciphers... chacha20-ietf-poly1305
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2166]: listening at 0.0.0.0:1100
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2166]: tcp port reuse enabled
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2166]: UDP relay enabled
Tue Aug 11 10:04:50 2020 daemon.err /usr/bin/ss-redir[2166]: [udp] getaddrinfo: System error
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2166]: running from root user
Tue Aug 11 10:04:50 2020 user.info : using tcp fast open
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2168]: enable TCP no-delay
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2168]: resolving hostname to IPv6 address first
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2168]: initializing ciphers... chacha20-ietf-poly1305
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2168]: listening at 0.0.0.0:1100
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2168]: tcp port reuse enabled
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2168]: UDP relay enabled
Tue Aug 11 10:04:50 2020 daemon.err /usr/bin/ss-redir[2168]: [udp] getaddrinfo: System error
Tue Aug 11 10:04:50 2020 daemon.info /usr/bin/ss-redir[2168]: running from root user
Tue Aug 11 10:04:50 2020 user.notice omr-tracker: Launching...
Tue Aug 11 10:04:50 2020 user.notice omr-tracker: Launched
Tue Aug 11 10:04:52 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 1.1.1.1)
Tue Aug 11 10:04:52 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.252.5 dev tun0
Tue Aug 11 10:05:08 2020 user.notice Shadowsocks: Rules DOWN
Tue Aug 11 10:05:08 2020 user.notice Shadowsocks: Rules UP
Tue Aug 11 10:05:08 2020 user.notice Shadowsocks: Reload omr-bypass rules
Tue Aug 11 10:05:08 2020 user.notice omr-bypass: Starting OMR-ByPass...
Tue Aug 11 10:05:09 2020 user.notice omr-bypass: Reload dnsmasq...
Tue Aug 11 10:05:09 2020 daemon.info dnsmasq[21990]: read /etc/hosts - 4 addresses
Tue Aug 11 10:05:09 2020 daemon.info dnsmasq[21990]: read /tmp/hosts/odhcpd - 0 addresses
Tue Aug 11 10:05:09 2020 daemon.info dnsmasq[21990]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Tue Aug 11 10:05:09 2020 daemon.info dnsmasq-dhcp[21990]: read /etc/ethers - 0 addresses
Tue Aug 11 10:05:09 2020 user.notice omr-bypass: OMR-ByPass is running
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-local[5692]: using tcp fast open
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-local[5692]: enable TCP no-delay
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-local[5692]: initializing ciphers... chacha20-ietf-poly1305
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-local[5692]: listening at 127.0.0.1:1111
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-local[5692]: tcp port reuse enabled
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-local[5692]: udprelay enabled
Tue Aug 11 10:05:09 2020 daemon.err /usr/bin/ss-local[5692]: [udp] getaddrinfo: System error
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-local[5692]: running from root user
Tue Aug 11 10:05:09 2020 user.info : using tcp fast open
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5693]: enable TCP no-delay
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5693]: resolving hostname to IPv6 address first
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5693]: initializing ciphers... chacha20-ietf-poly1305
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5693]: listening at 0.0.0.0:1100
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5693]: tcp port reuse enabled
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5693]: UDP relay enabled
Tue Aug 11 10:05:09 2020 daemon.err /usr/bin/ss-redir[5693]: [udp] getaddrinfo: System error
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5693]: running from root user
Tue Aug 11 10:05:09 2020 user.info : using tcp fast open
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5694]: enable TCP no-delay
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5694]: resolving hostname to IPv6 address first
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5694]: initializing ciphers... chacha20-ietf-poly1305
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5694]: listening at 0.0.0.0:1100
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5694]: tcp port reuse enabled
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5694]: UDP relay enabled
Tue Aug 11 10:05:09 2020 daemon.err /usr/bin/ss-redir[5694]: [udp] getaddrinfo: System error
Tue Aug 11 10:05:09 2020 daemon.info /usr/bin/ss-redir[5694]: running from root user
Tue Aug 11 10:05:09 2020 daemon.err openvpn(omr)[16536]: event_wait : Interrupted system call (code=4)
Tue Aug 11 10:05:09 2020 daemon.notice openvpn(omr)[16536]: /sbin/ifconfig tun0 0.0.0.0
Tue Aug 11 10:05:09 2020 daemon.debug dnsmasq[21990]: stopped listening on tun0(#106): 10.255.252.6 port 53
Tue Aug 11 10:05:09 2020 daemon.notice netifd: Network device 'tun0' link is down
Tue Aug 11 10:05:09 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Tue Aug 11 10:05:09 2020 daemon.notice netifd: Interface 'omrvpn' is now down
[... cut ...]
Tue Aug 11 10:11:04 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.1.1 dev eth1
Tue Aug 11 10:11:10 2020 user.notice post-tracking-post-tracking: wan1 (eth1) switched up
Tue Aug 11 10:11:10 2020 user.notice post-tracking-post-tracking: Set server vps (54.38.188.175) route via 192.168.1.1 metric 3
Tue Aug 11 10:11:18 2020 daemon.err /usr/bin/ss-redir[24557]: remote recv: Connection reset by peer
Tue Aug 11 10:12:18 2020 daemon.err /usr/bin/ss-redir[24557]: remote recv: Operation timed out`

Ysurac, 2020-08-11:
" OMR-VPS: You can't redirect ports >= 65000, they are needed by OpenMPTCProuter Server part"
Donc il doit y avoir un port > 65000 qui est définis dans les régles. Effectivement c'est le seul cas où je vire une régle sauvagement (peut être trop sauvagement).

AdrienDssy, 2020-08-11:
J'ai mis 65000 comme port max il faut que je modifie par 64999 ?

Ysurac, 2020-08-11:
oui, à partir de 65000 c'est réservé.

AdrienDssy, 2020-08-11:
ça disparaît quand même si je met 64999 mais sur le VPS c'est bien modifier

Ysurac, 2020-08-11:
Ca doit pas la. Regarde à nouveau dans le journal système si il y a le même message (ou un autre).

AdrienDssy, 2020-08-11:
Non j'ai plus le message (enfin de ce que j'ai vu)
>Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-local[22765]: udp port reuse enabled
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-local[22765]: running from root user
Tue Aug 11 12:07:14 2020 user.info : using tcp fast open
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22766]: enable TCP no-delay
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22766]: resolving hostname to IPv6 address first
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22766]: initializing ciphers... chacha20-ietf-poly1305
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22766]: listening at :::1100
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22766]: tcp port reuse enabled
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22766]: UDP relay enabled
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22766]: udp port reuse enabled
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22766]: running from root user
Tue Aug 11 12:07:14 2020 user.info : using tcp fast open
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22767]: enable TCP no-delay
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22767]: resolving hostname to IPv6 address first
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22767]: initializing ciphers... chacha20-ietf-poly1305
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22767]: listening at :::1100
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22767]: tcp port reuse enabled
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22767]: UDP relay enabled
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22767]: udp port reuse enabled
Tue Aug 11 12:07:14 2020 daemon.info /usr/bin/ss-redir[22767]: running from root user
Tue Aug 11 12:07:14 2020 user.notice Shadowsocks: Rules DOWN
Tue Aug 11 12:07:14 2020 daemon.err openvpn(omr)[5907]: event_wait : Interrupted system call (code=4)
Tue Aug 11 12:07:14 2020 daemon.notice openvpn(omr)[5907]: /sbin/ifconfig tun0 0.0.0.0
Tue Aug 11 12:07:14 2020 daemon.debug dnsmasq[10981]: stopped listening on tun0(#118): 10.255.252.6 port 53
Tue Aug 11 12:07:14 2020 daemon.notice netifd: Network device 'tun0' link is down
Tue Aug 11 12:07:14 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Tue Aug 11 12:07:14 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Tue Aug 11 12:07:14 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Tue Aug 11 12:07:14 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Tue Aug 11 12:07:14 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Tue Aug 11 12:07:14 2020 daemon.notice openvpn(omr)[5907]: /usr/libexec/openvpn-hotplug down omr tun0 1500 1562 10.255.252.6 10.255.252.5 init
Tue Aug 11 12:07:14 2020 daemon.notice openvpn(omr)[5907]: SIGTERM[hard,] received, process exiting
Tue Aug 11 12:07:14 2020 daemon.notice openvpn(omr)[23033]: OpenVPN 2.4.9 x86_64-openwrt-linux-gnu [SSL (OpenSSL)] [LZO] [LZ4] [EPOLL] [MH/PKTINFO] [AEAD]
Tue Aug 11 12:07:14 2020 daemon.notice openvpn(omr)[23033]: library versions: OpenSSL 1.1.1g  21 Apr 2020, LZO 2.10
Tue Aug 11 12:07:14 2020 daemon.warn openvpn(omr)[23033]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Tue Aug 11 12:07:14 2020 daemon.warn openvpn(omr)[23033]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Tue Aug 11 12:07:14 2020 daemon.notice openvpn(omr)[23033]: TCP/UDP: Preserving recently used remote address: [AF_INET]54.38.188.175:65301
Tue Aug 11 12:07:14 2020 daemon.notice openvpn(omr)[23033]: Attempting to establish TCP connection with [AF_INET]54.38.188.175:65301 [nonblock]
Tue Aug 11 12:07:14 2020 user.notice omr-tracker: Launching...
Tue Aug 11 12:07:14 2020 user.notice omr-tracker: Launched
Tue Aug 11 12:07:14 2020 daemon.debug dnsmasq[10981]: stopped listening on ifb4eth1(#127): fe80::7879:a9ff:fe95:90fe%ifb4eth1 port 53
Tue Aug 11 12:07:14 2020 daemon.debug dnsmasq[10981]: stopped listening on lo(#1): ::1 port 53
Tue Aug 11 12:07:14 2020 daemon.debug dnsmasq[10981]: stopped listening on eth1(#10): fe80::210:18ff:fe8c:92fa%eth1 port 53
Tue Aug 11 12:07:14 2020 daemon.debug dnsmasq[10981]: stopped listening on eth2(#11): fe80::210:18ff:fe8c:92fc%eth2 port 53
Tue Aug 11 12:07:14 2020 daemon.debug dnsmasq[10981]: stopped listening on eth4(#13): fd88:d699:d82f::1 port 53
Tue Aug 11 12:07:14 2020 daemon.debug dnsmasq[10981]: stopped listening on eth4(#13): fe80::2e0:4cff:feb0:f852%eth4 port 53
Tue Aug 11 12:07:14 2020 daemon.debug dnsmasq[10981]: stopped listening on 6in4-omr6in4(#117): fe80::a00:2%6in4-omr6in4 port 53
Tue Aug 11 12:07:14 2020 daemon.debug dnsmasq[10981]: stopped listening on 6in4-omr6in4(#117): fe80::aff:fb02%6in4-omr6in4 port 53
Tue Aug 11 12:07:14 2020 user.notice omr-6in4: Disable IPv6
Tue Aug 11 12:07:14 2020 daemon.notice netifd: Interface 'omr6in4' is now down
Tue Aug 11 12:07:14 2020 user.notice ModemManager: hotplug: remove network interface 6in4-omr6in4: event processed
Tue Aug 11 12:07:14 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=6in4-omr6in4, subsystem=net
Tue Aug 11 12:07:14 2020 daemon.info vnstatd[6440]: Info: SIGTERM received, exiting.
[... cut ...]
Tue Aug 11 12:20:56 2020 daemon.info dnsmasq[16146]: read /etc/hosts - 4 addresses
Tue Aug 11 12:20:56 2020 daemon.info dnsmasq[16146]: read /tmp/hosts/odhcpd - 0 addresses
Tue Aug 11 12:20:56 2020 daemon.info dnsmasq[16146]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Tue Aug 11 12:20:56 2020 daemon.info dnsmasq-dhcp[16146]: read /etc/ethers - 0 addresses
Tue Aug 11 12:20:56 2020 user.notice omr-bypass: OMR-ByPass is running
Tue Aug 11 12:20:57 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)

AdrienDssy, 2020-08-11:
et la v6 toujours non fonctionnelle

Ysurac, 2020-08-11:
Il n'y a rien qui peut virer la régle la.
Pour l'IPv6 ce n'est pas le même soucis. Si il n'y a pas d'erreur lorsqu'il est activé dans la page Système->OpenMPTCProuter onglet Etat, alors il est fonctionnel sur le routeur. Le soucis est au niveau de la propagation au client. Voir dans Réseau->Interface puis Lan et onglet Serveur DHCP et onglet Paramètres IPv6 ce qui est activé ou pas.

AdrienDssy, 2020-08-11:
Oui l'IPv6 l'était déjà et l'est toujours

Ysurac, 2020-08-11:
Il va falloir être plus précis. Le "Service de publicité de routeur" et "DHCPv6-Service" sont réglés comment ?

Ysurac, 2020-08-11:
Ainsi que "Toujours annoncer le routeur par défaut" est bien coché ?

AdrienDssy, 2020-08-11:
[image]

Ysurac, 2020-08-11:
Peux tu lancer sur OpenMPTCProuter via ssh ou sur un poste client ```tcpdump -vvvv -ttt -i eth0 icmp6 and 'ip6[40] = 134'``` et le laisser lancer tant qu'il n'y a pas un ou 2 messages ?
La commande permet de capturer les paquets Router Advertisement si il y en a.

AdrienDssy, 2020-08-11:
[image]

AdrienDssy, 2020-08-11:
ça fait rien d'autre que ça

Ysurac, 2020-08-11:
Etrange.
Dans Réseau->Interfaces et l'onglet Options globales de réseau, qu'elle est la valeur de "Préfixe ULA IPv6" ?

AdrienDssy, 2020-08-11:
fd88:d699:d82f::1/60

Ysurac, 2020-08-11:
Et si tu remplaces par ```fd88:d699:d82f::/48``` ?

AdrienDssy, 2020-08-11:
Non ça change rien, j'utilise OpenVPN ça peut peut être dû à ça non ?

Ysurac, 2020-08-11:
En 3 minutes ça semble court pour voir le résultat. Les RA ne sont pas envoyés toute les 5s ;)
Si tout est vert ça veut dire que l'IPv6 fonctionne sur le routeur. Visiblement le soucis est au niveau de la distribution des IPv6 privées.

AdrienDssy, 2020-08-11:
Comment y remédié du coup ? :+1:

Ysurac, 2020-08-11:
Sur le comment y remédié, c'est ce que j'essaye de faire depuis déjà un moment en te faisant faire des tests. Il me faudra ta configuration avec IPv6 activé dispo dans Système->OpenMPTCProuter puis onglet "Voir tous les paramètres" (ne pas oublier de virer les clefs ou au moins l'IP du serveur)

Ca passera toujours via un tunnel (en tout cas tant que j'aurais pas implémenté le support du full IPv6 avec WAN et serveur en IPv6 only).
Mais sinon il est possible si on possède un /48 d'utiliser un /56 sur le routeur et d'avoir donc des IPs publiques.

AdrienDssy, 2020-08-11:
> dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.ra='server'
dhcp.lan.dhcpv6='server'
dhcp.lan.ra_preference='high'
dhcp.lan.ra_management='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].mac='3C:D9:2B:0B:9F:E2'
dhcp.@host[0].dns='1'
dhcp.@host[0].ip='192.168.100.139'
dhcp.@host[0].name='Ded-Adky'
dhcp.@host[1]=host
dhcp.@host[1].name='adky-vps-linux'
dhcp.@host[1].dns='1'
dhcp.@host[1].mac='0E:FD:5C:A0:66:85'
dhcp.@host[1].ip='192.168.100.154'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
[... 1387 lines cut ...]
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='0'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth0' 'wan4' 'eth2' 'eth1' 'eth2'
vnstat.@vnstat[0].backup='0'

closed 2020-08-12 by AdrienDssy