Ysurac/openmptcprouter #1129: ACL

Nadeistos, 2020-08-11:
Hi !

ACL doen't work... invalid user/password with new created user even after reboot

  - OpenMPTCProuter version: 0.55
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: x86_64

I'll try with a rasspberry 3

Ysurac, 2020-08-11:
ACL is working but you need to make sure that you grant access to all needed ACL group.

Nadeistos, 2020-08-11:
Yes i done it ^^
no user created in /etc/passwd, normal ?

Ty

Ysurac, 2020-08-11:
Yes. But if a new user is created via SSH, you can use it if you select "Use Unix password" in Password variant when you create an account.

Nadeistos, 2020-08-11:
Je la fais en fr, désolé :/
Alors création utilisateur par mot simple ou password UNIX après creation SSH toujours la même erreur de mdp...
Cela fonctionne avec un mot de passe crypté par hachage, par contre sur le menu "niveau d'acces" le choix "complet" ou "lecture seule" cela donne une erreur 403, par contre en passant via "individuel" les options "lecture seule" ou "acces complet"  tout fonctionne...
Petit détail... comme le login par défaut arrive sur luci-mod-status-index, si il est désactivé on a forcement une erreur 403

Ysurac, 2020-08-11:
Oui j'avais remarqué ces soucis. Ce module provient d'OpenWrt, et effectivement il faut définir les niveaux d’accès manuellement et seul le mot de passe chiffré par hachage fonctionne aussi pour moi. Le principal c'est que ça fonctionne ;)
Il faudra que je regarde si le comportement est le même avec OpenWrt ou si c'est lié à des modifications que j'ai pu faire.

Nadeistos, 2020-08-11:
Oui ça fonctionne nickel :) Apres avoir installé toutes les betas et bricolé un peu, j'avais peur que cela vienne de chez moi
merci pour tout !

closed 2020-12-16 by github-actions[bot]