Ysurac/openmptcprouter #1130: update issue

numerobis08, 2020-08-11:
## Expected Behavior

tried to update VPS server running under Debian 10

## Current Behavior
tried with both commands (connected as root) : 
wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh
or

wget https://www.openmptcprouter.com/server/debian10-x86_64.sh
sh debian10-x86_64.sh

Result : error message 
Impossible d’écrire dans « debian10-x86_64.sh » (Système de fichiers accessible en lecture seulement).

Can someone support me ? 
Thank you in advance for the help

Adorfer, 2020-08-12:
when invoking the wget, your current work directory is either
- full (drive has not enough space left)
- the filesystem is damaged/mounted read-only
- the user has not enough rights to write to that directory

kevin39, 2020-08-12:
Yeah, your VPS is r/o only. Thats sound bad but it's not a bug.

Nadeistos, 2020-08-12:
Hi 

df -h to see remaining space and if partition is full try
apt autoremove

numerobis08, 2020-08-12:
I checked and still have available space. 
I opened a ticket to Firstheberg let’s see ... 
otherwise I’ll do a fresh install 
Thank you for quick feedback:-)

Ysurac, 2020-08-12:
You tried a reboot ? I had the same problem with FirstHeberg that was resolved only after a reboot.

xillibit, 2020-08-12:
I have the same issue with my instance on scaleway with debian buster, it'a actually in version 0.1015 5.4.52-mptcp, i have tried the update to 0.1017 but even after a reboot of the instance it stay to 0.10.15. I have checked with a df -h but it's not full

Ysurac, 2020-08-12:
@xillibit if you don't have a read only alert it's not the same issue.
Try to run the script again on the VPS, yesterday CDN wasn't updated and still in 0.1015

numerobis08, 2020-08-12:
> You tried a reboot ? I had the same problem with FirstHeberg that was resolved only after a reboot.

after contacting Firstheberg, they need to perform a fsck manually and now it's working :-) 
Many thanks

closed 2020-08-12 by numerobis08