Ysurac/openmptcprouter #1131: Lost Lan IP after updating Linksys 3200 to 0.55

Corrado56, 2020-08-12:
## Expected Behavior
Selected "Flash image"
selected  "openmptcprouter-v0.55-r0+14022-18b7d87a8f-mvebu-cortexa9-linksys_wrt3200acm-squashfs-sysupgrade.bin" as the image file
Received this message
[image]
Selected "force upgrade"
The modem is working for some two minutes with blinking lights, the stop with only the "Power" on. **"Internet" is not on.**
Power off and power on.
Same result. I cannot connect the LAN side of the modem and start the web interface. It seems no IP is assugned to the phisical port "internet".
If I connect a cable to the WAN side of the modem (port 1,2,3,4) the corrisponding led becomes on and the modem is responding to ping, but from this side of the modem I cannot start the web interface. the answer is connection denied from ip ........
 

At the end of the flashing process the modem should restart with the "internet" port assigned to the previous local IP.

## Current Behavior
the modem is not working at all. 
After a few moments of despair, I found that it's still possible to start the modem from the "alternate" image that it's actually containing 0.54.7.
I don't how to exit from this problem.

## Possible Solution
A new image for Linksys 3200 ?

## Steps to Reproduce the Problem
See above

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.54.7
  - OpenMPTCProuter VPS version:  Debian 10 + MPTCP 0.1017 
  - OpenMPTCProuter platform: Linksys WRT3200 ACM

Ysurac, 2020-08-12:
Can you try from a fresh image ? Without keeping settings ?
There is a change in network configuration for Linksys WRT from swconfig to DSA as you can see in your screenshot.

Corrado56, 2020-08-12:
Boot the modem from alternate.
Selected "Flash image"
selected "openmptcprouter-v0.55-r0+14022-18b7d87a8f-mvebu-cortexa9-linksys_wrt3200acm-squashfs-factory.img" as the image file
Disabled "Keep settings"
Enabled "Force upgrade"
[image]
Unfortunately, same result. 
After the flashing the modem reboot,  check wi fi, but stops without enabling "Internet" (the lan side).
This is the expected behavior, but after the reboot the second light is off
[image]

Corrado56, 2020-08-12:
Hi Yannik, good news. 
After a second manual reboot (not the one forced by the flashing, the router is working, the release is 0.55 and 192.168.100.1 is assigned to "internet" port.
Now I will redo the configuration. The configuration backups from 0.54.7 could be loaded and aree expected to function or it's better to redo ?

Ysurac, 2020-08-12:
You need to redo, the network part configuration is not the same this will fail.
You can copy from the .gz backup files in /etc/config to same directory on the router if you don't want to do all configuration (but not /etc/config/network).

closed 2020-08-12 by Corrado56