Ysurac/openmptcprouter #1132: Using OMR as a gateway for pfsense/opnsense firewall

rootella, 2020-08-12:
## Expected Behavior
Manage NAT and firewalling from opnsense/pfsense

## Context (Environment)
WAN1/2/3 -- OMR -- FIREWALL -- LAN

## Specifications
  - OpenMPTCProuter version: 0.55
  - OpenMPTCProuter VPS version: Latest
  - OpenMPTCProuter platform: x86_64

It would be handy to have a wiki guide with best practices

Ysurac, 2020-08-12:
This will be dual NAT. You can redirect all ports from OpenMPTCProuter to pfsense.
Instead of using pfsense you can use internal OpenMPTCProuter firewall.

uryupinsk, 2020-08-12:
I am doing exactly this, although I may be using OMR firewall exclusively in the future for my somewhat limited firewall needs. I do inter-VLAN routing through an L3 switch and only need 2 port forwards at the moment.

For your needs, you have 2 choices :

- dual NAT: not the best design but simpler, let pfSense do all the routing and NATing stuff
- routing without NAT on pfSense: a little more complicated, port forwarding and NATing is only done on OMR

### Dual NAT solution

1. On pfSense, add a WAN interface with OMR as the default gateway. By default, pfSense will NAT to OMR.
2. If you need port forwarding, in OMR, redirect ports 1-64999 from vpn zone to lan zone, with pfSense WAN IP. Do the actual port forwarding on pfSense.

###  No NAT solution

1. On pfSense, add a WAN interface with OMR as the default gateway.
2. Disable Outbound NAT on pfSense.
3. On OMR, add static route(s) to your LAN network(s).
4. If you need port forwarding, redirect the needed ports to the target LAN hosts _**on OMR**_, then add a Pass WAN firewall rule _**on pfSense**_ to allow inbound traffic to the target host and port.

rootella, 2020-08-12:
Thank you Ysurac, any chance to bypass double nat with some config? Or a better config..

What I've done so far:
- System > OpenMPTCPRouter > Advanced > Redirects all ports from server to this router (on)
- Network > Firewall > General Settings: https://imgur.com/a/pG6VGpa
- Network > Firewall > Port Forwards: https://imgur.com/a/Sg4pswj

uryupinsk, 2020-08-12:
Your port forwards are wrong:

1. Port forwarding only occurs from VPN zone, not WAN zone, so remove your From WAN zone rule.
2. You only forward ports 1-64999, because ports > 64999 are used by OMR.

Ysurac, 2020-08-12:
Thanks @uryupinsk for the answer. Can I add this answer to the doc ? This can help all pfSense users :)

rootella, 2020-08-12:
Thank you @uryupinsk, wan forwarding doesn't help if OMR-bypass is present?
Got a voip trunk that rely on a specific WAN and I don't want to incapsulate into VPN

uryupinsk, 2020-08-12:
@Ysurac Absolutely! Glad I can help some users. :-)

@rootella In this case I think you are right, but maybe @Ysurac can confirm it.

closed 2020-12-16 by github-actions[bot]

psylou, 2023-01-27:
Everything seems to work fine for me (80 and 443) but I can't get Wireguard to work. I don't know why. I can receive  the data in opnsense but it nothing gets received on client side. :( Can anybody help?