Ysurac/openmptcprouter #1133: A few notes about OMR v0.55

uryupinsk, 2020-08-12:
Hi @Ysurac ,

This is not a proper issue but there is no suggestion channel. Feel free to delete it if it concerns you.

I just installed v0.55 and everything is working so far. Upgrading process was seamless.

A few notes and questions :

- I don't think that turning `autorate-ingress` on by default in SQM-QoS is a good idea. It may be a good thing for unstable connections, but for a stable DSL, cable or fiber connection it makes things a lot worse. Maybe add a setting in the wizard to ask if it's a stable connection.
- Have you tested `autorate-ingress` on LTE connections? If yes, what are the results?
- What is the status of Glorytun TCP vs DSVPN? I had to revert to Glorytun TCP back in v0.54.7 due to stability issues with DSVPN, despite its better resources usage and performance.

Thanks a lot for all the very hard work you've been putting these past months. OMR is getting better and better and I am glad to be able to use it.

Ysurac, 2020-08-12:
Did you make some test with and without autorate-ingress on a stable connections ? This shouldn't have a big impact.
More tests may be needed to validate on each connections type, results was quite good on some tests.
I also find Glorytun TCP is more stable than DSVPN, I didn't make many test on that for 0.55. I try to have a better solution for UDP in 0.56

uryupinsk, 2020-08-12:
There is a delay of > 15-20s with `autorate-ingress` before saturating the link with no noticeable improvement of latency. Without it, and with correct bandwidth settings in SQM, I saturate almost instantly the link with +3-5 ms of latency.

I will test `autorate-ingress` with LTE connection and will report there.

Ok, no problem. I will try DSVPN later. When it's working correctly, it's more CPU efficient than Glorytun TCP. What do you plan for UDP next? The big drawback on my LTE connection (Free) is that it throttles UDP, so I need to use a TCP VPN.

uryupinsk, 2020-08-15:
I made a lot of tests last night between VPNs and the results are pretty interesting.

### Test methodology

1. On OMR, WAN 1 is a 100 Mbps cable connection and WAN 2 is a Free Mobile LTE connection.
2. I set up a Wireguard site-to-site VPN between me and a friend's network. He has a 1 Gbps fiber connection and is able to fully saturate his link with Wireguard with another Gigabit network.
3. I choose between Glorytun TCP, DSVPN and Glorytun UDP.
4. I download a file on a HTTP server on his side with 8 concurrent streams.
5. I measure download speeds and latency.

### Notes

On a pure TCP connection (going through Shadowsocks instead of the aforementioned VPNs) :

- download speed : 15-20 mo/s
- latency with unloaded links : 17 ms
- latency with loaded links : 20-23 ms

Using a Wireguard (or OpenVPN) on OMR is not very efficient because of the double overhead for encapsulating and encrypting. Download speed will be more negatively impacted.

### Results

Glorytun TCP :

- download speed : 4-10 mo/s
- latency : 80 to timeout
- CPU usage is very low because of AES-NI support

DSVPN :

- download speed : 4-6 mo/s
- latency : 80 ms to timeout
- CPU usage is higher because AES-NI is not implemented

#### Glorytun UDP

That's the interesting part. I abandoned it a few months ago because stability and speed was horrendous on my Free Mobile LTE WAN, which I think throttles UDP traffic. However here, it has the better throughput of the 3. On the bad side of things, stability is much worse when downloading, with pings skyrocketing even more than with the TCP VPNs, and connection hangs at times. It may be due to out of order packets.

### Conclusion

All 3 VPN solution are suboptimal compared to MPTCP for throughput. I hope more users can share here their experiences, and I want to extensively test next version of OMR if you implement another VPN solution for UDP.

Adorfer, 2020-08-16:
"a few notes about OMRv0.55" is a strange title for a thread wich is dealing about delection of tunneling mechanisms in terms of overhead/throughput and latency. 
in other words: Please change the title to a less generic, matching the topic you like to be discussed here.

uryupinsk, 2020-08-17:
@Adorfer Thanks for your input. Indeed this post was initially for generic notes and remarks on v0.55. I will change the tittle according to the actual topic.

closed 2020-12-20 by github-actions[bot]