Ysurac/openmptcprouter #1134: Not possible to update VPS to 0.1017

xillibit, 2020-08-12:
On my instance on scaleway with debian buster, it'a actually in version 0.1015 5.4.52-mptcp, i have tried the update to 0.1017 but even after a reboot of the instance it stay to 0.10.15. I have checked with a df -h but it's not full

_Originally posted by @xillibit in https://github.com/Ysurac/openmptcprouter/issues/1130#issuecomment-672791699_

Ysurac, 2020-08-12:
try a ```wget https://www.openmptcprouter.com/server/debian10-x86_64.sh```on the VPS and check version at the beginning defined by "OMR_VERSION" variable.
Try a ```ping www.openmptcprouter.com``` to check if it correctly resolve to the CDN.

xillibit, 2020-08-12:
It resolve correctly the CDN : 

[image]

I have done a `wget https://www.openmptcprouter.com/server/debian10-x86_64.sh` and checked the code, it's right the 0.1017 version : 

[image]

I have launched the script but at the end i'am disconnected  : 

[image]

Ysurac, 2020-08-12:
You can try to run the script using ```screen``` (```apt-get install screen``` if not already installed), this will keep the script running even if you have a connection error.

xillibit, 2020-08-12:
I have tired a new time with screen, it has passed this time to 0.1017 but without finish

Ysurac, 2020-08-12:
without finish ? You have an error ?

xillibit, 2020-08-12:
No i don't have an error, i have closed the session but it was at the same place like on the screen above

Ysurac, 2020-08-12:
And the script stopped inside the screen app ?

xillibit, 2020-08-12:
yes it has stopped inside, i have launched screen then pressed the space and launched the command sh debian10-x86_64.sh

Ysurac, 2020-08-12:
Strange. You looked at the result of ```dmesg``` ? maybe it was killed by kernel

xillibit, 2020-08-12:
How do-i look the result of dmesg ?

Ysurac, 2020-08-12:
By typing ```dmesg``` under SSH on the VPS.

xillibit, 2020-08-12:
I show plenty of lines like that :  

[image]

Ysurac, 2020-08-12:
It's firewall log, nothing special here.

Adorfer, 2020-08-13:
i do not know if it is related: 
i had a breaking vps after going from vpn revision (the one up to date when b5 was out), not chance to connect with the router, even a full reinstall of the router (sysupgrade -n) did not help, all settings looked fine, status screen was "connected", but neither incoming ports were forwareded, nor did the lan work for longer than 10-20s each time of restart oder reconfig/reload. 
(the server#s ip was used for the clients meanwhile). 
Tried to disable shadowsocks, or changed masters on the interfaces, etc , no luck
Since i had no time to dig into it it, i killed the vm and did a full reinstall: works perfect now, better than before in terms of latency/packetloss (at least in my environment)

or to put it a different way: since there is not much stateful data in the vps: if it does not work after upgrade: tear it down and make it new.

closed 2020-08-15 by xillibit