Ysurac/openmptcprouter #1135: How to setup WIFI AP on linksys 3200

Corrado56, 2020-08-12:
## Expected Behavior
This is not an issue, it's an help request.

I am trying to define a WIFI AP on a linksys 3200 

## Current Behavior
I have activated the radio signal and associated to a Custom SSID. 
I succesfully connect to the Wifi, BUT "Internet not available".
[image]

I have no idea how an interface should be created to associate the WIFI signal to the LAN side of the switch.

Please note that I had successful created a Client DHCP interface to add a phone on WAN side, this is not the problem I have now.

## Specifications

  - OpenMPTCProuter version: 0.55
  - OpenMPTCProuter VPS version:  Debian 10 mptcp 0.1017
  - OpenMPTCProuter platform: Linksys

Ysurac, 2020-08-12:
The doc for RPI may help: https://github.com/Ysurac/openmptcprouter/wiki/Configure-Wifi
Else I will try to do a doc for WRT ;)

Corrado56, 2020-08-12:
Hi Yannik, 
I made a lot of test following RPI documentation but without success. 
I defined the wifi_ap interface as static address with an ip 192.168.100.2 gateway 192.168.100.1 (the lan IP), **DHCP enabled**.
I was able to connect to the wifi, but the wifi interface of the local computer (the client) never received a valid IP from MPTCPRouter
[image]
--
[image]

Ysurac, 2020-08-12:
If you set a static IP it's working ?

Corrado56, 2020-08-12:
No
I tested both gateway 192.168.100.1 and .2
[image]

Ysurac, 2020-08-13:
By following the RPI wifi guide I have a working AP on WRT3200ACM.
I think the problem is related to your wifi_ap interface configuration.

Rwestrem, 2020-08-13:
I had to delete and then re add the ssid's from each radio in order to get an IP from wifi.  Wifi seems to drop data a lot though even after this fix.  It will stay connected but show no internet available then comes and goes.

Corrado56, 2020-08-14:
Fixed with Rwestrem suggestion. (Thanks)
Deleting all the default SSID's, then recreating ex novo. 
And now I receive an IP from DHCP (!) and navigate

closed 2020-08-14 by Corrado56

Rwestrem, 2020-08-17:
> Fixed with Rwestrem suggestion. (Thanks)
> Deleting all the default SSID's, then recreating ex novo.
> And now I receive an IP from DHCP (!) and navigate

Is your wifi stable?  Mine was unstable so I used an AP instead.  If yours is stable I may investigate my setup.

Corrado56, 2020-08-17:
It's actually stable, with the problem of issue #1152