Ysurac/openmptcprouter #1136: mise a jour

philoun8, 2020-08-12:
Bonsoir

je viens pour un petit souci depuis la mise a jour d'OpenMPTCProuteur en version 0.55, que j'ai fais aujourd'hui

J'ai le message "more than one default VPN is enabled", alors que j'ai uniquement Glorytun d'activer!

[image]

et de ce fait les port ne s'ouvrent pas depuis l'exterieur

Ysurac, 2020-08-12:
Il faut faire le tour des VPN dans le menu VPN pour voir si un autre VPN n'est pas activé.

Merci d'utiliser le template à l'avenir.

philoun8, 2020-08-12:
c'est ce que je fais avant de poster et évidement tous les vpn sont désactivés. mais bon a priori mes questions dérangent donc je vais arrêter d'en poser

philoun8, 2020-08-12:
j'ai fais le tour des vpn et comme j'avais précisé, seul Glorytun est activé.
Et même en désactivant TOUS les vpn j'ai le même message

Ysurac, 2020-08-12:
Il me faudrait la configuration, dans Système->OpenMPTCProuter puis onglet "Voir tous les paramètres" (attention de supprimer l'IP du serveur et/ou les clefs). Peut être un reste d'une ancienne version...

philoun8, 2020-08-12:
je pense avoir tout viré

`
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/kobo.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/canalplus.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/jeuxvideo.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/tenor.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/netflix.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflxext.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflximg.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflximg.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflxvideo.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/nflxso.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/s3.ll.dash.row.aiv-cdn.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/s3-dub.cf.dash.row.aiv-cdn.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/dmqdd6hw24ucf.cloudfront.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/d25xi40x97liuc.cloudfront.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/aiv-delivery.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/aiv-cdn.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/1s3.lvlt.dash.us.aiv-cdn.net.c.footprint.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/s.loris.llnwd.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/atv-ext.amazon.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/c.media-amazon.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/wbagora.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/wbplay.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/e13555.b.akamaiedge.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/e1800.d.akamaiedge.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/e1879.e7.akamaiedge.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/playstation.net/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/playstation.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/sonyentertainmentnetwork.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/android.clients.google.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1' '/worldofwarcraft.com/omr_dst_bypass_wan1,omr6_dst_bypass_wan1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].mac='*************'
dhcp.@host[0].name='***'
dhcp.@host[0].dns='1'
dhcp.@host[0].ip='192.168.100.184'
dhcp.@host[1]=host
dhcp.@host[1].mac='************'
dhcp.@host[1].name='***'
dhcp.@host[1].dns='1'
dhcp.@host[1].ip='192.168.100.202'
dhcp.@host[2]=host
dhcp.@host[2].mac='************'
dhcp.@host[2].name='*****'
dhcp.@host[2].dns='1'
dhcp.@host[2].ip='192.168.100.230'
dhcp.@host[3]=host
dhcp.@host[3].mac='************'
dhcp.@host[3].name='****'
dhcp.@host[3].dns='1'
dhcp.@host[3].ip='192.168.100.192'
dhcp.@host[4]=host
dhcp.@host[4].mac='**********'
dhcp.@host[4].name='***'
dhcp.wwan=dhcp
dhcp.wwan.start='100'
dhcp.wwan.leasetime='12h'
dhcp.wwan.limit='150'
dhcp.wwan.interface='wwan'
dhcp.@host[13]=host
dhcp.@host[13].mac='***********'
dhcp.@host[13].name='*****'
dhcp.@host[13].dns='1'
dhcp.@host[13].ip='192.168.8.2'
dhcp.@host[14]=host
[... cut ...]
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='wan1' 'wan2'
vnstat.@vnstat[0].backup='0'
wireless.radio0=wifi-device
wireless.radio0.type='mac80211'
wireless.radio0.path='platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1'
wireless.radio0.hwmode='11g'
wireless.radio0.txpower='13'
wireless.radio0.country='FR'
wireless.radio0.channel='auto'
Powered by openmptcprouter v0.55 r0+14022-18b7d87a8f
`

Ysurac, 2020-08-12:
Le VPN définis est MLVPN mais c'est Glorytun UDP qui est activé. C'est surement le soucis.
Vu que MLVPN n'est plus proposé dans les choix car il crash au démarrage, il faut dans Système->OpenMPTCProuter onglet Assistant de configuration coché "Configuration avancé" et choisir un VPN puis revalider la page. Je pense que ça réglera le soucis.

philoun8, 2020-08-12:
nickel ca a fonctionné, merci

[image]

Ysurac, 2020-08-12:
ok parfait, il faudrait que je mette un message plus explicite ou que je change le comportement...

closed 2020-08-12 by Ysurac