Ysurac/openmptcprouter #1137: VnStat Database - Search Best practice

oOSphaXOo, 2020-08-12:
## Current Behavior
Hi @Ysurac  This is not a proper issue but there is no suggestion channel or HowTo configure channel.

After all upgrade of OpenMPTCPRouter vnstat's database is flush ... I search a best practice to backup / externalise database or all other suggestion to keep this database.

I have search and found some possibility: 
 * Modify /etc/vnstat.conf and define an other Database_dir. But I need modify this every time I upgrade or this file was backup with OpenMPTCPRouter GUI ? Can I use nfs share / usb key / other to store database ?
 * use a backup_script like /etc/init.d/vnstat_backup which don't work. But I need recreate It after upgrade

Which solution can I use to backup or to external store this database ???

## Specifications

  - OpenMPTCProuter version: latest 0.55 (and anterior)
  - OpenMPTCProuter VPS version: latest 0.1017 (and anterior)
  - OpenMPTCProuter platform:

Ysurac, 2020-08-13:
vnstat backup script should work. You checked "Save vnstats stats" in System->OpenMPTCProuter, "Advanced settings" tab ?
If it's the case the backup file is then preserved for update and restored at boot.

oOSphaXOo, 2020-08-13:
My bad ... I don't see this option. I search it in "vnstat menu configuration" ...
I try reboot and backup is done. I monitor it at the next update if it's OK (I'll create a schedule script to externalise it) 

Thanks for your works

closed 2020-08-13 by oOSphaXOo