Ysurac/openmptcprouter #1139: Wifi as wan

Rwestrem, 2020-08-13:
## Expected Behavior
I would like to have my wan3 on openMPTCP Linksys wrt3200 wifi tether to my phone hotspot.  This is normally simple to do with just scanning the wifi and joining and creating a wwan.  Im not sure how the correct procedure to do this would be.

## Current Behavior

when joining a wifi network as wan it just creates a new wwan outside of the wizard.

Rwestrem, 2020-08-13:
I figured it out.

closed 2020-08-13 by Rwestrem

Adorfer, 2020-08-14:
Could you perhaps give a hint where to improve the documentation?
In order to prevent other people to trip over the same difficulty.

Rwestrem, 2020-08-14:
> Could you perhaps give a hint where to improve the documentation?
> In order to prevent other people to trip over the same difficulty.

It was very simple.  I was just unfamiliar with the way to do it.  I didn't realize there was a way to just enable it in network/mptcp.