Ysurac/openmptcprouter #1140: unable to update to 0.1017

vinceducat, 2020-08-13:
## Expected Behavior
update 0.1015 to 0.1017

## Current Behavior
after launch update via debian script
still in 0.1015

## Possible Solution

## Steps to Reproduce the Problem

  1. sh debian10-x86_64.sh
  2. end with no erro
  3.restart vps
  4. still in 0.1015... :-(

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.55
  - OpenMPTCProuter VPS version: 0.1015
  - OpenMPTCProuter platform:  x86_64 on debian 10 in milkywan VPS

Ysurac, 2020-08-14:
try a ```wget https://www.openmptcprouter.com/server/debian10-x86_64.sh``` on the VPS and check version at the beginning defined by "OMR_VERSION" variable.
Try a ```ping www.openmptcprouter.com``` to check if it correctly resolve to the CDN.
If the version is 0.1017, you can try to relaunch it. What do you have at the end when you launch it ?

vinceducat, 2020-08-14:
OMR_VERSION was 0.1017

ping not test my vps is in rebuilding by milky ;-) ...

to do a complete noew install ...

vinceducat, 2020-08-16:
hello after complete vps reinstall seems to be ok but i have question about low rate : 

when my vps is down, i have un flowrate a 250-300mbit/s througt OMR, but now throught my vps  i have only 45mbit/s :-(
@ycarus do you think it is a problem of perf on OMR or on OMR-vps

OMR is 0.55 Pentium(R) Dual-Core CPU T4400 @ 2.20GHz
OMR-VPS is milkywan VM Bidouille : 1vCPU, 1Go de RAM, 30Go de HDD

vinceducat, 2020-08-16:
@Ysurac seems to be a problem of perf bidouille is not recommed over 100mbit ... you can closed this issu

closed 2020-08-16 by vinceducat