Ysurac/openmptcprouter #1141: [FR] translation: german

Adorfer, 2020-08-13:
## Expected Behavior
language in LUCI beeing consistent, according to the web browser preferred language, here: german ("deutsch-de").

## Current Behavior
only the openwrt-luci part of the framework are localized to german, most openmptcprouter specific dialogues are in english language.

## Possible Solution
 1. please see complete(!) proposal at https://weblate.openmptcprouter.com/languages/de/omr/
 2. other users are invited to contribute too

## Steps to Reproduce the Problem

  see above.  

## Context (Environment)

  Firefox Webbroser, set to preferred language "de", additional languages "en" and "fr"

## Specifications

  - OpenMPTCProuter version: OpenMPTCProuter v0.55 "release"
  - OpenMPTCProuter VPS version: not relevant
  - OpenMPTCProuter platform: not relevant

Ysurac, 2020-08-14:
I will add it to next release (0.55.1) fix some fixes in a few days.

Ysurac, 2020-08-18:
I forgot to add it... This will be for next release :)

Quenty-tolosan, 2020-08-18:
Thanks, it's really appreciated :) Got more time to finish

closed 2020-08-20 by Adorfer