Ysurac/openmptcprouter #1142: OMR bypass Hulu

Rwestrem, 2020-08-14:
## Expected Behavior

Hulu bypass protocol should direct hulu over wan instead of VPN
## Curent Behavior

It is still going over the VPN
## Possible Solution

Anyone have any solutions or could they share the domains they use to bypass Hulu?
## Steps to Reproduce the Problem

 Use the protocol

Ysurac, 2020-08-14:
Check that you use OpenMPTCProuter IP as DNS.

I can't test hulu (not available in France), but, if DNS is OK, you can check using your browser developer tools what domain is/are blocked.

cron410, 2020-08-20:
I'm having the same issue after upgrading to v0.55.1 today. Wife was watching Hulu just before the upgrade, went to lunch, I upgraded to 0.55.1 and we have been unable to stream Hulu since. 

Here's my OMR-bypass settings for Hulu, Netflix, and Disney+

```
omr-bypass.@dpis[0]=dpis
omr-bypass.@dpis[0].proto='NetFlix'
omr-bypass.@dpis[0].interface='eth1'
omr-bypass.@dpis[0].note='Netflix wont allow VPS'
omr-bypass.@domains[0]=domains
omr-bypass.@domains[0].name='netflix.com'
omr-bypass.@domains[0].interface='eth1'
omr-bypass.@domains[1]=domains
omr-bypass.@domains[1].name='netflix.net'
omr-bypass.@domains[1].interface='eth1'
omr-bypass.@domains[2]=domains
omr-bypass.@domains[2].name='nflxext.com'
omr-bypass.@domains[2].interface='eth1'
omr-bypass.@domains[3]=domains
omr-bypass.@domains[3].name='nflximg.com'
omr-bypass.@domains[3].interface='eth1'
omr-bypass.@domains[4]=domains
omr-bypass.@domains[4].name='nflxvideo.com'
omr-bypass.@domains[4].interface='eth1'
omr-bypass.@domains[5]=domains
omr-bypass.@domains[5].name='nflxso.com'
omr-bypass.@domains[5].interface='eth1'
omr-bypass.@domains[6]=domains
omr-bypass.@domains[6].name='nflxsearch.net'
omr-bypass.@domains[6].interface='eth1'
omr-bypass.@domains[7]=domains
omr-bypass.@domains[7].name='ip2location.com'
omr-bypass.@domains[7].interface='eth1'
omr-bypass.@asns[0]=asns
omr-bypass.@asns[0].asn='2906'
omr-bypass.@asns[0].interface='eth1'
omr-bypass.@asns[1]=asns
omr-bypass.@asns[1].asn='55095'
omr-bypass.@asns[1].interface='eth1'
omr-bypass.@asns[2]=asns
omr-bypass.@asns[2].asn='40027'
omr-bypass.@asns[2].interface='eth1'
omr-bypass.@asns[3]=asns
omr-bypass.@asns[3].asn='394406'
omr-bypass.@asns[3].interface='eth1'
omr-bypass.@asns[4]=asns
omr-bypass.@asns[4].interface='all'
omr-bypass.@domains[8]=domains
omr-bypass.@domains[8].name='hulu.com'
omr-bypass.@domains[8].interface='eth1'
omr-bypass.@domains[9]=domains
omr-bypass.@domains[9].name='huluim.com'
omr-bypass.@domains[9].interface='eth1'
omr-bypass.@domains[10]=domains
omr-bypass.@domains[10].name='huluad.com'
omr-bypass.@domains[10].interface='eth1'
omr-bypass.@domains[11]=domains
omr-bypass.@domains[11].name='akamaihd.net'
omr-bypass.@domains[11].interface='eth1'
omr-bypass.@domains[12]=domains
omr-bypass.@domains[12].name='disneyplus.com'
omr-bypass.@domains[12].interface='eth1'
omr-bypass.@domains[13]=domains
omr-bypass.@domains[13].name='bamgrid.com'
omr-bypass.@domains[13].interface='eth1'
omr-bypass.@domains[14]=domains
omr-bypass.@domains[14].name='bam.nr-data.net'
omr-bypass.@domains[14].interface='eth1'
omr-bypass.@domains[15]=domains
omr-bypass.@domains[15].name='cdn.registerdisney.go.com'
omr-bypass.@domains[15].interface='eth1'
omr-bypass.@domains[16]=domains
omr-bypass.@domains[16].name='cws.conviva.com'
omr-bypass.@domains[16].interface='eth1'
omr-bypass.@domains[17]=domains
omr-bypass.@domains[17].name='d9.flashtalking.com'
omr-bypass.@domains[17].interface='eth1'
omr-bypass.@domains[18]=domains
omr-bypass.@domains[18].name='disney-portal.my.onetrust.com'
omr-bypass.@domains[18].interface='eth1'
omr-bypass.@domains[19]=domains
omr-bypass.@domains[19].name='disneyplus.bn5x.net'
omr-bypass.@domains[19].interface='eth1'
omr-bypass.@domains[20]=domains
omr-bypass.@domains[20].name='disney-plus.net'
omr-bypass.@domains[20].interface='eth1'
omr-bypass.@domains[21]=domains
omr-bypass.@domains[21].name='dssott.com'
omr-bypass.@domains[21].interface='eth1'
omr-bypass.@domains[22]=domains
omr-bypass.@domains[22].name='js-agent.newrelic.com'
omr-bypass.@domains[22].interface='eth1'
omr-bypass.@domains[23]=domains
omr-bypass.@domains[23].name='adobedtm.com'
omr-bypass.@domains[23].interface='eth1'
omr-bypass.lo=interface
omr-bypass.lo.id='1'
omr-bypass.eth0=interface
omr-bypass.eth0.id='2'
omr-bypass.eth1=interface
omr-bypass.eth1.id='3'
omr-bypass.eth2=interface
omr-bypass.eth2.id='4'
omr-bypass.tun0=interface
omr-bypass.tun0.id='5'
```

cron410, 2020-08-20:
Update: Netflix is also having issues. Seems OMR Bypass may not be working at all in 0.55.1

Adorfer, 2020-08-21:
Confirm for "by asn" on 0.55

Ysurac, 2020-08-21:
There is an issue. I'm working on this and will compile a new release asap.

Ysurac, 2020-08-21:
A new image is compiling... This should take some hours. Then I will test and this will be released.

closed 2020-12-24 by github-actions[bot]