Ysurac/openmptcprouter #1144: Unable to u

LeprovostNoam, 2020-08-14:
## Expected Behavior
The router should start correctly

## Current Behavior
Unable to access gateway at 192.168.100.1, no internet access.

## Possible Solution
I've try to reset the router using the red button, not working

## Steps to Reproduce the Problem

  1. Upload .bin
  2. start update

## Context (Environment)
I'm trying to update the router

## Specifications

  - OpenMPTCProuter version: v0.55
  - OpenMPTCProuter platform: Linksys WRT3200ACM

Ysurac, 2020-08-14:
After upload it doesn't reboot correctly (or at all?). You need to reboot manually using the I/O button.
If it's an upgrade you can't keep current settings.

LeprovostNoam, 2020-08-14:
I've save my backup, i've tried to restart witht I/O, i've tried to reset but it's not working the Ethernet is not detected

LeprovostNoam, 2020-08-14:
[image]

Ysurac, 2020-08-14:
Are you sure that you don't keep settings while upgrading ?
I've tested on my WRT3200ACM. Try again to upload the image after switching to the other partition: https://openwrt.org/toh/linksys/linksys_wrt3200acm

LeprovostNoam, 2020-08-14:
How can i switch to the other partition? It's the original partition?

Ysurac, 2020-08-14:
It's in firmware recovery section of the link I gave by playing with the power switch.

Adorfer, 2020-08-14:
> [image]
> I've this page and after ERR_ADDRESS_UNREACHABLE

Please try a private browser window or/and a different browser.
(don't ask, just try. if it does not help, please let me know)

LeprovostNoam, 2020-08-14:
I've done it, it start on the openmptcprouter.
 I've update to the last firmware but when it start it always 0.54.7 why?

Adorfer, 2020-08-14:
> i have done it
you have done what?

LeprovostNoam, 2020-08-14:
Now, i can access the router. I try again to update by uploading .img, i uncheck the setting case and check force update but after the router is unavailable and i have to play with the power switch to access and when i go to the settings it always 0.54.7 and not the last version

Adorfer, 2020-08-14:
did you try a private browserwindow in a different browser? yes or no?

LeprovostNoam, 2020-08-14:
Yes, i try again 3 time. Now it's working on 0.55. Do i have to upload the old config or do i have to re-configure manually

Adorfer, 2020-08-14:
you tried what 3 times? reupload? 
never mind, you can just upload the backup you downloaded prior upgrading. 
Or you just put the VPS ip, server-username and server-key and in "advanced" cross "retrieve all keys". 
after that you can retrieve the backup from the vps, in case you pushed one in the past

Ysurac, 2020-08-14:
You have to reconfigure manually on a WRT, network configuration is not the same as previous version.

LeprovostNoam, 2020-08-14:
Oups so i upload the config? So now i have to re install and manually configure

Ysurac, 2020-08-14:
With the old config I don't think WRT can boot.

LeprovostNoam, 2020-08-14:
All is done! Thank you

closed 2020-08-14 by LeprovostNoam