Ysurac/openmptcprouter #1146: UDP traffic overwhelming VPN?

alexciarlillo, 2020-08-14:
## Expected Behavior
UDP traffic should have no adverse impact on overall connection performance / health as long as I am not saturating my connection.

## Current Behavior
When utilizing RTC voice connections (which typically use UDP) I see my connection health severely impacted. I see increased latency on ICMP pings (even on 0.55), and on the UDP packets themselves. This happens with fairly low traffic volume, essentially talking sitting in Discord listening to 3-4 other users speaking. I am not saturating the link at this level of use. I understand ICMP ping is not always reliable and is (or was?) de-prioritized, but the issue itself disappears when not using any VPN with the same level of traffic.

I expect some additional latency from the overhead of encrypting/decrypting the VPN traffic, but what I am experiencing is on the order of 100's of ms additional latency.

## Possible Solution
Disabling the use of VPN fixes this issue. I would like to understand better exactly what this means for my connection though. If it really is just that I have lost the security of the VPN and it is not different than using my connection normally, un-bonded with no VPN, than I am fine with this solution. I believe TCP traffic is still handled by Shadowsocks in this configuration?

I have also verified that the CPU utilization on my VPS has remained low, <5% utilization.

## Steps to Reproduce the Problem

  1. Start ICMP ping to www.google.com (or anywhere)
  2. Join a discord with a couple people talking
  3. Check bandwidth usage either from task manger or router graphs, see you are well within bandwidth capabilities of your connection.
  4. Watch ICMP pings increase in latency.
  5. Connection to discord becomes unstable, also increasing latency and having adverse effects on voice comms.
  6. Game latency also increases.

## Context (Environment)

I have seen many other posts on here about UDP issues when using a VPN, and you generally say it is bad to disable the VPN as a solution, so I was hoping you could expand on why that is. My main goal here is bonding my connections with the lowest possible latency, privacy is absolute a plus, but not required for my needs.

I also develop WebRTC based communications software for a living (not Discord) and have seen this issue manifest there as well. This is just to say I have some experience with this and would be glad to help provide extra information if you can tell me what would be helpful.

## Specifications

  - OpenMPTCProuter version: v0.55 r0+14022-18b7d87a8f
  - OpenMPTCProuter VPS version: Version 0.1017 4.19.104-mptcp
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-08-14:
You can try another VPN and some MPTCP settings (scheduler BLEST and TCP congestion Cubic). You can also enable SQM.
By disabling VPN, anything that is not TCP use direct connection. It's not a good idea because some applications use UDP and TCP, if both doesn't have same exit IP this can make some strange result.

UDP will work better in 0.56.

Your kernel is not the good one for 0.1017, did you reboot the VPS after install ?

alexciarlillo, 2020-08-14:
Ah I did not reboot. I will try that and re-enabling Glorytun TCP to start with. I will find a way to reliably get the exact traffic that is problematic (e.g. using a music bot in Discord) to see if I can reproduce the issue and then carefully work through each VPN and try the different schedulers. I have attempted both already, but not in a very methodical way, so I will gather as much data as possible.

alexciarlillo, 2020-08-14:
I am going to close this issue and look forward to any improvements in 0.56! But while I am here... after restarting my VPS I decided to run a few quick tests. These are far from scientific and need reproduced a few times especially given the variability of LTE, but since I took the time I figured would share my results for posterity and anyone else who is curious. Feel free to stop reading here :)

To summarize: there definitely appears to be some major spikes and deviations from the norm when sending a lot of UDP traffic using Glorytun TCP (possibly reduced with BLEST + Cubic). Glorytun UDP seems to be within the margin of error for my rudimentary testing and is what I am going to use for now. DSVPN does not exhibit the same spiking as Glrytun TCP, but performs worse overall. I don't think changing the scheduler/congestion to BLEST + cubic resulted in any statistically significant difference, but I am going to leave them enabled for now anyway and read up on BLEST.

Results: The only things I really considered here are avg. and mdev. I will likely repeat this a few times over the next couple days, also with longer ping times and also tcping to verify the issue is only impact UDP/ICMP data.

```
No VPN - No RTC Connection

--- 97.107.133.234 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29029ms
rtt min/avg/max/mdev = 51.025/79.028/104.004/15.326 ms

No VPN - Active RTC Connection

--- 97.107.133.234 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29003ms
rtt min/avg/max/mdev = 55.587/69.047/101.908/13.952 ms

=================

Glorytun TCP - No RTC Connection - Default + bbr

--- 97.107.133.234 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29030ms
rtt min/avg/max/mdev = 47.364/76.705/145.502/18.784 ms

Glorytun TCP - Active RTC (UDP) Traffic - Default + bbr

--- 97.107.133.234 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29025ms
rtt min/avg/max/mdev = 55.720/519.142/3831.664/892.679 ms

Glorytun TCP - Active RTC (UDP) Traffic - BLEST + Cubic

--- 97.107.133.234 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29017ms
rtt min/avg/max/mdev = 50.668/151.896/292.451/69.788 ms

Glorytun TCP - BLEST + Cubic - No Traffic

--- 97.107.133.234 ping statistics ---
60 packets transmitted, 60 received, 0% packet loss, time 59058ms
rtt min/avg/max/mdev = 54.495/85.256/178.476/23.285 ms

===========

Glorytun UDP - No RTC Connection - Default + bbr

--- 97.107.133.234 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29028ms
rtt min/avg/max/mdev = 55.514/83.303/121.666/19.115 ms

Glorytun UDP - Active RTC (UDP) Traffic - Default + bbr

--- 97.107.133.234 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29029ms
rtt min/avg/max/mdev = 61.518/90.430/124.942/17.198 ms

Glorytun UDP - Active RTC (UDP) Traffic - BLEST + Cubic

--- 97.107.133.234 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29028ms
rtt min/avg/max/mdev = 46.845/82.548/103.946/16.479 ms

Glorytun UDP - BLEST + Cubic - No Traffic

--- 97.107.133.234 ping statistics ---
60 packets transmitted, 60 received, 0% packet loss, time 59059ms
rtt min/avg/max/mdev = 51.419/82.832/112.303/15.145 ms

=================

DSVPN - No RTC Connection - Default + bbr

--- 97.107.133.234 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29028ms
rtt min/avg/max/mdev = 60.442/107.110/276.460/46.074 ms

DSVPN - Active RTC (UDP) Traffic - Default + bbr

--- 97.107.133.234 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29027ms
rtt min/avg/max/mdev = 66.800/120.126/228.907/33.768 ms

DSVPN - Active RTC (UDP) Traffic - BLEST + Cubic

--- 97.107.133.234 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29001ms
rtt min/avg/max/mdev = 57.126/108.021/257.392/47.528 ms

DSVPN - BLEST + Cubic - No Traffic

--- 97.107.133.234 ping statistics ---
30 packets transmitted, 30 received, 0% packet loss, time 29025ms
rtt min/avg/max/mdev = 61.367/106.120/294.723/43.774 ms

```

closed 2020-08-14 by alexciarlillo