Ysurac/openmptcprouter #1147: ipv6 enabled in openmptcprouter but not working in wan

xillibit, 2020-08-15:
## Expected Behavior
ipv6 should work

## Current Behavior
In openmptcprouter 0.55 i have enabled ipv6, i have no errors displayed in the system state. But when checking ipv6 with https://ip.lafibre.info/ it said that it's not working : 

[image]

The system state : 

[image]

## Possible Solution

## Steps to Reproduce the Problem

  1. Enable ipv6 in LAN2 with is the LAN where computer are present and enable too ipv6 in wizard

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.55
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-08-15:
Do you have an IPv6 on your client computer ?

xillibit, 2020-08-15:
I have that on my computer on the LAN, so yes it has an ipv6 : 

[image]

Ysurac, 2020-08-15:
No, it's should have an IP in fdxx range.
There is an issue in IPv6 RA, but I don't know which and I'm not able to reproduce it for now.

xillibit, 2020-08-15:
What is IPv6 RA ?

I have just looked into interfaces, it seems to be reversed. The IPv6 with prefix fdxx is affected to LAN but it should be on LAN2, the LAN2 has the computers and the LAN has just the SFR box and the huawei B715-23c : 

[image]

Ysurac, 2020-08-15:
IPv6 RA = IPv6 Router Advertisements. The router send to everything that is connected the prefix and route.
But in your case, you need to configure IPv6 on LAN2 in General settings tab.

xillibit, 2020-08-15:
For LAN2 this is the settings : 

[image]

and : 

[image]

For LAN it has IPv6 disabled and DHCP server disabled for IPv4 and IPv6 : 

[image]

and : 

[image]

Ysurac, 2020-08-15:
As LAN2 as no IPv6 configured, this must be set in General settings tab (_Paramètres généraux_).

xillibit, 2020-08-15:
I have configured like that : 

[image]

The computer on lan after renew the dhcp, it still get a fexx IPv6 : 

[image]

Ysurac, 2020-08-15:
fe80 will always be here, it's available when IPv6 is enabled: it's a link local address.
fdxx are unique local address, used for a private IPv6 network.

You can try to set a IPv6 routed prefix, or disabling "builtin IPv6-management" and copy exactly LAN IPv6 config should work.

Adorfer, 2020-08-16:
settings look right on the lan2. 
(it's actually really the same which i run here on my OMR system, despite it's LAN instead of LAN2)
[image]

due to some reason, your windows machine does not learn the prefix. 

please do this on your OMR (you have to replace "eth0" by the physical IF you are using for LAN2. 
(In case tcpdump is not installed, do a _"opkg update; opkg install tcpdump"_, i am not sure if it's included in the default distro)

`tcpdump -vvvv -ttt -i eth0 icmp6 and 'ip6[40] = 134' `

it should give after some 5-10 minutes(!) results like this, where you see the prefix you defined plus as well the name of the OMR defined in the setup (per default: "openmptcprouter")

[image]

xillibit, 2020-08-17:
By the way i have updated to 0.55.1. I'am don't know enough on IPv6 to be sure about IP set on each LAN

I don't know if it's related but the dhcp in IPv6 hasn't given an IPv6 : 

[image]

> tcpdump -vvvv -ttt -i eth0 icmp6 and 'ip6[40] = 134'

It's eth1 the physical interface, so i have tried it but it has stopped by iteslf : 

[image]

Adorfer, 2020-08-25:
for note: On 0.55.2 i had to put "enable-ra" in the /etc/dnsmasq.conf to have working ipv6 router for windows-lan clients. 
(did not investigate further)

Ysurac, 2020-08-25:
By default odhcpd is used for RA and DHCPv6, it's why it's not enabled in dnsmasq.

Adorfer, 2020-08-25:
ok then, the odhcpd did not work on my setup.

Ysurac, 2020-08-25:
I tested with a windows 10 image and I get an IPv6, so the client OS doesn't seems to be the problem.
Can you give me the result of ```uci show dhcp``` via SSH on the router of not working config ?

Adorfer, 2020-08-26:
the full stuff is several hundred lines, but i guess we are looking for the odhcpd? 
in this case the "maindhcp=0" looks a bit strange for me. 

```
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'

```

Ysurac, 2020-08-26:
I need also "dhcp.lan" entry.

Adorfer, 2020-08-27:
```
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.ra_preference='high'
dhcp.lan.leasetime='17220'
dhcp.lan.dns='fd00:2283:524d::1'

```

Ysurac, 2020-08-27:
In this config, "Router Advertisement-Service" is not set to "Server mode" in Network->interface->LAN, DHCP Server tab and IPv6 settings tab.

xillibit, 2020-09-12:
> I tested with a windows 10 image and I get an IPv6, so the client OS doesn't seems to be the problem.
> Can you give me the result of `uci show dhcp` via SSH on the router of not working config ?

[image]

xillibit, 2021-02-20:
I'am now on 0.57.3, my VPS is on scaleway it has an IPv6 :

[image]

Can-i should use it under wizard in IPv6 Prefix ?

[image]

Adorfer, 2021-02-21:
> But still not working :

windows will by default NOT use ipv6 if the networkrange is from the ULA-"fd..."-space. 

change the local range to a different block: 
https://openwrt.org/docs/guide-user/network/ipv6/ipv6.nat6#network

```

# Configure network
uci set network.globals.ula_prefix="$(uci get network.globals.ula_prefix | sed -e "s/^./d/")"
uci commit network
/etc/init.d/network restart
```

xillibit, 2021-02-21:
I have applied the commands, after an `ipconfig /renew` i'am still with fd something 👍 

[image]

But now ipv6 is working : 

[image]

closed 2021-02-21 by xillibit

xillibit, 2021-02-21:
I have time to time the IPv4 offline but i don't know why

Adorfer, 2021-02-21:
can you a) please update the the issue name ("not working in wan" sounds misleading, since the OMR will not use ipv6 on any wan)
b) if you have ipv4 issues: please open a new issue, describing the problem, using the template.