Ysurac/openmptcprouter #1150: OMR-Bypass questions

benichal, 2020-08-16:
Hey there, first of all, thanks for the solution.

I don't really know if it's correct to open an issue just for asking questions, let me know if i'm wrong

I'd like to have some precisions about the OMR-Bypass.
I put rules in the protocols & services part for netflix as I saw on several posts and it works well.
I also saw that the aggregation is not possible with this trick so i putted my biggest wan connexion as interface.

My question is, if i put 2 rules on this part what is the expected behavior ? 
- 1 netflix protocol with my wan1 as interface
- 1 netflix protocol with my wan2 as interface

(You have to know that my wan1 interface is 4g, so sometimes it can be really slow.)

Does both interfaces will balance depending off the bandwidth ?
Does the interface defined as main will be chosen before the other ?
If it a good way to be sure the service will be always up ?

Thanks for your answersn and have a great day.

OpenMPTCProuter version: v0.55 r0+14022-18b7d87a8f
OpenMPTCProuter VPS version: debian 10
OpenMPTCProuter platform: RPI4b

Ysurac, 2020-08-17:
First rule will be used, so no load balancing possible here.
If you set no interface, this use interface defined as multipath master and if the interface is down this will use another interface.

benichal, 2020-08-17:
Thanks a lot.

closed 2020-08-17 by benichal