Ysurac/openmptcprouter #1151: FTP Port forwarding not working

Corrado56, 2020-08-16:
## Expected Behavior
I Expect when I connect VPS Public IP port 21, to redirect the call to a FTP server in the Lan side of the router 

## Current Behavior
After setting the port forwarding rule, something happen and a connection is made to the FTP server.
But the FTP server reports as the source of the connection the Wan master connection IP, instead of the VPS public ip.
And the connection then closes.

## Possible Solution
No idea.
No idea if this problem is related to port 21 and FTP, or is general from Port forwarding

## Steps to Reproduce the Problem
This is the configuration I have:
[image]

  1. set the following port forward rule
[image]

  2. Connect FTP plain text. VPS public IP, port 21
[image]

  3. The FTP server reports Master Wan IP as the source, instead of the expected VPS IP
[image]
same result testing from FTP client
[image]

## Context (Environment)
I need to redirect connections from VPS Public IP to Lan computers / Printers

## Specifications

  - OpenMPTCProuter version:  0.55
  - OpenMPTCProuter VPS version: <0.1017
  - OpenMPTCProuter platform:  Linksys 3200

Ysurac, 2020-08-17:
Did you try to add a SNAT rule ? Like this: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

Corrado56, 2020-08-17:
Added this NAT rule
[image]

No changes
[image]

Corrado56, 2020-08-17:
Added this NAT rule
[image]

Nothing changes
[image]

Ysurac, 2020-08-17:
Strange. I don't know how it can get WAN IP, redirected traffic use OpenMPTCProuter VPN only.

Corrado56, 2020-08-18:
Hi Yannick, I make more test and i suspect we have a problem with the outgoing traffic, not the incoming.
This is the full picture.
[image]
I am working on MY PC where a FTP client is installed.
When I contact a remote FTP server 188.14.xxx.xxx my PC present itself with the VPS ip address 195.231.xxx.xxx (correct)
[image]
When I contact FTP server 195.231.xxx.xxx (the VPS address) my PC present itself with the master WAN address (wrong), and i receive some connections error.
[image]
**BUT**
If I contact FTP server 195.231.xxx.xxx (the VPS address) from a remote FTP client, I have no errors and could connect the nas1 without any problem, so incoming traffic is correctly forwarded to the NAS1.
So it seems that something is wrong when the FTP client is under the bonding. At this point I am not sure the outbound traffic ever reached the VPS IP, but I am not technical enough to evaluate this Hypothesis.

Corrado56, 2020-08-26:
Hi Yannick, any news on this problem ? 
My FTP forwarding is actually stable and works well **>>  when the call arrives from internet <<**
But
I suppose the red box above should be treated ad a "black" box so any call from 192.168.100.xxx should exit from the VPS IP, then eventually reenter. I doesn't understand why the FTP server receive the master connection IP in this case.

Ysurac, 2020-08-26:
VPS IP is bypassed, so when you want to reach VPS IP you use direct master WAN, this doesn't use the proxy/VPN.
I think a SNAT is needed on VPS side ;)

closed 2020-09-11 by Corrado56