Ysurac/openmptcprouter #1153: UDP data flow across more than one interface

pbapba512, 2020-08-17:
## Expected Behavior

Hello, I'm tested the OpenMPTCP Router for a few days. I do a lot of livestreaming, because of this, I tested with RTMP and RTMPS protocols (which are TCP based). The result was really good, I used 4 pcs 4G modem, and the outgoing data flow was divided across the data connections, it was perfect and stable - I'm really happy with the result!

However, I often use SRT protocol (from Haivision) for point-to-point video connections. The main benefit of the SRT protocol: its open source, robust, packet loss recovery, and jitter prevention. Its an UDP based protocol. 
Unfortunatelly during my tests the SRT data flow was not distrubted across the data connections, only the master connection was used for the transmission. I'm tried to play with the settings, VPN-s, etc, but I can't figure out the solution. I know, UDP is a different animal when we talk about a real bonding-aggregating.

I'm curious, is there any best-practice, suggested settings-configuration for UDP livestreaming data flow?

Thank you!

## Current Behavior

The UDP SRT protocol based outgoing livestreaming dataflow is not distributed across the data connections, only the master connection used for the outgiong dataflow.

## Possible Solution

Maybe any special config or settings?

## Steps to Reproduce the Problem

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:  v0.55 
  - OpenMPTCProuter VPS version: 0.1017 5.4.52
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-08-18:
With every VPN, UDP should use aggregation. Glorytun UDP use own method of aggregation, not based on MPTCP.
I tested with iperf3 and UDP on glorytun TCP and this is aggregated. How do you check if it's aggregated or not ? All is green in OpenMPTCProuter status page ? VPN is up ?

pbapba512, 2020-08-20:
I did some test today. The result is the same. Okay, I'm not a network expert, so probably there is some wrong configuration settings or etc. in my side, but I can't figure out the solution. :( 

So, here is my status screen, as you see everything is green:

[image]

Here is my test with RTMP livestreaming to Vimeo platform with 20Mbit/s, as you can see the traffic is distributed nicely:

[image]

The second test with 5Mbit to Vimeo, the result is not as good as the previous test, mainly was used the master interface for the traffic:

[image]

And here is my last test with SRT UDP streaming protocol at 3.5Mbit, the traffic was not distributed across the interfaces. The result is the same with 10 or 20Mbit. 

[image]

So I tried a lot of things, settings, without any sucess. I don't know its a protocol related problem, or something not good with my configuration... I have no idea... is there any best practice or suggested configuration for this?

Thank you for your time and support!
Peter

Ysurac, 2020-08-20:
The result for 3.5Mb/s and 5Mb/s is normal. Interface with lowest latency is used while bandwidth is not full.
I would like to see the result with 20Mb/s, I think lot of packets are lost ?