Ysurac/openmptcprouter #1154: OMR-Bypass should switch wan connections if primary fails

jrvanator, 2020-08-17:
This is more of a request for enhancement than an issue.

If are using OMR-Bypass (in my case, bypass by source MAC), I leave the interface unspecified, which defaults to the master.  If the master connection fails, ideally, the bypass would switch to the remaining connection(s).

Is this possible?

Thank you, your software is awesome.

Rwestrem, 2020-08-18:
It would be great if it also had the option to load balance between the wans if it was bypassed.

Ysurac, 2020-08-18:
@jrvanator It should have been the case, but it seems that I forget to add changes needed since balancing mode is the default. I will fix that in next release.
Not sure for the load balance when it's bypassed... maybe...

jrvanator, 2020-08-18:
@Ysurac, I believe that @Rwestrem is asking for the option to have the connections do a round-robin usage of the available wan connections when in bypass, not a true load-based balancing.

Ysurac, 2020-08-18:
It's what I was talking about too. When no VPS is defined, OpenMPTCProuter do a load balancing using round robin algorithm pondered since 0.55 by default (it's balancing mode in OpenMPTCProuter advanced parameters).
But using omr-bypass is often done on site banning IPs, not sure if load balancing work very well on them. 
I can add it but if I want to add it as additional choice (like interfaces, master and balancing), this need some code :)

closed 2020-12-21 by github-actions[bot]