Ysurac/openmptcprouter #1155: More then one default VPN is enabled

alexkomsp, 2020-08-18:
## Expected Behavior
VPN is worked.

## Current Behavior
**More then one default VPN is enabled**, but only Glorytun is enabled.

## Specifications

  - OpenMPTCProuter version: v0.55.1
  - OpenMPTCProuter VPS version: 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform:

Ysurac, 2020-08-18:
Check in VPN menu that there is no other VPN enabled, and use only wizard to set a VPN. You can validate again wizard to fix the issue in most case. If it's an update, this can be related to the previous use of MLVPN, disabled now because of crash.

alexkomsp, 2020-08-18:
@Ysurac only Glorytun was enabled, now it's changed to VPN tunnel DOWN.

Ysurac, 2020-08-18:
Can you give me the configuration, available in System->OpenMPTCProuter and "Show all settings" tab. Remove server ip and/or keys before posting.

alexkomsp, 2020-08-18:
dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.LWAN=dhcp
dhcp.LWAN.start='100'
dhcp.LWAN.leasetime='12h'
dhcp.LWAN.limit='150'
dhcp.LWAN.interface='LWAN'
dhcp.wifi=dhcp
dhcp.wifi.start='100'
dhcp.wifi.leasetime='12h'
dhcp.wifi.limit='150'
dhcp.wifi.interface='wifi'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
[... 1523 lines cut ...]
wireless.wifinet1.mode='ap'
wireless.wifinet1.key='****.'
wireless.wifinet1.network='lan wwan'
wireless.wifinet2=wifi-iface
wireless.wifinet2.ssid='***'
wireless.wifinet2.device='radio0'
wireless.wifinet2.mode='sta'
wireless.wifinet2.key='***'
wireless.wifinet2.network='wwan'
wireless.wifinet2.encryption='psk2'
wireless.wifinet3=wifi-iface
wireless.wifinet3.ssid='***'
wireless.wifinet3.device='radio0'
wireless.wifinet3.mode='sta'
wireless.wifinet3.network='wwan'
wireless.wifinet3.encryption='psk2'
wireless.wifinet3.key='***'
wireless.wifinet3.disabled='1'
wireless.wifinet4=wifi-iface
wireless.wifinet4.ssid='***'
wireless.wifinet4.device='radio1'
wireless.wifinet4.mode='sta'
wireless.wifinet4.key='***'
wireless.wifinet4.encryption='psk2'
wireless.wifinet4.network='wifi'

Ysurac, 2020-08-18:
You still have "More than one default VPN is enabled" ?
If not, look at State->System log to check what error about glorytun you can have. You can also try to reboot the router.

alexkomsp, 2020-08-18:
"More than one default VPN is enabled" is gone. I rebooted the router and now there is only "VPN tunnel DOWN".
And what i found in sys logs about glorytun:

Tue Aug 18 12:36:00 2020 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/glorytun-udp
Tue Aug 18 12:36:00 2020 user.notice firewall: Reloading firewall due to ifup of wan1 (br-wan1)
Tue Aug 18 12:36:00 2020 user.notice ucitrack: Setting up /etc/config/glorytun reload dependency on /etc/config/openmptcprouter-vps
Tue Aug 18 12:36:00 2020 daemon.info dnsmasq[1993]: exiting on receipt of SIGTERM

Tue Aug 18 12:36:01 2020 daemon.info glorytun: starting glorytun vpn instance vpn
Tue Aug 18 12:36:01 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Aug 18 12:36:01 2020 user.notice ModemManager: hotplug: cached event found: action=add, name=tty24, subsystem=tty, sysfspath=/sys/devices/virtual/tty/tty24
Tue Aug 18 12:36:01 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tty24, subsystem=tty
Tue Aug 18 12:36:01 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Tue Aug 18 12:36:01 2020 daemon.info glorytun[6274]: INITIALIZED tun0

Tue Aug 18 12:36:05 2020 daemon.err glorytun[6274]: 185.***.***.***.65001: connected
Tue Aug 18 12:36:05 2020 daemon.info glorytun[6274]: STARTED tun0

Ysurac, 2020-08-18:
What can be interesting is after the STARTED.

alexkomsp, 2020-08-18:
```
Tue Aug 18 12:36:05 2020 daemon.err glorytun[6274]: 185.***.***.***.65001: connected
Tue Aug 18 12:36:05 2020 daemon.info glorytun[6274]: STARTED tun0
Tue Aug 18 12:36:05 2020 user.err vnstat_backup: cannot restore, backup file does not exist (yet)
Tue Aug 18 12:36:05 2020 user.notice adb-enablemodem: INFO: waiting for device
Tue Aug 18 12:36:05 2020 user.notice omr-bypass: Starting OMR-ByPass...
Tue Aug 18 12:36:06 2020 daemon.info [4550]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.1': not supported by any plugin
Tue Aug 18 12:36:06 2020 daemon.info [4550]: <info>  [base-manager] couldn't check support for device '/sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3': not supported by any plugin
Tue Aug 18 12:36:06 2020 user.notice omr-bypass: Restart dnsmasq...
Tue Aug 18 12:36:06 2020 daemon.info dnsmasq[5346]: exiting on receipt of SIGTERM
Tue Aug 18 12:36:06 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Aug 18 12:36:06 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Aug 18 12:36:06 2020 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: started, version 2.82 cachesize 150
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: DNS service limited to local subnets
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq-dhcp[11393]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: using only locally-known addresses for domain test
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: using only locally-known addresses for domain onion
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: using only locally-known addresses for domain localhost
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: using only locally-known addresses for domain local
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: using only locally-known addresses for domain invalid
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: using only locally-known addresses for domain bind
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: using only locally-known addresses for domain use-application-dns.net
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: using nameserver 127.0.0.1#5353
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: using only locally-known addresses for domain lan
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: read /etc/hosts - 4 addresses
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq[11393]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Tue Aug 18 12:36:07 2020 user.notice omr-bypass: OMR-ByPass is running
Tue Aug 18 12:36:07 2020 daemon.info dnsmasq-dhcp[11393]: read /etc/ethers - 0 addresses
Tue Aug 18 12:36:08 2020 user.notice OMR-VPS: Can't get token, try later
Tue Aug 18 12:36:08 2020 kern.info kernel: [   20.938664] br-wan1: port 1(eth1) entered blocking state
Tue Aug 18 12:36:08 2020 kern.info kernel: [   20.944025] br-wan1: port 1(eth1) entered forwarding state
Tue Aug 18 12:36:08 2020 daemon.notice netifd: Network device 'eth1' link is up
Tue Aug 18 12:36:08 2020 daemon.notice netifd: bridge 'br-wan1' link is up
Tue Aug 18 12:36:08 2020 daemon.notice netifd: Interface 'wan1' has link connectivity
Tue Aug 18 12:36:08 2020 user.notice unbound: default memory configuration
Tue Aug 18 12:36:08 2020 daemon.notice unbound: [11844:0] notice: init module 0: iterator
Tue Aug 18 12:36:08 2020 daemon.info unbound: [11844:0] info: start of service (unbound 1.11.0).
Tue Aug 18 12:36:10 2020 user.notice post-tracking-post-tracking: Set server vps (185.***.***.***) route via 192.168.1.1 metric 7
Tue Aug 18 12:36:10 2020 user.notice post-tracking-post-tracking: Can't find Shadowsocks, restart it...
Tue Aug 18 12:36:10 2020 user.notice Shadowsocks: Rules DOWN
Tue Aug 18 12:36:11 2020 daemon.err omr-tracker[9500]: Command failed: Not found
Tue Aug 18 12:36:11 2020 user.notice Shadowsocks: Rules UP
Tue Aug 18 12:36:11 2020 user.notice Shadowsocks: Reload omr-bypass rules
Tue Aug 18 12:36:11 2020 user.notice omr-bypass: Starting OMR-ByPass...
Tue Aug 18 12:36:11 2020 daemon.notice procd: /etc/rc.d/S99openmptcprouter-vps: {"result":"done","reason":"no changes","route":"shadowsocks"}
Tue Aug 18 12:36:12 2020 user.notice post-tracking-post-tracking: wan1 (br-wan1) switched off because gateway down and ping from 192.168.8.2 error (4.2.2.1,8.8.8.8)
Tue Aug 18 12:36:12 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.8.1 dev br-wan1
Tue Aug 18 12:36:12 2020 user.notice omr-bypass: Reload dnsmasq...
Tue Aug 18 12:36:12 2020 daemon.info dnsmasq[11393]: read /etc/hosts - 4 addresses
Tue Aug 18 12:36:12 2020 daemon.info dnsmasq[11393]: read /tmp/hosts/dhcp.cfg01411c - 1 addresses
Tue Aug 18 12:36:12 2020 daemon.info dnsmasq-dhcp[11393]: read /etc/ethers - 0 addresses
Tue Aug 18 12:36:12 2020 user.notice omr-bypass: OMR-ByPass is running
Tue Aug 18 12:36:13 2020 user.info : using tcp fast open
Tue Aug 18 12:36:13 2020 user.info : using tcp fast open
Tue Aug 18 12:36:13 2020 user.info : using tcp fast open
[... cut ...]
Tue Aug 18 12:36:38 2020 daemon.notice unbound: [14165:0] notice: init module 0: iterator
Tue Aug 18 12:36:38 2020 daemon.info unbound: [14165:0] info: start of service (unbound 1.11.0).
Tue Aug 18 12:36:41 2020 user.notice post-tracking-post-tracking: wan1 (br-wan1) switched up
Tue Aug 18 12:36:41 2020 user.notice post-tracking-post-tracking: Set server vps (185.***.***.***) route via 192.168.8.1 metric 3
Tue Aug 18 12:36:48 2020 daemon.err uhttpd[3917]: luci: accepted login on / for root from 192.168.100.142
```

Ysurac, 2020-08-18:
wan1 is a bridge, and this seems to be a problem as it's switched up/down. Should be a macvlan.

alexkomsp, 2020-08-18:
I changed my bridge to macvlan, now I got "Gateway DOWN" on wan1.

Ysurac, 2020-08-18:
what is the result of ```ip a``` and ```ip r``` via SSH on the router (and without server ip) ?

alexkomsp, 2020-08-19:
```
root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,ALLMULTI,PROMISC,UP,LOWER_UP,80000> mtu 1500 qdisc mq master br-lan state UP group default qlen 1000
    link/ether dc:a6:32:ae:4b:79 brd ff:ff:ff:ff:ff:ff
3: ip6tnl0@NONE: <NOARP,80000> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd ::
4: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
5: gre0@NONE: <NOARP,80000> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
6: gretap0@NONE: <BROADCAST,MULTICAST,80000> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: erspan0@NONE: <BROADCAST,MULTICAST,80000> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
8: ip6gre0@NONE: <NOARP,80000> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd ::
9: teql0: <NOARP,80000> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void
10: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 100
    link/ether dc:a6:32:ae:4b:7a brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.2/24 brd 192.168.1.255 scope global wlan0
       valid_lft forever preferred_lft forever
14: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 qdisc fq_codel state UP group default qlen 100
    link/ether 0c:5b:8f:27:9a:64 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::e5b:8fff:fe27:9a64/64 scope link
       valid_lft forever preferred_lft forever
18: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP,80000> mtu 1500 qdisc noqueue state UP group default qlen 1000
    link/ether dc:a6:32:ae:4b:79 brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global br-lan
       valid_lft forever preferred_lft forever
19: wlan1: <NO-CARRIER,BROADCAST,MULTICAST,UP,80000> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether 84:c9:b2:82:dd:fa brd ff:ff:ff:ff:ff:ff
20: wlan1-1: <NO-CARRIER,BROADCAST,MULTICAST,ALLMULTI,PROMISC,UP,80000> mtu 1500 qdisc mq master br-lan state DOWN group default qlen 1000
    link/ether 84:c9:b2:82:dd:fb brd ff:ff:ff:ff:ff:ff
21: tun0: <POINTOPOINT,MULTICAST,NOARP,80000> mtu 1500 qdisc fq_codel state DOWN group default qlen 500
    link/none
23: br-omrvpn: <BROADCAST,MULTICAST,80000> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 66:fa:d3:58:ab:2b brd ff:ff:ff:ff:ff:ff
24: wan1@eth1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 0c:5b:8f:27:9a:64 brd ff:ff:ff:ff:ff:ff
    inet 192.168.8.2/24 brd 192.168.8.255 scope global wan1
       valid_lft forever preferred_lft forever

```

```
root@OpenMPTCProuter:~# ip r
default via 192.168.1.1 dev wlan0
default via 192.168.1.1 dev wlan0 metric 7
127.0.0.0/8 dev lo proto static scope link metric 1
185.*.*.* via 192.168.1.1 dev wlan0 metric 1
185.*.*.* via 192.168.1.1 dev wlan0 metric 7
192.168.1.0/24 dev wlan0 scope link metric 7
```

alexkomsp, 2020-08-20:
@Ysurac no idea what is the problem?

Ysurac, 2020-08-20:
You have many errors in your configuration: why omrvpn is a bridge ? this should not be the case. Why lan is also a bridge ?
I would try a fresh configuration.

alexkomsp, 2020-08-20:
Don't remember what i switched lan or something else to bridge, but now i installed fresh image on my rpi and all is good, vpn is work! Thanks so much for your help @Ysurac !

Ysurac, 2020-08-20:
cool :)
Thank you for the feedback

closed 2020-08-20 by Ysurac