Ysurac/openmptcprouter #1156: issue running VPS script.

Rwestrem, 2020-08-18:
Get:1 https://repo.openmptcprouter.com stretch InRelease [2119 B]
Err:1 https://repo.openmptcprouter.com stretch InRelease
  The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 608FD17B2B24D936
Reading package lists... Done
W: GPG error: https://repo.openmptcprouter.com stretch InRelease: The following signatures couldn't be verified because the public key is not available: NO_PUBKEY 608FD17B2B24D936
E: The repository 'https://repo.openmptcprouter.com stretch InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

Ysurac, 2020-08-19:
Try again, if still in error give me more log. The key is downloaded before that.

Rwestrem, 2020-08-19:
I was missing repositories.  I'll close this.  Thank you.

closed 2020-08-19 by Rwestrem