Ysurac/openmptcprouter #1158: Problem of perf

vinceducat, 2020-08-19:
## Expected Behavior
have aggregation flow rate egal to sum of each flow  275 mb/s
[https://pic.nperf.com/r/3272421640973452-gjN3eSVk.png](url)
## Current Behavior
rate only half of the sum 77 mb/s
[https://pic.nperf.com/r/3272422147777817-zJoPjkvV.png](url)
## Possible Solution
want to have opinion about perf

## Steps to Reproduce the Problem

## Context (Environment)

  - OpenMPTCProuter version: 0.55.1
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform:

Ysurac, 2020-08-19:
How many connections do you have ? what is the speed of each connection ? Did you test with ```omr-test-speed``` and ```omr-test-speed wan1``` via SSH on the router ?
And what is the platform ? What is the VPS ?

vinceducat, 2020-08-19:
x86_64 for platform

I have 2 connection 
_ adsl wan 1 direct speed without pass  trought vps 1000k/s
_ 4g wan 2 direct test without pass  trought vps 350mb/s

test trought vps ans omr-test-speed
omr-test-speed  10M
omr-test-speed wan 2 1800k => strange :-(

Ysurac, 2020-08-19:
So you seems to have a problem with wan2. omr-test-speed only download a file from OVH, directly when an interface is specified.
It's a dedicated ethernet card with an ethernet router or an USB LTE dongle ?

vinceducat, 2020-08-19:
its dedicated ethernet card with internet router huawei b825 in ethernet

Ysurac, 2020-08-20:
Check cables, try to plug it on wan1,...
Check kernel and system log of OpenMPTCProuter in state menu, maybe a problem with this connection is detected.

vinceducat, 2020-08-20:
its not a problem of cable.. all test are made trought the same hardware, the only difference is passing throught or not the vps... but allways test pass trought OMR ...

it seems than my VPS is 100% load on the two vCPU in the milkywan perf1 vps... i ever change from bidouille to perf1 ... no more best perff :-(

Ysurac, 2020-08-20:
With ```omr-test-speed wan2``` this doesn't use the VPS at all.
For the CPU load, check that AES is disabled on advanced checkbox in the OpenMPTCProuter wizard, milkywan VPS doesn't seems to support AES.
You can also disable encryption for testing at least.
And with htop (```apt-get install htop```) on the VPS you can check CPU load and what is using it.

vinceducat, 2020-08-20:
i have chacha20 in the in the encryption.... must i disable it ?

Ysurac, 2020-08-20:
To test if it's encryption that use all the CPU and so make everything slow: yes

vinceducat, 2020-08-20:
does i have to reboot beetween each change ?

Ysurac, 2020-08-20:
No, only validate the wizard.

Ysurac, 2020-08-20:
So about 8Mb/s and 27Mb/s when encryption is disabled. Still not really good. And still high CPU usage.
Did you try with htop to check what is using so much CPU ?
What is the load on the VPS when it is idle ?

vinceducat, 2020-08-20:
idle 2/5% ...

during omr-test-speed on the vps the first process loading cpu is ss-manager ... shadowsocks...

vinceducat, 2020-08-20:
yes its ss-server...

Hugues from milkywan just now give me a vps which support aes-ni what was the best encryption type. ?

resultat of rate are really better :-)

Ysurac, 2020-08-20:
Yes, I asked to him why AES was not activated and he moved your VPS on a host with AES.
Check in the wizard that AES-256-GCM is really used.
Rate shouldn't improve as encryption was disabled, but host CPU can be better too :)

vinceducat, 2020-08-20:
thanks ;-)

aes-256-gcm enable in the wizard ;-)

vinceducat, 2020-08-24:
@Ysurac  no idea about this difference ?

Ysurac, 2020-08-24:
Difficult to compare vCPU, some can have more dedicated ressources...

rifaterdemsahin, 2020-09-30:
I gave up the VGPUs last month and i will go for i5 with enough ram and nvme

Rationale : routers should be easy reset start stop...

my planned purchase
https://www.amazon.co.uk/gp/product/B076B6SWG5/ref=as_li_qf_asin_il_tl?ie=UTF8&tag=rifaterdemsah-21&creative=6738&linkCode=as2&creativeASIN=B076B6SWG5&linkId=7eb4dbb17a7cbf769408adc483f40887

closed 2021-01-04 by github-actions[bot]