Ysurac/openmptcprouter #1159: unable to update port forward

zulu14, 2020-08-19:
hello, this issues is for all openmptcprouter version and vps tested, 
hardware rpi4

the port worward into the VPN to LAN now working alwas, sometimes its works , sometimes its not works, restart the vps or omr not solve the problem, the only things  into the syslog is "orm-vps: Remove old firewall rules"

zulu14, 2020-08-19:
(all of the omr statuts are green)

zulu14, 2020-08-19:
(the direct all port of vps to omr are activated)
(the disable of the port already into the firewall are desactivated)

thank you for your helping and aweseome software

zulu14, 2020-08-19:
update: (sorry for the multiple post)
on bot the vps screen show " failed to start shorewall ipv4 firewall"

zulu14, 2020-08-19:
[image]

zulu14, 2020-08-19:
[image]

zulu14, 2020-08-19:
i fixed the issues, when the ICMP protocol are select the shorewall not start (screen above) we need to unselect icmp on port forward to enable the shorewall

Ysurac, 2020-08-20:
I will fix this for next release.

closed 2020-08-20 by zulu14