Ysurac/openmptcprouter #1160: opkg failure for  http://packages.openmptcprouter.com/v0.55rc2/x86_64

gnosoz, 2020-08-20:
## Expected Behavior
opkg update / opkg install package up to vers 54.7 was working fine but now it seems there is a failure downloading the repo in  http://packages.openmptcprouter.com/v0.55rc2/x86_64
this is preventing installation of additional packages as the update fails with error 4

## Current Behavior
```
opkg update
Downloading http://packages.openmptcprouter.com/v0.55rc2/x86_64/luci/Packages.gz
Failed to send request: Operation not permitted
*** Failed to download the package list from http://packages.openmptcprouter.com/v0.55rc2/x86_64/luci/Packages.gz

Downloading http://packages.openmptcprouter.com/v0.55rc2/x86_64/packages/Packages.gz
Failed to send request: Operation not permitted
*** Failed to download the package list from http://packages.openmptcprouter.com/v0.55rc2/x86_64/packages/Packages.gz

Downloading http://packages.openmptcprouter.com/v0.55rc2/x86_64/base/Packages.gz
Failed to send request: Operation not permitted
*** Failed to download the package list from http://packages.openmptcprouter.com/v0.55rc2/x86_64/base/Packages.gz

Downloading http://packages.openmptcprouter.com/v0.55rc2/x86_64/routing/Packages.gz
Failed to send request: Operation not permitted
*** Failed to download the package list from http://packages.openmptcprouter.com/v0.55rc2/x86_64/routing/Packages.gz

Downloading http://packages.openmptcprouter.com/v0.55rc2/x86_64/telephony/Packages.gz
Failed to send request: Operation not permitted
*** Failed to download the package list from http://packages.openmptcprouter.com/v0.55rc2/x86_64/telephony/Packages.gz

Downloading http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/targets/x86/64/packages/Packages.gz
Failed to send request: Operation not permitted
*** Failed to download the package list from http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/targets/x86/64/packages/Packages.gz

Downloading http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/packages/x86_64/base/Packages.gz
Failed to send request: Operation not permitted
*** Failed to download the package list from http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/packages/x86_64/base/Packages.gz

Downloading http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/packages/x86_64/luci/Packages.gz
Failed to send request: Operation not permitted
*** Failed to download the package list from http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/packages/x86_64/luci/Packages.gz

Downloading http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/packages/x86_64/openmptcprouter/Packages.gz
Failed to send request: Operation not permitted
*** Failed to download the package list from http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/packages/x86_64/openmptcprouter/Packages.gz

Downloading http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/packages/x86_64/packages/Packages.gz
Failed to send request: Operation not permitted
*** Failed to download the package list from http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/packages/x86_64/packages/Packages.gz

Collected errors:
 * opkg_download: Failed to download http://packages.openmptcprouter.com/v0.55rc2/x86_64/luci/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://packages.openmptcprouter.com/v0.55rc2/x86_64/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://packages.openmptcprouter.com/v0.55rc2/x86_64/base/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://packages.openmptcprouter.com/v0.55rc2/x86_64/routing/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://packages.openmptcprouter.com/v0.55rc2/x86_64/telephony/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/targets/x86/64/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/packages/x86_64/base/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/packages/x86_64/luci/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/packages/x86_64/openmptcprouter/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

 * opkg_download: Failed to download http://download.openmptcprouter.com:80/release/v0.55rc2/x86_64/packages/x86_64/packages/Packages.gz, wget returned 4.
 * opkg_download: Check your network settings and connectivity.

```

## Possible Solution
Please check that the repository is accessible, I believe something similar happened on a previous release and was sorted by the magic Ysurac very quickly

## Steps to Reproduce the Problem

  1. install v0.55.1 r0+14022-18b7d87a8f
  2. opkg update
  3. opkg install "name_package"

## Context (Environment)
opkg update / opkg install package up to vers 54.7 was working fine but now it seems there is a failure downloading the repo in  http://packages.openmptcprouter.com/v0.55rc2/x86_64
this is preventing installation of additional packages as the update fails with error 4

## Specifications

  - OpenMPTCProuter version: v0.55.1 r0+14022-18b7d87a8f
  - OpenMPTCProuter VPS version: < OpenMPTCProuter VPS 0.1015 >
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-08-20:
Do you have OpenMPTCProuter configured and working ?
Can you test, from the router via SSH: ```wget -4 http://packages.openmptcprouter.com/v0.55rc2/x86_64/luci/Packages.gz``` ?
If it's working, can you also test: ```opkg update``` ?

gnosoz, 2020-08-20:
Hi Yannick,
forcing ipv4 only works like a charm but then opkg update I think is still trying with ipv6 hence the error I am getting

on regards with router setting in luci I have the ipv6 disabled with no prefix set either, not sure if that might be causing this?

Ysurac, 2020-08-20:
I'm not able to reproduce the issue, even if only IPv4 is enabled. If it's an update, this can be related to an old setting.