Ysurac/openmptcprouter #1161: omr-test-speed error

vistman, 2020-08-20:
## Expected Behavior
I guess omr-test-speed wan1/2/3 should work

## Current Behavior
omr-test-speed is working, but omr-test-speed wan1/2/3 returns curl: (45) Couldn't bind to 'wan1/2/3'

## Possible Solution
Don't know.

## Steps to Reproduce the Problem

  1. SSH to OMR
  2. omr-test-speed wan 1
  3. omr-test-speed wan 1
  4. omr-test-speed wan 3

## Context (Environment)
I have 3 routers attached to OMR via USB2Eth adapters (wan1 - vDSL, wan2 - LTE, wan3 - LTE)
I was checking out https://github.com/Ysurac/openmptcprouter/issues/1158 and decided to try it as I had some funny side effects after upgrading to v0.55.1 where one of the routers was constantly reported as unreachable although it was. Therefore I downgraded to rc1 which was last working for me as I haven't tried rc2.
 
Also don't know if it is related to this issue, but not to open a new issue - I've noticed another side effect with my 3 routers where the 3rd one seemed not to be used at all if it wasn't explicitly set as master. More precisely I've just changed a LTE provider on 3rd router (wan3) which has by far greatest transfer rate and until it isn't set as master my overall speed are 1/3 of those when I set it as master.

## Specifications

  - OpenMPTCProuter version: 0.55rc1
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: RPi4b/x86_64)

Ysurac, 2020-08-20:
```omr-test-speed wan1``` should work but with wan1 must be a real interface visible via ```ip link```.
USB2Ethernet adapter often doesn't give a good result.
For the unreachable, I think it's about gateway ? Did you try to ping it via SSH on the router ?

vistman, 2020-08-20:
USB2Eth adapters are actually showing USB2Eth adapters ...
As for the gateway ... you're right I cannot ping my gateway from OMR ... will have to take a look at set firewall rules ... Thanks, this hasn't occurred this could be a problem ...

vistman, 2020-08-20:
If we're talking about ISP routers/modems then they are all reachable without any problems from OMR.

vistman, 2020-08-29:
How can I know if my provider is blocking MPTCP?
I doubt this is happening as as soon as I've set one of the links/interfaces to use MPTCP over VPN

Anyway this is the result from my interface not using MPTCP over VPN (anonymized):
1: <WAN2_GATEWAY_IP>
2: * 
3: 212.151.136.17 IP::TTL IP::CheckSum 
4: 130.244.64.171  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
5: 212.151.234.216 IP::TTL IP::CheckSum 
6: 130.244.39.90 IP::TTL IP::CheckSum 
7: 212.151.233.140 IP::TTL IP::CheckSum 
8: 130.244.39.206 IP::TTL IP::CheckSum 
9: 62.115.175.116 IP::TTL IP::CheckSum 
10: 62.115.14.117  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
11: 144.208.208.145 IP::TTL IP::CheckSum [Extra headers: ICMPExtension ICMPExtensionObject ICMPExtensionMPLS ] 
12: 144.208.208.151  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
13: 144.208.211.19  [PARTIAL] +PartialTCP IP::TTL IP::CheckSum 
14: x.x.x.x TCP::SrcPort TCP::DstPort TCP::SeqNumber TCP::AckNumber TCP::DataOffset TCP::Flags TCP::WindowsSize TCP::CheckSum IP::TotalLength IP::Identification IP::TTL IP::CheckSum IP::SourceIP IP::DestinationIP +TCPOptionMaxSegSize TCPOptionMPTCPCapable::Sender's Key 

Result:
< Ethernet (14 bytes) :: DestinationMAC = <DEST_MAC> , SourceMAC = <SRC_MAC> , Type = 0x800 , >
< IP (20 bytes) :: Version = 4 , HeaderLength = 5 , DiffServicesCP = 0 , ExpCongestionNot = 0 , TotalLength = 56 , Identification = 0x0 , Flags = 2 , FragmentOffset = 0 , TTL = 51 , Protocol = 0x6 , CheckSum = 0x28cc , SourceIP = <VPS_IP> , DestinationIP = <DHCP_IP_BY_WAN2_ROUTER> , >
< TCP (20 bytes) :: SrcPort = 65101 , DstPort = 9122 , SeqNumber = 3185481877 , AckNumber = 2060445656 , DataOffset = 9 , Reserved = 0 , Flags = ( SYN ACK ) , WindowsSize = 43200 , CheckSum = 0x2b29 , UrgPointer = 0 , >
< TCPOptionMaxSegSize (4 bytes) :: Kind = 2 , Length = 4 , MaxSegSize = 1420 , >
< TCPOptionMPTCPCapable (12 bytes) :: Kind = 30 , Length = 12 , Subtype = 0 , Version = 0 , Checksum = 1 (Checksum Enabled) , Flags = 0 , Crypto = 1 (HMAC-SHA1) , Sender's Key = Sender's Key = <SENDER_KEY> , >

What does [PARTIAL] mean?
Also currently I have 2 LTE modem/routers connected to OMR and I can set only one of them to use MPTCP over VPN - if I set both everything is reported to be OK on status page but I don't have Internet access.

Ysurac, 2020-08-29:
There is the "TCPOptionMPTCPCapable" so not problem here, at least for this interface. But there is test done over each interface so when there is a problem a warning is displayed.

vistman, 2020-08-29:
OK, thanks .. this is test from other-slower (not master) connection ...
And could you please tell me if you have any idea why can't I use MPTCP over VPN on both routers at the same time?

Another funny thing is if I use MPTCP over VPN for my faster (master) connection I'm getting bonded (greater) bandwidth rate (wit about 10% greater ping) but if I use faster (master) connection without MPTCP over VPN, OMR is using just one - master connection (I've tested it numerous times until I figured this was happening) - hence my conclusion my ISP on faster connection is blocking MPTCP ...

I'll post test results for my faster (master) connection with MPTCP over VPN disabled later ...

closed 2021-01-02 by github-actions[bot]