Ysurac/openmptcprouter #1172: issues

zulu14, 2020-08-23:
hi, i have multiple instance of OMR, 
one of my instance are the status page all green, but i cannot ping openmptcprouter.com

can you plese make a little check to the syslog, and tell me if you see an issues on it ! thanxs
i know is not the 0.50.2 version, but this istance is berry mportant, and i cannot update it now,

Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain 1.1.1.1
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain use-application-dns.net
Sun Aug 23 20:09:37 2020 daemon.infoSun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain 1.1.1.1
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain use-application-dns.net
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using nameserver 127.0.0.1#5353
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using 1 more local addresses
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: read /etc/hosts - 4 addresses
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq-dhcp[21097]: read /etc/ethers - 0 addresses
Sun Aug 23 20:09:37 2020 daemon.err dnsmasq[21097]: cannot read server=1.0.0.1: No such file or directory
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain test
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain onion
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain localhost
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain local
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain invalid
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain bind
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain 1.1.1.1
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain use-application-dns.net
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using nameserver 127.0.0.1#5353
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using 1 more local addresses
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: read /etc/hosts - 4 addresses
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq-dhcp[21097]: read /etc/ethers - 0 addresses
Sun Aug 23 20:09:37 2020 daemon.err dnsmasq[21097]: cannot read server=1.0.0.1: No such file or directory
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain test
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain onion
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain localhost
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain local
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain invalid
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain bind
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain 1.1.1.1
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain use-application-dns.net
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using nameserver 127.0.0.1#5353
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using 1 more local addresses
Sun Aug 23 20:09:37 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Aug 23 20:09:37 2020 user.notice Shadowsocks: Rules DOWN
Sun Aug 23 20:09:37 2020 user.notice mptcp: Reloading mptcp config due to ifup of wan2 (eth1)
Sun Aug 23 20:09:38 2020 user.notice Shadowsocks: Rules UP
Sun Aug 23 20:09:38 2020 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Sun Aug 23 20:09:39 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Sun Aug 23 20:09:39 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sun Aug 23 20:09:39 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun Aug 23 20:09:40 2020 user.notice omr-bypass: Reload dnsmasq...
Sun Aug 23 20:09:40 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 104.16.1.1)
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: read /etc/hosts - 4 addresses
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq-dhcp[21097]: read /etc/ethers - 0 addresses
Sun Aug 23 20:09:40 2020 daemon.err dnsmasq[21097]: cannot read server=1.0.0.1: No such file or directory
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain test
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain onion
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain localhost
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain local
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain invalid
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain bind
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain 1.1.1.1
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain use-application-dns.net
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using nameserver 127.0.0.1#5353
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using 1 more local addresses
Sun Aug 23 20:09:40 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Aug 23 20:09:40 2020 daemon.info procd: - init complete -
Sun Aug 23 20:09:40 2020 daemon.info vnstatd[23056]: Info: vnStat daemon 2.4 started. (pid:23056 uid:0 gid:0 64-bit)
Sun Aug 23 20:09:40 2020 daemon.info vnstatd[23056]: Info: Monitoring (12): wlan0 (1000 Mbit) wan4 (1000 Mbit) wan3 (1000 Mbit) wan2 (1000 Mbit) wan1 (1000 Mbit) tun0 (10 Mbit) teql0 (1000 Mbit) ifb4tun0 (1000 Mbit) ifb1 (1000 Mbit) ifb0 (1000 Mbit) eth1 (1000 Mbit) eth0 (1000 Mbit)
Sun Aug 23 20:09:40 2020 daemon.info vnstatd[23056]: Info: Interface "wan4" disabled.
Sun Aug 23 20:09:40 2020 daemon.info vnstatd[23056]: Info: Interface "wan3" disabled.
[... 1988 lines cut ...]
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-redir[25852]: listening at 0.0.0.0:1101
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-redir[25852]: tcp port reuse enabled
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-redir[25849]: tcp port reuse enabled
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-redir[25849]: UDP relay enabled
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-redir[25852]: UDP relay enabled
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-redir[25849]: udp port reuse enabled
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-redir[25852]: udp port reuse enabled
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-redir[25852]: running from root user
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-redir[25849]: running from root user
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-local[25848]: using tcp fast open
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-local[25848]: enable TCP no-delay
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-local[25848]: initializing ciphers... chacha20-ietf-poly1305
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-local[25848]: listening at 127.0.0.1:1111
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-local[25848]: tcp port reuse enabled
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-local[25848]: udprelay enabled
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-local[25848]: udp port reuse enabled
Sun Aug 23 20:11:51 2020 daemon.info /usr/bin/ss-local[25848]: running from root user
Sun Aug 23 20:11:51 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sun Aug 23 20:11:55 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.255.1 dev tun0
[... cut ...] .254
Sun Aug 23 20:13:55 2020 user.notice post-tracking-post-tracking: omrvpn down. Replace default route by 192.168.0.254 dev wan1
Sun Aug 23 20:13:55 2020 user.notice post-tracking-post-tracking: omrvpn down because gateway down and ping from 10.255.255.2 error (4.2.2.1,8.8.8.8)
Sun Aug 23 20:13:55 2020 user.notice post-tracking-post-tracking: Tunnel down use ShadowSocks for UDP
Sun Aug 23 20:13:55 2020 user.notice Shadowsocks: Rules DOWN
Sun Aug 23 20:13:56 2020 user.notice Shadowsocks: Rules

Ysurac, 2020-08-23:
There is an error in DHCP&DNS configuration, in DNS forwarding I think and concurrent DNS queries are reached, so this need to be increased in advanced parameters tab.
Check that you can ping an internet IP like 8.8.8.8

zulu14, 2020-08-23:
thank for you quick answer, 
i need to update here ? 
[image]

zulu14, 2020-08-23:
ping a dns not work :/
[image]

zulu14, 2020-08-23:
some time the 192.168.100.1  are not reachable, i have try to reset omr, and its not solve the problem.

the following syslog has get when the interface are reUP, 

Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain 1.1.1.1
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain use-application-dns.net
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using nameserver 127.0.0.1#5353
Sun Aug 23 20:09:37 2020 daemon.info dnsmasq[21097]: using 1 more local addresses
Sun Aug 23 20:09:37 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Aug 23 20:09:37 2020 user.notice Shadowsocks: Rules DOWN
Sun Aug 23 20:09:37 2020 user.notice mptcp: Reloading mptcp config due to ifup of wan2 (eth1)
Sun Aug 23 20:09:38 2020 user.notice Shadowsocks: Rules UP
Sun Aug 23 20:09:38 2020 user.notice post-tracking-post-tracking: Get status and settings for wan1... Done
Sun Aug 23 20:09:39 2020 user.notice post-tracking-post-tracking: Get status and settings for wan2... Done
Sun Aug 23 20:09:39 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sun Aug 23 20:09:39 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sun Aug 23 20:09:40 2020 user.notice omr-bypass: Reload dnsmasq...
Sun Aug 23 20:09:40 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 104.16.1.1)
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: read /etc/hosts - 4 addresses
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: read /tmp/hosts/dhcp.cfg01411c - 4 addresses
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq-dhcp[21097]: read /etc/ethers - 0 addresses
Sun Aug 23 20:09:40 2020 daemon.err dnsmasq[21097]: cannot read server=1.0.0.1: No such file or directory
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain test
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain onion
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain localhost
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain local
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain invalid
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain bind
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain 1.1.1.1
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using only locally-known addresses for domain use-application-dns.net
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using nameserver 127.0.0.1#5353
Sun Aug 23 20:09:40 2020 daemon.info dnsmasq[21097]: using 1 more local addresses
Sun Aug 23 20:09:40 2020 user.notice omr-bypass: OMR-ByPass is running
Sun Aug 23 20:09:40 2020 daemon.info procd: - init complete -
Sun Aug 23 20:09:40 2020 daemon.info vnstatd[23056]: Info: vnStat daemon 2.4 started. (pid:23056 uid:0 gid:0 64-bit)
Sun Aug 23 20:09:40 2020 daemon.info vnstatd[23056]: Info: Monitoring (12): wlan0 (1000 Mbit) wan4 (1000 Mbit) wan3 (1000 Mbit) wan2 (1000 Mbit) wan1 (1000 Mbit) tun0 (10 Mbit) teql0 (1000 Mbit) ifb4tun0 (1000 Mbit) ifb1 (1000 Mbit) ifb0 (1000 Mbit) eth1 (1000 Mbit) eth0 (1000 Mbit)
Sun Aug 23 20:09:40 2020 daemon.info vnstatd[23056]: Info: Interface "wan4" disabled.
Sun Aug 23 20:09:40 2020 daemon.info vnstatd[23056]: Info: Interface "wan3" disabled.
Sun Aug 23 20:09:40 2020 daemon.info vnstatd[23056]: Info: Interface "wan2" disabled.
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info: service stopped (unbound 1.10.0).
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info: server stats for thread 0: 206 queries, 6 answers from cache, 200 recursions, 0 prefetch, 0 rejected by ip ratelimiting
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info: server stats for thread 0: requestlist max 61 avg 28.05 exceeded 0 jostled 0
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info: average recursion processing time 25.807899 sec
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info: histogram of recursion processing times
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info: [25%]=10.3529 median[50%]=23.1724 [75%]=42.4727
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info: lower(secs) upper(secs) recursions
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info:    0.016384    0.032768 1
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info:    0.032768    0.065536 4
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info:    0.065536    0.131072 5
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info:    0.131072    0.262144 9
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info:    0.262144    0.524288 2
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info:    0.524288    1.000000 1
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info:    2.000000    4.000000 5
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info:    4.000000    8.000000 13
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info:    8.000000   16.000000 34
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info:   16.000000   32.000000 58
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info:   32.000000   64.000000 55
Sun Aug 23 20:09:46 2020 daemon.info unbound: [11080:0] info:   64.000000  128.000000 13
[... cut ...]
Sun Aug 23 22:02:24 2020 daemon.info dnsmasq-dhcp[21594]: DHCPACK(eth0) 192.168.100.39 1c:1b:0d:6f:f8:61 MAXIME
Sun Aug 23 22:03:35 2020 daemon.err /usr/bin/ss-redir[25850]: server recv: Host is unreachable
Sun Aug 23 22:04:03 2020 daemon.info dnsmasq-dhcp[21594]: DHCPREQUEST(eth0) 192.168.100.42 fc:3f:db:46:31:c0
Sun Aug 23 22:04:03 2020 daemon.info dnsmasq-dhcp[21594]: DHCPACK(eth0) 192.168.100.42 fc:3f:db:46:31:c0 MAXIME-PC
Sun Aug 23 22:04:48 2020 daemon.err uhttpd[3681]: luci: accepted login on /admin for root from 192.168.100.42

Ysurac, 2020-08-23:
No, in Network->DHCP and DNS menu.

Ysurac, 2020-08-23:
But as ping doesn't work, I think there is a bigger problem...
Can you ping the server IP via SSH on the router ?

zulu14, 2020-08-23:
okay thanks, 

if that can help:

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/use-application-dns.net/' '/1.1.1.1/'
dhcp.@dnsmasq[0].serversfile='server=1.0.0.1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct' 'ihost.netflix.com'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1' '/netflix.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxext.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflximg.com/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflximg.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxvideo.net/omr_dst_bypass_all,omr6_dst_bypass_all' '/nflxso.net/omr_dst_bypass_all,omr6_dst_bypass_all'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.lan.limit='49'
dhcp.lan.start='1'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'

the maximal lease are already in unlimited
[image]

i will test to ping the router
I recently saw that the project and maybe you, live thanks to the donation, or can we make a donation?
thank!

Ysurac, 2020-08-23:
Many error in dnsmasq config, but I don't think it's the problem here: you added 1.1.1.1 and 1.0.0.1 in a wrong way.
The limit is "Nombre maximal de requêtes concurrentes" limitées à 150
I also see that eth0 go up and down, this make many problem. Check network cable and if there is no temperature problem.

For donation, this can be made on https://www.openmptcprouter.com/ there is link to paypal, patreon or what you want ;) (and yes it's me, I'm the developper)

zulu14, 2020-08-23:
the ping of the vps ip on the ssh work , 
0% packets loss

Ysurac, 2020-08-23:
And if on the VPS you ping 8.8.8.8 ?
And the result of a traceroute from the router to 8.8.8.8  (in Network->diagnostic menu or via ssh) ?

zulu14, 2020-08-23:
on the vps , the ping of 8.8.8.8 work,

traceroute to 8.8.8.8 (8.8.8.8), 30 hops max, 46 byte packets
 1  10.255.255.1  35.124 ms
 2  *
 3  *
 4  *
 5  *
 6  *
 7  *
 8  *
 9  *
10  *
and its the same "*" up to 30

(we are send a donation with paypal)

Ysurac, 2020-08-23:
10.255.255.1 is the VPN end on the VPS, so at least the ping seems to reach the VPS part.
As it's blocked on the VPS I think there is something wrong here.
You can try to restart shorewall on the VPS : ```systemctl restart shorewall```

Ysurac, 2020-08-23:
If restarting shorewall doesn't change anything, reboot the VPS.
No more idea for this evening and I need to sleep now sorry. Good luck.

zulu14, 2020-08-23:
okay thanks , the systemctl restart shorewall, fix the issues, the ping now wokrking

good night to you !

Adorfer, 2020-08-24:
What kind of emergency-service was affected? 
From the description it looks more like "OMR not interfacing the VPS", not like problem with OMR-bypass or whatever emergency-service you tried to route "specially".
(Tryping to understand the ticket)

kevin39, 2020-08-24:
I think the "emergency" word is just a bad (french) translation ;)
I think he would have preferred "important" or "urgent".

zulu14, 2020-08-24:
the emergency is not strictly speaking on omr, i have six omr instances, and the one that broke is networking a small fleet of 10 dedicated servers where there are about 500 customers. a reset of omr and a restart of the vps didn't fix the problem, which is somehow odd considering that a shorewall restart was enough to bring the network back.

closed 2020-08-24 by zulu14