Ysurac/openmptcprouter #1173: Aggregated connection not working

jhackeur, 2020-08-24:
## Expected Behavior
Aggregated connection for my ADSL, 4G, and second 4G access

## Current Behavior
My connection are not aggregated - only master connection is used

## Possible Solution
fix multipath vpn ?

## Steps to Reproduce the Problem

  1. connect my 3 access
  2. run a speedtest and see that bandwidth/latency are the same as "master" access only

## Context (Environment)
It's only a bandwidth issue, new design so was not work before.
If you have a look on the screenshot, you can see that my 3 access are working:
1/ if I connect my 3 access I got this:
[image]
[image]
2/if I connect only ADSL I got this: 
[image]
[image]
3/if I connect only 4G got this: 
[image]
[image]
4/if I connect only 4G-BR200 got this: 
[image]
[image]

The question is: why my access are not aggregated ?

## Specifications

  - OpenMPTCProuter version: v0.55.2
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: ESXI

Ysurac, 2020-08-24:
Please reboot the VPS, it was updated but not rebooted: it seems to use an old kernel.
Check in Network->MPTCP and "MPTCP fullmesh" tab that all IPs are visible.
To test speed, use via SSH on the router ```omr-test-speed``` to test aggregated speed and ```omr-test-speed wan1``` to test wan1 speed (where wan1 is real interface name visible via ```ip link```). Test during 2 minutes and stop with Ctrl+C

jhackeur, 2020-08-24:
Thanks for the answer!

I have rebooted both (server and router), but still have the same issue.
Regarding MPTCP fullmesh:
[image]

Please find below the omr-test-speed:
[image]

Ysurac, 2020-08-24:
What is the result if you remove ADSL ?
What is your VPS provider ?

jhackeur, 2020-08-24:
Here the result without ADSL (eth1):
[image]

I use an an ESX hosted on dedicated server provided by OneProvider

Ysurac, 2020-08-24:
Can you check using htop on the router and on the VPS the CPU load and memory usage when omr-test-speed (only, no need to test ethX this time) is running ?

jhackeur, 2020-08-24:
Here the result:
- for router side:
[image]
- for server side:
[image]

jhackeur, 2020-08-25:
I don't think this is related to perf issue, isn't ?
But if needed, I can add ressources.

Ysurac, 2020-08-25:
This doesn't seems to be the problem.

Ysurac, 2020-08-25:
There is no router/firewall like Cisco material somewhere on the path ?
Can you check latency from each wan to the VPS ip using ```ping -B -I eth1 -c 10 <vpsip>``` for each interface ?

jhackeur, 2020-08-25:
Indeed there is firewalls involved in this design, please find below:
[image]

Regarding the latency:
[image]
Second 4G access is down at this time... 

What I can't explain is: why each tunnel separately works fine but not works in aggregated mode ?
Any other check I can do ?

To be honnest I'm a security guys and not a network guys, and I don't really understand how multipath VPN dispatching works....

Ysurac, 2020-08-25:
Multipath TCP add a multipath flag in TCP header, some firewall remove it.
And this may be the case here: https://supportcenter.checkpoint.com/supportcenter/portal?solutionid=sk114666

jhackeur, 2020-08-25:
mmmmhhh indeed, seems to be stripped here.....
I double check this tomorrow and perform capture before and after firewall to track this flag - thanks a lot for this informations!

jhackeur, 2020-08-27:
So for your record, Checkpoint firewall remove the flag - you point indeed the issue!
And it's not possible to allow this flag as Multipath VPN is considered as security issue by checkpoint...

So I bypass the firewall for test, and working as expected yet!

thanks again for your good job :)

closed 2020-08-27 by jhackeur