Ysurac/openmptcprouter #1175: script update vps

chris77170, 2020-08-24:
## Expected Behavior

- update of vps in 0.1017

## Current behavior

- vps Milkywan (master) functional in 0.1015 but impossible to update to 0.1017

- vps Scaleway (backup) functional in 0.1017 update done without worries

- on the vps master, when validating the command line in order to update the vps, I only get as a result a simple return to the empty line

## Possible Solution

## Steps to Reproduce the Problem

## Specifications

  - OpenMPTCProuter version: 0.55
  - VPS version of OpenMPTCProuter: 0.1015
  - OpenMPTCProuter platform: x86_64
  - VPS Milkywan: Linux xxxxx.xxxxx 4.19.104-mptcp #1 SMP Mon Feb 17 11:08:15 UTC 2020 x86_64 
  - VPS Scaleway: Linux xxxxx.xxxxx 5.4.52-mptcp #1 SMP Mon Jul 20 16:09:04 UTC 2020 x86_64 GNU/

Ysurac, 2020-08-24:
Without any output it's difficult to say anything. The script should always exit with at least a comment...

vinceducat, 2020-08-24:
il have simiral problem with 0.1015 to 0.1017 update au Milky VPS... il do a complete reinstalla of the vps... 

but for now if you can send info to @Ysurac  the probleme could be identified and solve ;-)

chris77170, 2020-08-24:
@Ysurac just, there is no output from the script.
when i pass the command line to update the vps i only have a carriage return and nothing else. it is as if the script was empty (but I checked it is good ..) and on the other hand it runs without problem on a scaleway vps.

robertoglxda, 2020-08-24:
I had the same issue today. In my case, the script failed at line 91 (script for Ubuntu 20):
`apt-get check >/dev/null 2>&1`

After some hours, I found out that this was caused by LXD. My apt package manager was somehow 'broken', and I had to fix the LXD issue manually: in my case, I had to use 'snap' to install LXD ('snap install lxd'), then I did a manual 'apt --fix-broken install' and then the script was working fine.
For some reason, the check at line 91 is not working: it does not report the error in the 'if' as it should, but it does not continue neither.

chris77170, 2020-08-25:
Hello,
So I moved forward a little, well ... After an apt update & upgrade, the script starts and gives me this :

```
Reading package lists... Done
Building dependency tree
Reading state information... Done
curl is already the newest version (7.64.0-4+deb10u1).
libcurl4 is already the newest version (7.64.0-4+deb10u1).
git is already the newest version (1:2.20.1-2+deb10u3).
rename is already the newest version (1.10-1).
unzip is already the newest version (6.0-23+deb10u1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Install kernel linux-image-5.4.52-mptcp_1.11+206826e
 !!! if kernel install fail run: dpkg --remove --force-remove-reinstreq linux-image-5.4.52-mptcp !!!
Selecting previously unselected package linux-image-5.4.52-mptcp.
(Reading database ... 77108 files and directories currently installed.)
Preparing to unpack .../linux-image-5.4.52-mptcp_1.11+206826e_amd64.deb ...
Ok, aborting, since modules for this image already exist.
dpkg: error processing archive /tmp/linux-image-5.4.52-mptcp_1.11+206826e_amd64.deb (--install):
 new linux-image-5.4.52-mptcp package pre-installation script subprocess returned error exit status 1
Examining /etc/kernel/postrm.d .
run-parts: executing /etc/kernel/postrm.d/initramfs-tools 5.4.52-mptcp /boot/vmlinuz-5.4.52-mptcp
run-parts: executing /etc/kernel/postrm.d/zz-update-grub 5.4.52-mptcp /boot/vmlinuz-5.4.52-mptcp
Errors were encountered while processing:
 /tmp/linux-image-5.4.52-mptcp_1.11+206826e_amd64.deb
```
I looked on the forum and in on github but I found nothing conclusive.

chris77170, 2020-08-25:
Well, not liking to remain in front of a failure, I found a solution, I do not say that C the solution to all but that to work for me.
I put the manipulation if it can help someone, @Ycarus if need to modify what I write do not hesitate.

So from what I understand the script didn't want to hear anything ...

So I did an

```
sudo -s
apt update
apt upgrade
```

there the script started, but it stopped on setting up the kernel 5.4.52-mptcp

So I run the following two lines in order to erase the old kernel ... why???
```
/etc/kernel/postrm.d/initramfs-tools 5.4.52-mptcp /boot/vmlinuz-5.4.52-mptcp
/etc/kernel/postrm.d/zz-update-grub 5.4.52-mptcp /boot/vmlinuz-5.4.52-mptcp
```
a reboot of the vm

change to root
and the small order that goes well 
`wget -O - https://www.openmptcprouter.com/server/debian10-x86_64.sh | sh
`
and an updated OMR VM.

Thank you @Ysurac  for the work and the availability, and thank you for what contributes to the help given ...

Ysurac, 2020-08-25:
Else the solution was in the log:  ``` !!! if kernel install fail run: dpkg --remove --force-remove-reinstreq linux-image-5.4.52-mptcp !!!``` ;)

chris77170, 2020-08-25:
>
> 
> Else the solution was in the log: ` !!! if kernel install fail run: dpkg --remove --force-remove-reinstreq linux-image-5.4.52-mptcp !!!` ;)

I read it right but the command didn't want to work either

Ysurac, 2020-08-25:
ok, maybe it was another issue

chris77170, 2020-08-25:
....
Finally as long as it works.
Thanks again for the help and the availability

closed 2020-08-25 by chris77170