Ysurac/openmptcprouter #1177: lan hard network problem

Kalimeiro, 2020-08-25:
Hi,

With 0.55.1 and 0.55.2 i have discovered a hard problem with network on my raspberry, i think the problem is on Kernel or LAN Driver.

Time to Time, when i perform shutdown, or reboot, i can't have any lan access, the router doesn't responding, i need to perform hard reboot electrically and then system is remounted as RO, reboot via gui to make fsck, and ... router doesn't responding ... etc... etc...

So i have attached HDMI cable to see with dmesg, and i discovered time to time "network unreachable" at boot start (~8/9s from start)

ip route show me "linkdown" but network cable is good, led green and orange flash...

i know it will work when i see at end startup : IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready

I have tested to removed 'force link' in eth0 interface, same thing.

## Specifications

  - OpenMPTCProuter version: 0.55.2
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: Raspberry Pi 3 Model B Plus Rev 1.3

Ysurac, 2020-08-25:
You don't have anything on USB port ?
No problem with the power supply ?
Shutdown or reboot are down via the interface ? Never any problem when electrically rebooted ?
You are RO, what is the FS you choose (ext4 or squashfs) ?

Kalimeiro, 2020-08-25:
> 
> 
> You don't have anything on USB port ?
No USB port used,

> No problem with the power supply ?
I have tested with many power supply and other motherboard RPI 3B+ REV 1.2 AND 1.3 : same thing.

> Shutdown or reboot are down via the interface ? Never any problem when electrically rebooted ?
i have shutdown/reboot with interface and when i don't have network i reboot in cli.

> You are RO, what is the FS you choose (ext4 or squashfs) ?
i use ext4 because bad experience with squashfs (lost configuration time to time when upgrade)

Kalimeiro, 2020-08-25:
I will send you a video to show in live the problem

Dilandau, 2020-08-26:
Same problem with my RPI 3B+ : 0.55.3 (64-bits)
Lan is UP with factory default, when i add basic config (ip/router/key) for MPTCP, lan do not respond.

Ysurac, 2020-08-26:
This doesn't seems to be same problem.
For Kalimeiro it's after a reboot, in your case is when you set some info in the wizard.

Malaga82, 2020-10-07:
With pi 3b+ there are several problems, i can confirm.
If you boot without video output it seems to boot fine, but the ethernet card is never really connected and you cannot ping it.
If you boot with video output, strangely, the ethernet card will be up and connected to the network, you can ping it and access web UI, create the tunnel etc.
Tried with 64 bit, 32 bit, dev build, same ethernet card issue.
If i remember well, there was several issue with 3B+ on openwrt itself, and i used custom build to ran it properly, i don't know if this will ever be fixed and i sadly switched to another HW. 3B+ is not reliable with openwrt base.
It will be great if other 3B+ owners could comment about this.

closed 2021-01-10 by github-actions[bot]