Ysurac/openmptcprouter #1178: Redirect port from router

LeprovostNoam, 2020-08-25:
## Expected Behavior
The traffic on the port 192.168.100.1:82 of my router will be redirected to 192.168.100.119:81

## Current Behavior
It's not working, the connection is refused: ERR_CONNECTION_REFUSED

## Steps to Reproduce the Problem
[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.55 r0+14022-18b7d87a8f / LuCI Master git-20.215.65300-c4b617a
  - OpenMPTCProuter VPS version: 0.55
  - OpenMPTCProuter platform: Linksys WRT3200ACM

Ysurac, 2020-08-25:
Did you configure port forward ? Check https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding

LeprovostNoam, 2020-08-25:
Ok, i config with port forward like the wiki but not working vpsip:81 to 192.168.100.118:81
[image]

Ysurac, 2020-08-25:
Try to select only TCP as protocol.

Ysurac, 2020-08-25:
And check on the VPS that you have the rules in ```/etc/shorewall/rules``` and that the correct VPN IP is in ```/etc/shorewall/params.vpn``` for OMR_ADDR (10.255.255.2 if glorytun tcp is used, it's the default).

LeprovostNoam, 2020-08-26:
ERR_CONNECTION_REFUSED when i'm just using tcp. When i put the ips of the router connected to the wans port (Who share the internet access) router its working but not the router connected on the lan port why?
`root@instance-1:~# cat /etc/shorewall/rules
#
# Shorewall version 4.0 - Sample Rules File for two-interface configuration.
# Copyright (C) 2006-2014,2007 by the Shorewall Team
#
# This library is free software; you can redistribute it and/or
# modify it under the terms of the GNU Lesser General Public
# License as published by the Free Software Foundation; either
# version 2.1 of the License, or (at your option) any later version.
#
# See the file README.txt for further details.
#------------------------------------------------------------------------------
# For information about entries in this file, type "man shorewall-rules"
######################################################################################################################################################################################################
#ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME            HEADERS         SWITCH          HELPER
#                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
#
#       Accept DNS connections from the firewall to the network
#
DNS(ACCEPT)     $FW             net
#
#       Allow Ping from/to the VPN
#
Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    $FW             vpn
#
#       Allow Ping from the firewall to the network
#
Ping(ACCEPT)    $FW             net
#
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
#	Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
#	Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
#	DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT)	$FW		vpn
#
#	Redirect all port from 1 to 64999 to the VPN client from the network
#
DNAT		net		vpn:$OMR_ADDR	tcp	1-64999
DNAT		net		vpn:$OMR_ADDR	udp	1-64999
ACCEPT		net		$FW		tcp	65101	# OMR openmptcprouter open shadowsocks port tcp
ACCEPT		net		$FW		udp	65101	# OMR openmptcprouter open shadowsocks port udp
ACCEPT		net		$FW		tcp	65001	# OMR openmptcprouter open glorytun port tcp
DNAT		net		vpn:$OMR_ADDR	tcp	8000	# OMR openmptcprouter redirect router 8000 port tcp
root@instance-1:~# cat /etc/shorewall/params.vpn
VPS_ADDR=10.255.255.1
VPS_IFACE=gt-tun0
OMR_ADDR=10.255.255.2
`

Ysurac, 2020-08-26:
router connected to lan port ? I don't understand.
Check that SNAT is configured, it's often needed.

LeprovostNoam, 2020-08-26:
OK, i have this configuration: 
[image]
Normaly, i'm connected on 192.168.1.1 so the openwrt router:
I have access to all that is above, so i access to 192.168.1.1:81 and 192.168.1.118:81 what are exactly the sames
No i've make a test, I connected directly on OpenMPTCP Router and i've not access to 192.168.118:81 and 192.168.118 lucy interface too so i think it's possible that 192.168.1.118 don't allow the connexion.
Is openwrt disallow the connexion by default? i've don't touched the firewall

Ysurac, 2020-08-26:
It should disable all wan connections by default.
I added tcpdump command to check if packets come to OpenMPTCProuter to the doc: https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding#debug

LeprovostNoam, 2020-08-26:
Ok, wait i've disable the firewall, no i've not access to the router. I'm looking for a way to have access again..

LeprovostNoam, 2020-08-26:
My router is back. Yes i know i'm a noob with openwrt xD.
Do you have an idea to allow wan connection on the openwrt router,  i don't wanna make another mistake by looking on google

Ysurac, 2020-08-26:
You can set Accept in firewall general settings for forward, input and ouput and same for zone if there is some defined.

closed 2020-12-29 by github-actions[bot]