Ysurac/openmptcprouter #1179: Pi-hole web interface ERR_CONNECTION_REFUSED

mack1726, 2020-08-26:
## Expected Behavior

## Current Behavior
After update OpenMPTCProuter from ver 0.54 to ver 0.55.3 Pi-hole web interface ERR_CONNECTION_REFUSED
I tried completely reinstalling VPS server and OpenMPTCProuter but it did not help.

## Possible Solution

## Steps to Reproduce the Problem

  1. Install Pi-hole on OpenMPTCProuter VPS and reboot vps
  2. Then  'Save & Apply' the wizard again in System->OpenMPTCProuter and reboot
  3. Web interface will be not available on 10.255.255.1 (ERR_CONNECTION_REFUSED)

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version: last version
  - OpenMPTCProuter platform:RPI4

Ysurac, 2020-08-26:
What is the result of ```curl 10.255.255.1``` from the router via SSH ?

mack1726, 2020-08-26:
root@OpenMPTCProuter:~# curl 10.255.255.1
curl: (7) Failed to connect to 10.255.255.1 port 80: Connection refused

Ysurac, 2020-08-26:
```ping 10.255.255.1``` is working ?

mack1726, 2020-08-26:
Yes.
[image]

Ysurac, 2020-08-26:
what is the result of ```netstat -laputen``` on the VPS ?

mack1726, 2020-08-26:
[image]

mack1726, 2020-08-26:
root@v277087:~# netstat -laputen
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State                                                                                                                                                                    User       Inode      PID/Program name
tcp        0      0 0.0.0.0:65401           0.0.0.0:*               LISTEN                                                                                                                                                                   0          6369       372/dsvpn
tcp        0      0 0.0.0.0:65500           0.0.0.0:*               LISTEN                                                                                                                                                                   0          9278       698/python3
tcp        0      0 0.0.0.0:65222           0.0.0.0:*               LISTEN                                                                                                                                                                   0          4559       286/sshd
tcp        0      0 127.0.0.1:4711          0.0.0.0:*               LISTEN                                                                                                                                                                   998        7662       616/pihole-FTL
tcp        0      0 0.0.0.0:65001           0.0.0.0:*               LISTEN                                                                                                                                                                   0          5043       325/glorytun-tcp
tcp        0      0 0.0.0.0:65101           0.0.0.0:*               LISTEN                                                                                                                                                                   0          5889       401/ss-server
tcp        0      0 0.0.0.0:53              0.0.0.0:*               LISTEN                                                                                                                                                                   998        7638       616/pihole-FTL
tcp        0      0 0.0.0.0:65301           0.0.0.0:*               LISTEN                                                                                                                                                                   0          5557       319/openvpn
tcp        0      0 94.103.85.162:65101     213.87.135.137:19570    ESTABLISHED                                                                                                                                                              0          680038     401/ss-server
tcp        0      0 94.103.85.162:65101     213.87.159.196:58416    ESTABLISHED                                                                                                                                                              0          248242     401/ss-server
tcp        0      0 94.103.85.162:14463     198.11.132.250:80       TIME_WAIT                                                                                                                                                                0          0          -
tcp        0      0 94.103.85.162:65101     213.87.135.137:54187    ESTABLISHED                                                                                                                                                              0          1275111    401/ss-server
tcp        0      0 94.103.85.162:25385     13.107.246.254:443      ESTABLISHED                                                                                                                                                              0          1272885    401/ss-server
tcp        0      0 94.103.85.162:65101     188.170.193.214:17333   ESTABLISHED                                                                                                                                                              0          272982     401/ss-server
tcp        0      0 94.103.85.162:65101     188.170.193.236:24006   ESTABLISHED                                                                                                                                                              0          60448      401/ss-server
tcp        0      0 94.103.85.162:45547     17.57.146.136:5223      ESTABLISHED                                                                                                                                                              0          312873     401/ss-server
tcp        0      0 94.103.85.162:65101     213.87.135.137:53597    ESTABLISHED                                                                                                                                                              0          306614     401/ss-server
tcp        0      0 94.103.85.162:65101     188.170.193.214:52890   ESTABLISHED                                                                                                                                                              0          1225775    401/ss-server
tcp        0      0 94.103.85.162:65101     188.170.193.214:6331    TIME_WAIT                                                                                                                                                                0      [... cut ...]
udp6       0      0 :::53                   :::*                                                                                                                                                                                             998        7639       616/pihole-FTL
udp6       0      0 :::65101                :::*                                                                                                                                                                                             0          5891       401/ss-server

Ysurac, 2020-08-27:
lighttpd is not listening so the pihole interface is down. You can check in ```/var/log/lighttpd/error.log``` if there is something and you can try to restart lighttpd ```systemctl restart lighttpd```

mack1726, 2020-08-27:
2020-08-26 18:21:09: (server.c.1464) server started (lighttpd/1.4.53) 
2020-08-26 18:21:21: (server.c.2059) server stopped by UID = 0 PID = 1 
2020-08-26 18:21:22: (server.c.1464) server started (lighttpd/1.4.53) 
2020-08-26 18:21:56: (server.c.2059) server stopped by UID = 0 PID = 1

Ysurac, 2020-08-27:
What do you have in ```/etc/lighttpd/external.conf``` ? and what is the result of ```ip a``` on the VPS ?

mack1726, 2020-08-27:
server.bind="10.255.255.1"
$SERVER["socket"] == "10.255.254.1:80" { }
$SERVER["socket"] == "10.255.252.1:80" { }
$SERVER["socket"] == "10.255.251.1:80" { }
$SERVER["socket"] == "10.255.253.1:80" { }

mack1726, 2020-08-27:
root@v277087:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP,80000> mtu 65536 qdisc noqueue state UNKNOWN group                                                                                                                                                              default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host
       valid_lft forever preferred_lft forever
2: ens3: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc prio state UP group de                                                                                                                                                             fault qlen 1000
    link/ether 52:54:00:04:3a:5f brd ff:ff:ff:ff:ff:ff
    inet 94.103.85.162/24 brd 94.103.85.255 scope global ens3
       valid_lft forever preferred_lft forever
    inet6 fe80::bf46:8478:31d6:82e8/64 scope link
       valid_lft forever preferred_lft forever
3: gt-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc prio                                                                                                                                                              state UNKNOWN group default qlen 500
    link/none
    inet 10.255.255.1/30 brd 10.255.255.3 scope global gt-tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::cfe3:dd3:9c9b:2792/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
4: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state                                                                                                                                                              UNKNOWN group default qlen 100
    link/none
    inet 10.255.252.1 peer 10.255.252.2/32 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::d9cd:8fcc:b89:9772/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
5: tun1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc fq state                                                                                                                                                              UNKNOWN group default qlen 100
    link/none
    inet 10.255.250.1 peer 10.255.250.2/32 scope global tun1
       valid_lft forever preferred_lft forever
    inet6 fe80::d45f:b1f7:b607:3697/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
6: dsvpn0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 9000 qdisc fq sta                                                                                                                                                             te UNKNOWN group default qlen 500
    link/none
    inet 10.255.251.1 peer 10.255.251.2/32 scope global dsvpn0
       valid_lft forever preferred_lft forever
    inet6 64:ff9b::aff:fb01 peer 64:ff9b::aff:fb02/96 scope global
       valid_lft forever preferred_lft forever
    inet6 fe80::47e1:5222:e74f:e555/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
7: gt-udp-tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP,80000> mtu 1500 qdisc f                                                                                                                                                             q state UP group default qlen 500
    link/none
    inet 10.255.254.1/30 brd 10.255.254.3 scope global gt-udp-tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::f990:fd5:9e4a:4991/64 scope link stable-privacy
       valid_lft forever preferred_lft forever
8: mlvpn0: <POINTOPOINT,MULTICAST,NOARP,80000> mtu 1444 qdisc noop state DOWN gr                                                                                                                                                             oup default qlen 500
    link/none
9: sit0@NONE: <NOARP,80000> mtu 1480 qdisc noop state DOWN group default qlen 10                                                                                                                                                             00
    link/sit 0.0.0.0 brd 0.0.0.0
10: omr-6in4-user0@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP,80000> mtu 1480 qdisc no                                                                                                                                                             queue state UNKNOWN group default qlen 1000
    link/sit 10.255.255.1 peer 10.255.255.2
    inet6 fe80::aff:ff01/64 scope link
       valid_lft forever preferred_lft forever
    inet6 fe80::a00:1/126 scope link
       valid_lft forever preferred_lft forever
root@v277087:~#

Ysurac, 2020-08-27:
The problem is that mlvpn is down. You can restart it ```systemctl restart mlvpn@mlvpn0``` or remove the line ```$SERVER["socket"] == "10.255.253.1:80" { }``` in /etc/lighttpd/external.conf and restart lighttpd.

mack1726, 2020-08-27:
Ysurac, Thank you very much. systemctl restart mlvpn@mlvpn0 did not help. But remove the line $SERVER["socket"] == "10.255.253.1:80" { } in /etc/lighttpd/external.conf and restart lighttpd. - it worked. Why did it happen? When restart vps, the problem will not return? Thanks again

Ysurac, 2020-08-27:
mlvpn is sometimes not stable, it's why I removed it (again) of wizard VPN list. If a VPN is not running on the VPS side, lighttpd server (user for Pi-Hole interface) can't listen to it IP, and by default it should listen to all VPN IPs, so it doesn't start.

The problem will not return after VPS restart.

mack1726, 2020-08-27:
Ok, thank you :)

closed 2020-12-30 by github-actions[bot]