Ysurac/openmptcprouter #1180: Compile question.

Rwestrem, 2020-08-26:
Not an issue.  Just a question about compiling for my unsupported device.  I'm having wifi issues with my linksys wrt3200 and I have a netgear r7800 that has moved to the new kernel in master.  I'm currently compiling right now but wondering if you're build environment will set the wan as lan and the lans as wan like the wrt3200 or is that something I will need to configure?

Ysurac, 2020-08-26:
No idea, if configuration is similar as a supported device this will correctly set network, else no.

Rwestrem, 2020-08-27:
> No idea, if configuration is similar as a supported device this will correctly set network, else no.

I compiled for the Netgear Nighthawk r7800.  It seems the only port that works is the 1st lan port to get to openmptcptrouter.  the rest including the wan port do not seem to function.  Is there anything in the settings that stands out?

dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].boguspriv='1'
dhcp.@dnsmasq[0].filterwin2k='0'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].resolvfile='/tmp/resolv.conf.d/resolv.conf.auto'
dhcp.@dnsmasq[0].nonwildcard='1'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
[... 1352 lines cut ...]
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='1'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth0.1' 'eth0.2' 'eth0.3' 'eth0.4'
Powered by openmptcprouter v0.55.3 r0+14198-da98603597

Rwestrem, 2020-09-05:
Works fine.  Just needed to set up internal switch correctly.

closed 2020-09-05 by Rwestrem