Ysurac/openmptcprouter #1181: Kernel Oops occurs every few minutes

milankragujevic, 2020-08-28:
## Expected Behavior
No Kernel oops should regularly occur. 

## Current Behavior
Every few minutes, the following message is logged in the syslog: 

```
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.377895] ------------[ cut here ]------------
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.378837] WARNING: CPU: 3 PID: 4356 at cfg80211_probe_status+0x54c/0x750 [cfg80211]
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.380571] Modules linked in: ath9k ath9k_common rtl8192cu rtl8192c_common rtl_usb rt2800usb rt2800lib qcserial pppoe ppp_async option iwlmvm iwldvm ipw cdc_mbim ath9k_hw ath usb_wwan usb_serial_simple ti_usb_3410_5052 sr9700 smsc95xx smsc75xx sierra_net sierra rtlwifi rtl8xxxu rtl8187 rt2x00usb rt2x00lib rndis_host qmi_wwan pppox ppp_generic pl2303 oti6858 mt7601u mos7720 mct_u232 mcs7830 mac80211 lzo keyspan kalmia iwlwifi ipt_REJECT huawei_cdc_ncm garmin_gps ftdi_sio ebtable_nat ebtable_filter ebtable_broute dm9601 cypress_m8 cp210x ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem belkin_sa ax88179_178a asix ark3116 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_state xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_ndpi xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iface xt_hl xt_helper xt_hashlimit xt_esp xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connbytes xt_condition xt_comment xt_addrtype xt_TRACE xt_TPROXY xt_TCPMSS xt_REDIRECT xt_MASQUERADE
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.380621]  xt_LOG xt_IPMARK xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_ACCOUNT visor via_velocity via_rhine usbserial usbnet tulip ts_fsm ts_bm solos_pci slhc sky2 skge sis900 sis190 sch_cake rtl8150 r8712u(C) r8169 r8152 r6040 pegasus pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount ne2k_pci mmc_spi macvlan lzo_decompress lzo_compress kaweth iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip6table_raw ip_tables hso forcedeth ezusb ethoc et131x ebtables ebt_vlan ebt_stp ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_among ebt_802_3 e1000e e100
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.392738]  crc7 crc_itu_t crc_ccitt compat_xtables compat cdc_wdm cdc_acm bnx2 atl2 atl1e atl1c atl1 asn1_decoder arptable_filter arpt_mangle arp_tables 8390 8139too 8139cp tcp_nanqinlang sch_teql sch_sfq sch_red sch_prio sch_pie sch_multiq sch_gred sch_fq sch_dsmark sch_codel em_text em_nbyte em_meta em_cmp act_simple act_police act_pedit act_ipt act_gact act_csum sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred configs evdev i2c_gpio i2c_dev spi_ks8995 ledtrig_heartbeat xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.404861]  ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 swconfig 3c59x ip6_gre ip_gre gre ixgbe igb e1000 ifb nat46 sit sctp libcrc32c mdio ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 ip6_tunnel netlink_diag tunnel6 tunnel4 ip_tunnel hfcpci hfcmulti veth tun mISDN_dsp l1oip mISDN_core xfrm_user xfrm_ipcomp af_key xfrm_algo vfat fat autofs4 br2684 atm nls_utf8 nls_iso8859_1 nls_cp437 zram zsmalloc natsemi eeprom_93cx6 sha256_ssse3 sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg md5 hmac ghash_generic gcm echainiv des_generic libdes deflate ctr cbc authenc crypto_acompress sdhci_pltfm xhci_plat_hcd fsl_mph_dr_of ehci_platform ehci_fsl gpio_button_hotplug button_hotplug tg3 b44 ssb ptp realtek broadcom bcm_phy_lib pps_core mii libphy
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.426241] CPU: 3 PID: 4356 Comm: rpcd Tainted: G        WC        5.4.58 #0
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.427955] Hardware name: GIGABYTE GB-BACE-3160/MZBSWMP-00, BIOS F6 06/13/2018
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.429686] RIP: 0010:cfg80211_probe_status+0x54c/0x750 [cfg80211]
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.430661] Code: 9a 00 00 00 48 89 df c7 45 d4 01 00 00 00 e8 bb d5 8a e0 85 c0 0f 84 35 ff ff ff 48 89 df e8 2b 0c cb e0 b8 a6 ff ff ff eb 81 <0f> 0b 48 89 df e8 1a 0c cb e0 b8 ea ff ff ff e9 6d ff ff ff eb db
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.433443] RSP: 0018:ffffc900006dfa50 EFLAGS: 00010202
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.434386] RAX: 0000000000000000 RBX: ffff888173d0ae00 RCX: 0000000000000000
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.436108] RDX: ffff888176cc0008 RSI: 0000000000000000 RDI: ffff888176cc02e0
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.437821] RBP: ffffc900006dfa88 R08: 0000000000000004 R09: ffff88817b6b4014
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.439539] R10: ffffffffa0ace000 R11: 0000000000000004 R12: ffffc900006dfab8
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.441252] R13: 0000000000000000 R14: ffff88817b6b4014 R15: ffff888176cc02e0
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.442966] FS:  00007f81ff2e4d48(0000) GS:ffff88817bf80000(0000) knlGS:0000000000000000
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.444710] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.445665] CR2: 00007fa23a75e000 CR3: 0000000176c4c000 CR4: 00000000001006e0
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.447382] Call Trace:
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.448245]  genl_family_rcv_msg+0x1ca/0x3e0
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.449159]  genl_rcv_msg+0x43/0x90
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.450047]  ? genl_family_rcv_msg+0x3e0/0x3e0
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.450964]  netlink_rcv_skb+0x4d/0x120
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.451862]  genl_rcv+0x23/0x40
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.452739]  netlink_unicast+0x169/0x1f0
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.453640]  netlink_sendmsg+0x1e4/0x380
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.454544]  ? netlink_unicast+0x1f0/0x1f0
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.455451]  ____sys_sendmsg+0x226/0x250
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.456352]  ? copy_msghdr_from_user+0xbd/0x130
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.457271]  ___sys_sendmsg+0x7a/0xb0
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.458166]  ? call_rcu+0xb/0x10
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.459052]  ? destroy_inode+0x60/0x70
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.459948]  ? evict+0x134/0x190
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.460829]  ? kmem_cache_free+0x286/0x290
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.461737]  ? _cond_resched+0x14/0x30
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.462636]  ? __dentry_kill+0x11f/0x170
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.463536]  __sys_sendmsg+0x40/0x70
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.464427]  __x64_sys_sendmsg+0x1a/0x20
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.465333]  do_syscall_64+0x54/0x380
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.466228]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.467163] RIP: 0033:0x7f81ff2c6373
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.468054] Code: c3 8b 07 85 c0 75 24 49 89 fb 48 89 f0 48 89 d7 48 89 ce 4c 89 c2 4d 89 ca 4c 8b 44 24 08 4c 8b 4c 24 10 4c 89 5c 24 08 0f 05 <c3> e9 19 d3 ff ff 41 54 b8 02 00 00 00 49 89 f4 be 00 08 08 00 55
Fri Aug 28 08:33:06 2020 kern.warn kernel: [293371.470840] RSP: 002b:00007fff874d4978 EFLAGS: 00000246 ORIG_RAX: 000000000000002e
[... cut ...]
The setup works normally, combines LTE (~ 50 - 75 Mbps depending on congestion) and VDSL 50 Mbps for up to 125 Mbps download, ping is stable at around 6ms. 

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.55.2 r0+14198-da98603597 / LuCI Master git-20.219.57394-35e2735
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1017 on Intel x64 KVM VPS with 1.5 GB RAM and 1 dedicated vCore, symmetric full-duplex 200 Mbps link, 5ms ping from modem to VPS
  - OpenMPTCProuter platform: x86_64 non-EFI ext4

Ysurac, 2020-08-28:
This seems to be related to "cfg80211", do you use wifi ?

milankragujevic, 2020-08-28:
No, I don't. I want to disable it then, to try what will happen. Can you tell me how?

Ysurac, 2020-08-28:
You can try using System->Software in installed tab to remove "kmod-cfg80211"

milankragujevic, 2020-08-28:
That has resolved the issue for me. Thank you.

closed 2020-08-28 by milankragujevic