Ysurac/openmptcprouter #1182: Interfaces omrip0gre and omrip1gre keep going down and being set up

milankragujevic, 2020-08-28:
## Expected Behavior
Interface should not go down and set up repeatedly.

## Current Behavior
These errors appear in the system log:

Fri Aug 28 19:53:44 2020 daemon.notice netifd: Interface 'omrip0gre' is setting up now
Fri Aug 28 19:53:44 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Fri Aug 28 19:53:44 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now
Fri Aug 28 19:53:44 2020 daemon.notice netifd: Interface 'omrip0gre' is now down
Fri Aug 28 19:53:44 2020 daemon.notice netifd: Interface 'omrip0gre' is setting up now
Fri Aug 28 19:53:44 2020 daemon.notice netifd: Interface 'omrip1gre' is now down
Fri Aug 28 19:53:44 2020 daemon.notice netifd: Interface 'omrip1gre' is setting up now

## Steps to Reproduce the Problem

  1. On GIGABYTE BRIX GB-BACE-3160, boot OpenMPTCProuter from USB drive
  2. Install OpenMPTCProuter VPS script on a VPS with 2 ip addreses
  3. This error appears in the log

## Context (Environment)
The VPS has 2 public IP addresses assigned. I tried removing one address and this still appears. The network works fine but this error is logged every second constantly. I do not need 2 IP addresses on the VPS, this is something I cannot change right now, so I just need a way to fix this. 

[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.55.2 r0+14198-da98603597 / LuCI Master git-20.219.57394-35e2735
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1017 on Intel x64 KVM VPS with 1.5 GB RAM and 1 dedicated vCore, symmetric full-duplex 200 Mbps link, 5ms ping from modem to VPS
  - OpenMPTCProuter platform: x86_64 non-EFI ext4

Ysurac, 2020-08-28:
All is green in status page ?
Do you have any ```OMR-VPS``` error in State->System log ?

milankragujevic, 2020-08-28:
Yes, all green and no errors. I removed the interfaces (omrip0, omrip0gre, omrip1, omrip1gre) manually and rebooted both the router and the VPS at the same time. The errors do not appear anymore, I am not sure why this happened... 

Sorry for wasting your time, and thank you so much for creating OpenMPTCProuter and for your quick support! ☺ 🙌🎉

closed 2020-08-28 by milankragujevic