Ysurac/openmptcprouter #1183: Question : Installation wifi

AdrienDssy, 2020-08-28:
## Current Behavior

Bonjour,
J'aimerais savoir comment je pourrais mettre en place le wifi sur omr (ça fonctionne dans une tour toute simple). Car pour l'instant je n'ai pas encore réussi à activer le wifi

Ysurac, 2020-08-28:
Il est fortement préférable de remplir le template.
Sinon la doc : https://github.com/Ysurac/openmptcprouter/wiki/Configure-Wifi

AdrienDssy, 2020-08-28:
Justement le menu (Network > wifi) n'est pas présent sinon je n'aurais pas demander 😅😅

Ysurac, 2020-08-28:
D'où la demande de remplir le template qui aurait indiquer l'architecture sur lequel tourne OMR.
Si le menu wifi n'est pas présent c'est que le wifi n'est pas détecté/supporté. Il faut ajouter le driver qui va bien pour la carte/clef/autre présente.

AdrienDssy, 2020-08-28:
Il y aurais une documentation pour ajouter mes clé wifi ? :)

Ysurac, 2020-08-28:
Le problème est plutôt de savoir la clef wifi utilisée et ainsi de trouver le chipset pour enfin trouver le driver nécessaire si il existe.

AdrienDssy, 2020-08-28:
J'ai deux clef mais aucune détectés sur OMR

AdrienDssy, 2020-08-28:
Une clef TP Link TL-WN823N et l'autre clef TP Link TL-WN725N

Ysurac, 2020-08-28:
Et tu as regardé les chipsets utilisés par chacun ?

Ysurac, 2020-08-28:
Visiblement il y a plusieurs versions avec des chipsets différent pour chacune des clefs.
Il faudrait le résultat de ```lsusb``` via SSH sur le routeur quand elles sont branchées.

AdrienDssy, 2020-08-28:
Bus 003 Device 001: ID 1d6b:0002
Bus 002 Device 002: ID 8087:8000
Bus 001 Device 001: ID 1d6b:0002
Bus 003 Device 008: ID 0bda:8179
Bus 001 Device 002: ID 8087:8008
Bus 004 Device 001: ID 1d6b:0003
Bus 003 Device 005: ID 2357:0109
Bus 002 Device 001: ID 1d6b:0002
Bus 003 Device 002: ID 0781:5581

Ysurac, 2020-08-28:
Donc on a :
* TL-WN823N v2/v3 [Realtek RTL8192EU]
* RTL8188EUS 802.11n Wireless Network Adapter

Alors peut être kmod-rtl8xxxu pour les deux, mais aucune garantie. Je l'intégrerais dans la prochaine version.

AdrienDssy, 2020-08-28:
D'accord, merci

AdrienDssy, 2020-08-29:
Une dernière question, je ne peux pas l'intégrer moi même en attendant ? :)

closed 2020-08-29 by AdrienDssy