Ysurac/openmptcprouter #1184: Impossible to block net access for a given device

patimages, 2020-08-29:
## Expected Behavior
block net access for a given device 

## Current Behavior
Does not work as expected (as discussed in another forum)

## Steps to Reproduce the Problem

set up a rule to block internet access to a device using IP or MAC address

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.55.3 r0+14198-da98603597 / LuCI Master git-20.219.57394-35e2735
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: x86_64)

Ysurac, 2020-08-29:
What is the result of ```tcpdump -i eth0 ether host <macaddress>``` from the router via SSH when you try to exit ?
There is also ```tcpdump -i eth0 -e``` that should show all traffic with mac addresses but can be really verbose.

patimages, 2020-08-29:
Thanks ! What do you mean by "when you try to exit ?"

Ysurac, 2020-08-30:
I mean when you try to exit to internet with the client that you want to block, so when you want to go to a website from this client.

closed 2021-01-02 by github-actions[bot]