Ysurac/openmptcprouter #1185: Soucis VPN

AdrienDssy, 2020-08-29:
## Current Behavior

Bonjour,

Je constate souvent des soucis avec le VPN ce qui rend souvent down certains de mes services :(

> Sat Aug 29 15:35:36 2020 user.notice Shadowsocks: Rules DOWN
Sat Aug 29 15:35:36 2020 user.notice Shadowsocks: Rules UP
Sat Aug 29 15:35:36 2020 daemon.err omr-tracker[8315]: RTNETLINK answers: Permission denied
Sat Aug 29 15:35:36 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sat Aug 29 15:35:36 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sat Aug 29 15:35:37 2020 user.notice omr-bypass: Reload dnsmasq...
Sat Aug 29 15:35:37 2020 daemon.info dnsmasq[13947]: read /etc/hosts - 4 addresses
Sat Aug 29 15:35:37 2020 daemon.info dnsmasq[13947]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Sat Aug 29 15:35:37 2020 daemon.info dnsmasq-dhcp[13947]: read /etc/ethers - 0 addresses
Sat Aug 29 15:35:37 2020 user.notice omr-bypass: OMR-ByPass is running
Sat Aug 29 15:35:37 2020 daemon.info dsvpn[11660]: Done.
Sat Aug 29 15:35:37 2020 daemon.notice netifd: Network device 'tun0' link is down
Sat Aug 29 15:35:37 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity loss
Sat Aug 29 15:35:37 2020 daemon.notice netifd: Interface 'omrvpn' is now down
Sat Aug 29 15:35:37 2020 daemon.debug dnsmasq[13947]: stopped listening on tun0(#294): 10.255.251.2 port 53
Sat Aug 29 15:35:37 2020 user.notice ModemManager: hotplug: remove network interface tun0: event processed
Sat Aug 29 15:35:37 2020 user.notice ModemManager: hotplug: event reported: action=remove, name=tun0, subsystem=net
Sat Aug 29 15:35:37 2020 daemon.notice netifd: Interface 'omrvpn' is disabled
Sat Aug 29 15:35:37 2020 daemon.err omr-tracker[8315]: Device "tun0" does not exist.
Sat Aug 29 15:35:39 2020 daemon.info dsvpn: starting dsvpn vpn instance vpn
Sat Aug 29 15:35:39 2020 daemon.notice netifd: Interface 'omrvpn' is enabled
Sat Aug 29 15:35:39 2020 daemon.notice netifd: Network device 'tun0' link is up
Sat Aug 29 15:35:39 2020 daemon.notice netifd: Interface 'omrvpn' has link connectivity
Sat Aug 29 15:35:39 2020 daemon.notice netifd: Interface 'omrvpn' is setting up now
Sat Aug 29 15:35:39 2020 daemon.info dsvpn[1255]: Interface: [tun0]
Sat Aug 29 15:35:39 2020 daemon.info dsvpn[1255]: Trying to reconnect
Sat Aug 29 15:35:39 2020 daemon.info dsvpn[1255]: Connecting to 54.38.188.175:65401...
Sat Aug 29 15:35:39 2020 daemon.notice netifd: Interface 'omrvpn' is now up
Sat Aug 29 15:35:39 2020 user.notice firewall: Reloading firewall due to ifup of omrvpn (tun0)
Sat Aug 29 15:35:39 2020 user.notice ModemManager: hotplug: add network interface tun0: event processed
Sat Aug 29 15:35:39 2020 user.notice ModemManager: hotplug: event reported: action=add, name=tun0, subsystem=net
Sat Aug 29 15:35:39 2020 user.notice ModemManager: hotplug: error: parent device sysfspath not found
Sat Aug 29 15:35:39 2020 daemon.debug dnsmasq[13947]: listening on tun0(#295): 10.255.251.2 port 53
Sat Aug 29 15:35:39 2020 daemon.err dsvpn[1255]: RTNETLINK answers: Permission denied
Sat Aug 29 15:35:39 2020 daemon.info dsvpn[1255]: Connected
Sat Aug 29 15:35:39 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sat Aug 29 15:35:40 2020 user.notice omr-bypass: Restart dnsmasq...
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[13947]: exiting on receipt of SIGTERM
Sat Aug 29 15:35:40 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Sat Aug 29 15:35:40 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Sat Aug 29 15:35:40 2020 user.notice dnsmasq: Allowing RFC1918 responses for domain plex.direct
Sat Aug 29 15:35:40 2020 daemon.err /usr/bin/ss-redir[5883]: remote recv: Network unreachable
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: started, version 2.82 cachesize 150
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: DNS service limited to local subnets
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack ipset auth DNSSEC no-ID loop-detect inotify dumpfile
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq-dhcp[2315]: DHCP, IP range 192.168.100.100 -- 192.168.100.249, lease time 12h
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: using only locally-known addresses for domain test
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: using only locally-known addresses for domain onion
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: using only locally-known addresses for domain localhost
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: using only locally-known addresses for domain local
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: using only locally-known addresses for domain invalid
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: using only locally-known addresses for domain bind
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: using only locally-known addresses for domain use-application-dns.net
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: using only locally-known addresses for domain lan
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: using nameserver 127.0.0.1#5353
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: using 1 more local addresses
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: read /etc/hosts - 4 addresses
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq[2315]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Sat Aug 29 15:35:40 2020 daemon.info dnsmasq-dhcp[2315]: read /etc/ethers - 0 addresses
Sat Aug 29 15:35:40 2020 user.notice omr-bypass: OMR-ByPass is running
Sat Aug 29 15:35:40 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sat Aug 29 15:35:41 2020 user.notice omr-bypass: Restart dnsmasq...
Sat Aug 29 15:35:41 2020 daemon.info dnsmasq[2315]: exiting on receipt of SIGTERM
Sat Aug 29 15:35:41 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
[... 642 lines cut ...]
Sat Aug 29 15:50:52 2020 daemon.info dsvpn[5929]: Connected
Sat Aug 29 15:50:54 2020 user.notice mptcp: Reloading mptcp config due to ifup of omrvpn (tun0)
Sat Aug 29 15:50:59 2020 user.notice post-tracking-post-tracking: Tunnel up : Replace default route by 10.255.251.1 dev tun0
Sat Aug 29 15:51:04 2020 daemon.err /usr/bin/ss-redir[5882]: remote recv: Operation timed out
Sat Aug 29 15:51:04 2020 daemon.err /usr/bin/ss-redir[5883]: remote recv: Operation timed out
Sat Aug 29 15:51:12 2020 user.notice post-tracking-post-tracking: Tunnel up disable use of ShadowSocks for UDP
Sat Aug 29 15:51:12 2020 user.notice Shadowsocks: Rules DOWN
Sat Aug 29 15:51:12 2020 daemon.err /usr/bin/ss-redir[5883]: remote recv: Operation timed out
Sat Aug 29 15:51:12 2020 user.notice Shadowsocks: Rules UP
Sat Aug 29 15:51:12 2020 user.notice Shadowsocks: Reload omr-bypass rules
Sat Aug 29 15:51:12 2020 user.notice omr-bypass: Starting OMR-ByPass...
Sat Aug 29 15:51:13 2020 user.notice omr-bypass: Reload dnsmasq...
Sat Aug 29 15:51:13 2020 daemon.info dnsmasq[8164]: read /etc/hosts - 4 addresses
Sat Aug 29 15:51:13 2020 daemon.info dnsmasq[8164]: read /tmp/hosts/dhcp.cfg01411c - 3 addresses
Sat Aug 29 15:51:13 2020 daemon.info dnsmasq-dhcp[8164]: read /etc/ethers - 0 addresses
Sat Aug 29 15:51:13 2020 user.notice omr-bypass: OMR-ByPass is running
Sat Aug 29 15:51:32 2020 daemon.err /usr/bin/ss-redir[5883]: remote recv: Operation timed out
Sat Aug 29 15:51:35 2020 daemon.err /usr/bin/ss-redir[5882]: remote recv: Operation timed out
Sat Aug 29 15:51:39 2020 daemon.err uhttpd[9176]: luci: accepted login on / for root from 192.168.100.246

## Specifications

  - OpenMPTCProuter version: (0.55.3)
  - OpenMPTCProuter VPS version: (0.1017 5.4.52-mptcp)
  - OpenMPTCProuter platform: (x86_64)

Ysurac, 2020-08-29:
Vu le log ce n'est pas que le VPN qui est down, shadowsocks semble avoir aussi du mal à joindre le VPS.
Le VPS est hébergé chez qui ?
Pourquoi DSVPN est utilisé ?
Il y a quoi comme connexion ?

AdrienDssy, 2020-08-29:
VPS hébergé chez OVHcloud, car j'ai vu que j'avais quasiment tout le temps VPN tunnel down (comme avec DSVPN) et justement je l'utilise car je vois "VPN tunnel down" ce qui rend down certains de mes services

Ysurac, 2020-08-29:
Oui, mais je vois beaucoup de ``` /usr/bin/ss-redir[5882]: remote recv: Operation timed out``` ce qui veut dire que le soucis n'est pas au niveau du VPN, mais de la communication entre le routeur et le VPS.
Il me faudrait les latences affichées dans l'onglet état de Système->OpenMPTCProuter et savoir le type de connexion (USB, routeurs ethernet, box de FAI, pigeon voyageur,...)

AdrienDssy, 2020-08-29:
[image]
Pour le gateway down c'est normal j'ai débrancher l'ADSL temporairement vu que je suis connecté dessus (et connexions directes sur des cartes réseaux (RJ45))

Ysurac, 2020-08-29:
Pour le master il est préférable d'avoir la connexion la plus stable (donc normalement l'ADSL, la 4G perdant forcément des paquets) et une latence de 122ms avec le VPS est assez importante.
OpenMPTCProuter est une nouvelle installation depuis la 0.55 ou une mise à jour ?

AdrienDssy, 2020-08-29:
L'ADSL à un débit très bas ce qui rendrais down la connexion entre le routeur et le VPS, la 4G prends bien la relève. L'installation est une mise a jour :) La latence est dû à la forte utilisation et la 4G qui est complètement saturée. Mais une antenne 4G est en attente de mise en service (qui est à peine à 300m de chez moi)

AdrienDssy, 2020-08-29:
[image]
Comme quoi le ping est assez variable mais la plus part du temps le ping est bas

AdrienDssy, 2020-08-30:
[image]
Des fois ça peux m'arriver d'avoir un ptit soucis sur ça mais c'est pas bien grave je pense

Ysurac, 2020-08-30:
Le VPN tunnel DOWN est gênant, le "Can't contact Server Admin Script" moins, mais sûrement lié. Au final il y a bien des difficultés pour joindre le serveur.
Tu ne m'as pas répondu si il s'agit d'une nouvelle installation ou de la mise à jour d'une ancienne ?

AdrienDssy, 2020-08-30:
Tout est une mise à jour (VPS et routeur)

Ysurac, 2020-08-30:
Tu peux essayer une installation depuis 0 du routeur ou dans Réseau->MPTCP de mettre MPTCP sheduler utilisé BLEST et cubic en TCP congestion control. Tu peux aussi augmenter le TCP syn retries à 3.

AdrienDssy, 2020-08-30:
J'ai refait la config de zéro sans réimporté l'ancienne et j'ai toujours pour l'instant des VPN tunnel down

Ysurac, 2020-08-30:
Garde Glorytun TCP en VPN.
Vérifie les clefs par rapport à celles sur le VPS.
Et si elles sont bonnes, donne le journal système et la configuration complète, présente dans Système->OpenMPTCProuter dans le dernier onglet (en enlevant les clefs).

AdrienDssy, 2020-08-30:
là ça à l'air de s'être stabilisé, je surveille le tout (j'utilise OpenVPN)

AdrienDssy, 2020-08-30:
Pour l'instant c'est pas gagner ça change rien même, cela viendrais du VPS ?

Ysurac, 2020-08-30:
C'est que les routeurs répondent parfois trop lentement.
Tu utilises un virtualiseur ?

AdrienDssy, 2020-08-30:
Non directement sur une machine avec un pentium

Ysurac, 2020-08-30:
Une carte réseau ou plusieurs ?
Et sinon toujours ma demande précédente.

AdrienDssy, 2020-08-30:
J'ai la carte réseau intégré à la carte mère (4G Box) et une carte réseau PCI (Réseau local et l'ADSL)

AdrienDssy, 2020-08-30:
J'avais essayer aussi ce que tu m'a dis au dessus mais ça n'a pas l'air de faire grand chose :/

Ysurac, 2020-08-30:
> Garde Glorytun TCP en VPN.
> Vérifie les clefs par rapport à celles sur le VPS.
> Et si elles sont bonnes, donne le journal système et la configuration complète, présente dans Système->OpenMPTCProuter dans le dernier onglet (en enlevant les clefs).

AdrienDssy, 2020-08-31:
> dhcp.@dnsmasq[0]=dnsmasq
dhcp.@dnsmasq[0].domainneeded='1'
dhcp.@dnsmasq[0].localise_queries='1'
dhcp.@dnsmasq[0].rebind_protection='1'
dhcp.@dnsmasq[0].rebind_localhost='1'
dhcp.@dnsmasq[0].local='/lan/'
dhcp.@dnsmasq[0].domain='lan'
dhcp.@dnsmasq[0].expandhosts='1'
dhcp.@dnsmasq[0].authoritative='1'
dhcp.@dnsmasq[0].readethers='1'
dhcp.@dnsmasq[0].leasefile='/tmp/dhcp.leases'
dhcp.@dnsmasq[0].localservice='1'
dhcp.@dnsmasq[0].server='127.0.0.1#5353' '/lan/' '/use-application-dns.net/'
dhcp.@dnsmasq[0].noresolv='1'
dhcp.@dnsmasq[0].nonegcache='1'
dhcp.@dnsmasq[0].rebind_domain='plex.direct'
dhcp.@dnsmasq[0].ipset='/googlevideo.com/omr_dscp-cs4,omr_dscp6-cs4' '/nflxvideo.net/omr_dscp-cs4,omr_dscp6-cs4' '/s3.ll.dash.row.aiv-cdn.net/omr_dscp-cs4,omr_dscp6-cs4' '/d25xi40x97liuc.cloudfront.net/omr_dscp-cs4,omr_dscp6-cs4' '/aiv-delivery.net/omr_dscp-cs4,omr_dscp6-cs4' '/vevo.com/omr_dscp-cs4,omr_dscp6-cs4' '/audio-fa.scdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/deezer.com/omr_dscp-cs4,omr_dscp6-cs4' '/sndcdn.com/omr_dscp-cs4,omr_dscp6-cs4' '/last.fm/omr_dscp-cs4,omr_dscp6-cs4' '/v.redd.it/omr_dscp-cs4,omr_dscp6-cs4' '/ttvnw.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs4,omr_dscp6-cs4' '/googletagmanager.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleusercontent.com/omr_dscp-cs2,omr_dscp6-cs2' '/google.com/omr_dscp-cs2,omr_dscp6-cs2' '/fbcdn.net/omr_dscp-cs4,omr_dscp6-cs4,omr_dscp-cs2,omr_dscp6-cs2' '/akamaihd.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.net/omr_dscp-cs2,omr_dscp6-cs2' '/whatsapp.com/omr_dscp-cs2,omr_dscp6-cs2' '/googleapis.com/omr_dscp-cs2,omr_dscp6-cs2' '/1e100.net/omr_dscp-cs2,omr_dscp6-cs2' '/hwcdn.net/omr_dscp-cs2,omr_dscp6-cs2' '/download.qq.com/omr_dscp-cs1,omr_dscp6-cs1' '/steamcontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gs2.ww.prod.dl.playstation.net/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropboxstatic.com/omr_dscp-cs1,omr_dscp6-cs1' '/dropbox-dns.com/omr_dscp-cs1,omr_dscp6-cs1' '/log.getdropbox.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/drive-thirdparty.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/docs.googleusercontent.com/omr_dscp-cs1,omr_dscp6-cs1' '/gvt1.com/omr_dscp-cs1,omr_dscp6-cs1' '/mmg-fna.whatsapp.net/omr_dscp-cs1,omr_dscp6-cs1' '/upload.youtube.com/omr_dscp-cs1,omr_dscp6-cs1' '/upload.video.google.com/omr_dscp-cs1,omr_dscp6-cs1' '/windowsupdate.com/omr_dscp-cs1,omr_dscp6-cs1' '/update.microsoft.com/omr_dscp-cs1,omr_dscp6-cs1'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='100'
dhcp.lan.limit='150'
dhcp.lan.leasetime='12h'
dhcp.lan.force='1'
dhcp.lan.ra_slaac='1'
dhcp.lan.ra_flags='managed-config' 'other-config'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
dhcp.odhcpd=odhcpd
dhcp.odhcpd.maindhcp='0'
dhcp.odhcpd.leasefile='/tmp/hosts/odhcpd'
dhcp.odhcpd.leasetrigger='/usr/sbin/odhcpd-update'
dhcp.odhcpd.loglevel='4'
dhcp.@host[0]=host
dhcp.@host[0].name='adkynet'
dhcp.@host[0].dns='1'
dhcp.@host[0].mac='3C:D9:2B:0B:9F:E2'
dhcp.@host[0].ip='192.168.100.139'
dropbear.@dropbear[0]=dropbear
dropbear.@dropbear[0].PasswordAuth='on'
dropbear.@dropbear[0].RootPasswordAuth='on'
dropbear.@dropbear[0].Port='22'
dscp.@classify[0]=classify
dscp.@classify[0].direction='both'
dscp.@classify[0].proto='icmp'
dscp.@classify[0].class='cs5'
dscp.@classify[0].comment='ICMP'
dscp.@classify[1]=classify
dscp.@classify[1].direction='both'
dscp.@classify[1].proto='udp'
dscp.@classify[1].class='cs5'
dscp.@classify[1].src_port='53,123,5353'
dscp.@classify[1].comment='DNS udp and NTP'
dscp.@classify[2]=classify
dscp.@classify[2].direction='both'
dscp.@classify[2].proto='tcp'
dscp.@classify[2].class='cs5'
dscp.@classify[2].src_port='53,5353'
dscp.@classify[2].comment='DNS tcp'
dscp.@classify[3]=classify
dscp.@classify[3].direction='both'
dscp.@classify[3].proto='tcp'
dscp.@classify[3].class='cs6'
dscp.@classify[3].dest_port='65001,65301,65011'
dscp.@classify[3].comment='OMR vpn'
dscp.@classify[4]=classify
dscp.@classify[4].direction='both'
dscp.@classify[4].proto='udp'
dscp.@classify[4].class='cs6'
dscp.@classify[4].dest_port='65001'
[... 1351 lines cut ...]
upnpd.config.enabled='0'
upnpd.config.enable_natpmp='1'
upnpd.config.enable_upnp='1'
upnpd.config.secure_mode='1'
upnpd.config.log_output='0'
upnpd.config.download='1024'
upnpd.config.upload='512'
upnpd.config.internal_iface='lan'
upnpd.config.port='5000'
upnpd.config.upnp_lease_file='/var/run/miniupnpd.leases'
upnpd.config.igdv1='1'
upnpd.@perm_rule[0]=perm_rule
upnpd.@perm_rule[0].action='allow'
upnpd.@perm_rule[0].ext_ports='1024-65535'
upnpd.@perm_rule[0].int_addr='0.0.0.0/0'
upnpd.@perm_rule[0].int_ports='1024-65535'
upnpd.@perm_rule[0].comment='Allow high ports'
upnpd.@perm_rule[1]=perm_rule
upnpd.@perm_rule[1].action='deny'
upnpd.@perm_rule[1].ext_ports='0-65535'
upnpd.@perm_rule[1].int_addr='0.0.0.0/0'
upnpd.@perm_rule[1].int_ports='0-65535'
upnpd.@perm_rule[1].comment='Default deny'
vnstat.@vnstat[0]=vnstat
vnstat.@vnstat[0].interface='eth4' 'eth3'

Ysurac, 2020-08-31:
Je vois pas de soucis dans la configuration.
Que donne un ```ping -c 20 <vpsip>``` depuis le routeur ?

AdrienDssy, 2020-08-31:
root@OpenMPTCProuter:~# ping -c 20 <vpsip>
-ash: syntax error: unexpected newline

AdrienDssy, 2020-08-31:
root@OpenMPTCProuter:~# ping -c 20 
PING () 56(84) bytes of data.
64 bytes from : icmp_seq=1 ttl=51 time=45.0 ms
64 bytes from : icmp_seq=2 ttl=51 time=75.4 ms
64 bytes from : icmp_seq=3 ttl=51 time=45.6 ms
64 bytes from : icmp_seq=4 ttl=51 time=67.6 ms
64 bytes from : icmp_seq=5 ttl=51 time=45.4 ms
64 bytes from : icmp_seq=6 ttl=51 time=105 ms
64 bytes from : icmp_seq=7 ttl=51 time=45.7 ms
64 bytes from : icmp_seq=9 ttl=51 time=60.7 ms
64 bytes from : icmp_seq=10 ttl=51 time=52.0 ms
64 bytes from : icmp_seq=11 ttl=51 time=45.1 ms
64 bytes from : icmp_seq=12 ttl=51 time=77.3 ms
64 bytes from : icmp_seq=13 ttl=51 time=44.8 ms
64 bytes from : icmp_seq=14 ttl=51 time=62.8 ms
64 bytes from : icmp_seq=15 ttl=51 time=53.3 ms
64 bytes from : icmp_seq=16 ttl=51 time=45.0 ms
64 bytes from : icmp_seq=17 ttl=51 time=44.9 ms
64 bytes from : icmp_seq=18 ttl=51 time=157 ms
64 bytes from : icmp_seq=19 ttl=51 time=80.3 ms
64 bytes from : icmp_seq=20 ttl=51 time=45.5 ms

--- ping statistics ---
20 packets transmitted, 19 received, 5% packet loss, time 19036ms
rtt min/avg/max/mdev = 44.764/63.077/157.218/27.521 ms

Ysurac, 2020-08-31:
Quelques paquets perdu mais ça ne devrait pas empêcher le fonctionnement.
Que donne ```omr-tracebox eth4``` sur le routeur via SSH ?

AdrienDssy, 2020-08-31:
> root@OpenMPTCProuter:~# omr-tracebox eth4
tracebox to 130.104.228.140 (multipath-tcp.org): 64 hops max
1: *
2: *
3: *
4: *
5: *
6: *
7: *
8: *
9: *
10: *
11: *
12: *
13: *
14: *
15: *
16: *
17: *
18: *
19: *
20: *
21: *
22: *
23: *
24: *
25: *
26: *
27: *
28: *
29: *
30: *
31: *
32: *
33: *
34: *
35: *
36: *
37: *
38: *
39: *
40: *
41: *
42: *
43: *
44: *
45: *
46: *
47: *
48: *
49: *
50: *
51: *
52: *
53: *
54: *
55: *
56: *
57: *
58: *
59: *
60: *
61: *
62: *
63: *
64: *
root@OpenMPTCProuter:~#

Ysurac, 2020-08-31:
Ah oui c'est pas très causant.
La 4G box c'est quoi ?
Le soucis se pose si uniquement l'ADSL est présent ?

AdrienDssy, 2020-08-31:
Le soucis ce pose dans les deux cas (qu'il y est l'ADSL ou non (mais plus quand y'a pas l'ADSL)) la 4G Box c'est une box Huawei B258-S (Avec une antenne 4G)

Ysurac, 2020-08-31:
Si dans System->OpenMPTCProuter, onglet paramètres avancés, "Selection de l'interface maître" est mis sur "Pas de changement" ça change quelque chose ?

AdrienDssy, 2020-08-31:
J'ai mis cet option depuis longtemps autant dire que ça change pas grand chose du coup :(

Ysurac, 2020-08-31:
Ce n'était pas le cas dans la configuration, d'où l'essai.

AdrienDssy, 2020-08-31:
J'ai refait le changement du coup, je surveille pour l'instant ça à l'air bon mais parlons pas trop vite

AdrienDssy, 2020-08-31:
Toujours du VPN tunnel down mais ça dure moins de 20 sec

Ysurac, 2020-08-31:
Comment ça ? Il passe up/down dans la page status ?
Il faudrait voir dans Etat->Journal système il y a quelque chose concernant glorytun et OMRVPN.
Egalement un test, via SSH sur le routeur: ```ping -c 20 10.255.255.1```, ça ping l'ip du VPN côté serveur (c'est que fait la page status).

AdrienDssy, 2020-08-31:
Bon ça down toujours autant :(

AdrienDssy, 2020-08-31:
root@OpenMPTCProuter:~# ping -c 20 10.255.255.1
PING 10.255.255.1 (10.255.255.1) 56(84) bytes of data.
64 bytes from 10.255.255.1: icmp_seq=1 ttl=64 time=16922 ms
64 bytes from 10.255.255.1: icmp_seq=2 ttl=64 time=16786 ms
64 bytes from 10.255.255.1: icmp_seq=3 ttl=64 time=16311 ms
64 bytes from 10.255.255.1: icmp_seq=4 ttl=64 time=15987 ms
64 bytes from 10.255.255.1: icmp_seq=5 ttl=64 time=15675 ms
64 bytes from 10.255.255.1: icmp_seq=6 ttl=64 time=15292 ms
64 bytes from 10.255.255.1: icmp_seq=7 ttl=64 time=15193 ms
64 bytes from 10.255.255.1: icmp_seq=8 ttl=64 time=14961 ms
64 bytes from 10.255.255.1: icmp_seq=9 ttl=64 time=14626 ms
64 bytes from 10.255.255.1: icmp_seq=10 ttl=64 time=15390 ms
64 bytes from 10.255.255.1: icmp_seq=11 ttl=64 time=16054 ms
64 bytes from 10.255.255.1: icmp_seq=12 ttl=64 time=16004 ms
64 bytes from 10.255.255.1: icmp_seq=13 ttl=64 time=15923 ms
64 bytes from 10.255.255.1: icmp_seq=14 ttl=64 time=16303 ms
64 bytes from 10.255.255.1: icmp_seq=15 ttl=64 time=16177 ms
64 bytes from 10.255.255.1: icmp_seq=16 ttl=64 time=17440 ms
64 bytes from 10.255.255.1: icmp_seq=17 ttl=64 time=18344 ms
64 bytes from 10.255.255.1: icmp_seq=18 ttl=64 time=18426 ms
64 bytes from 10.255.255.1: icmp_seq=19 ttl=64 time=19058 ms
64 bytes from 10.255.255.1: icmp_seq=20 ttl=64 time=19307 ms

--- 10.255.255.1 ping statistics ---
20 packets transmitted, 20 received, 0% packet loss, time 19317ms
rtt min/avg/max/mdev = 14626.059/16508.991/19307.445/1320.194 ms, pipe 17

AdrienDssy, 2020-08-31:
> Mon Aug 31 16:03:58 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:04:06 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:04:12 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:04:32 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:04:48 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:04:49 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:05:06 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:05:11 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:05:28 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:05:29 2020 user.notice post-tracking-post-tracking: wan1 (eth4) switched off because ping from 192.168.10.114 error (114.114.115.115,4.2.2.1)
Mon Aug 31 16:05:29 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.10.1 dev eth4
Mon Aug 31 16:05:40 2020 user.notice post-tracking-post-tracking: wan1 (eth4) switched up
Mon Aug 31 16:05:40 2020 user.notice post-tracking-post-tracking: Set server VPS (54.38.188.175) default route 54.38.188.175 nexthop via 192.168.10.1 dev eth4 weight 1
Mon Aug 31 16:05:47 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:05:48 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:05:52 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:05:57 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:06:03 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:06:06 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:06:07 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:06:10 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:06:11 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:06:13 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:06:18 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:06:23 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:06:28 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:06:33 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:06:34 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:06:37 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:06:40 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:06:46 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:06:53 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:07:04 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:07:09 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:07:22 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:07:37 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:07:39 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:07:40 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:07:48 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:07:48 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:08:45 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:08:59 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:09:00 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:09:09 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:09:14 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:10:03 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:10:24 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:10:40 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:10:57 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 16:11:27 2020 user.notice post-tracking-post-tracking: wan1 (eth4) switched off because ping from 192.168.10.114 error (9.9.9.9,1.1.1.1)
Mon Aug 31 16:11:27 2020 user.notice post-tracking-post-tracking: Delete default route via 192.168.10.1 dev eth4
Mon Aug 31 16:11:40 2020 user.notice post-tracking-post-tracking: wan1 (eth4) switched up
Mon Aug 31 16:11:40 2020 user.notice post-tracking-post-tracking: Set server VPS (54.38.188.175) default route 54.38.188.175 nexthop via 192.168.10.1 dev eth4 weight 1
Mon Aug 31 16:12:42 2020 daemon.err /usr/bin/ss-redir[1790]: remote recv: Operation timed out
Mon Aug 31 16:13:18 2020 daemon.err /usr/bin/ss-redir[1789]: server recv: Operation timed out
[... cut ...]
Mon Aug 31 17:30:04 2020 daemon.info dnsmasq-dhcp[25737]: DHCPACK(eth0) 192.168.100.240 3c:d9:2b:0b:9f:ea ILOCZ21410C80
Mon Aug 31 17:30:09 2020 daemon.err /usr/bin/ss-redir[1789]: server recv: Operation timed out
Mon Aug 31 17:30:30 2020 daemon.err /usr/bin/ss-redir[1790]: server recv: Operation timed out
Mon Aug 31 17:30:43 2020 daemon.err /usr/bin/ss-redir[1789]: remote recv: Operation timed out
Mon Aug 31 17:32:03 2020 daemon.err /usr/bin/ss-redir[1790]: server recv: Operation timed out
Mon Aug 31 17:32:11 2020 daemon.err /usr/bin/ss-redir[1790]: server recv: Operation timed out

AdrienDssy, 2020-08-31:
Actuellement j'ai que du VPN tunnel down

AdrienDssy, 2020-08-31:
J'ai changé par openVPN et j'ai plus le message pour l'instant, c'est bizarre cet histoire

Ysurac, 2020-08-31:
Ah oui avec un ping pareil c'est pas étonnant. Le ping dans le VPN est une véritable horreur.
Je vois dans le log que eth4, donc le wan1, est totalement instable. Pareil pour le wan2 d'ailleurs. OMR-Tracker a des difficultés pour pinger des IPs sur internet depuis ces interfaces.
Je pense pas que ça soit lié au VPS du coup.

AdrienDssy, 2020-08-31:
ça serais lié  à quoi ?

Ysurac, 2020-08-31:
Ca semble lié aux connexion. C'est vert car ça marche, mais par moment elles ne permettent pas d'aller sur internet.
Tu n'as pas de conflit d'IP sur ton réseau avec une autre box/machine ayant l'IP d'une box ?

AdrienDssy, 2020-08-31:
Non les deux box on une ip différente (192.168.1.1 box adsl, 192.168.10.1 côté 4G Box)

AdrienDssy, 2020-08-31:
> root@OpenMPTCProuter:~# ping -c 20 10.255.255.1
> PING 10.255.255.1 (10.255.255.1) 56(84) bytes of data.
> 64 bytes from 10.255.255.1: icmp_seq=1 ttl=64 time=538 ms
> 64 bytes from 10.255.255.1: icmp_seq=2 ttl=64 time=278 ms
> 64 bytes from 10.255.255.1: icmp_seq=3 ttl=64 time=467 ms
> 64 bytes from 10.255.255.1: icmp_seq=4 ttl=64 time=910 ms
> 64 bytes from 10.255.255.1: icmp_seq=5 ttl=64 time=1078 ms
> 64 bytes from 10.255.255.1: icmp_seq=6 ttl=64 time=768 ms
> 64 bytes from 10.255.255.1: icmp_seq=7 ttl=64 time=719 ms
> 64 bytes from 10.255.255.1: icmp_seq=8 ttl=64 time=898 ms
> 64 bytes from 10.255.255.1: icmp_seq=9 ttl=64 time=945 ms
> 64 bytes from 10.255.255.1: icmp_seq=10 ttl=64 time=1618 ms
> 64 bytes from 10.255.255.1: icmp_seq=11 ttl=64 time=1324 ms
> 64 bytes from 10.255.255.1: icmp_seq=12 ttl=64 time=323 ms
> 64 bytes from 10.255.255.1: icmp_seq=13 ttl=64 time=903 ms
> 64 bytes from 10.255.255.1: icmp_seq=14 ttl=64 time=1620 ms
> 64 bytes from 10.255.255.1: icmp_seq=15 ttl=64 time=971 ms
> 64 bytes from 10.255.255.1: icmp_seq=16 ttl=64 time=1259 ms
> 64 bytes from 10.255.255.1: icmp_seq=17 ttl=64 time=1270 ms
> 64 bytes from 10.255.255.1: icmp_seq=18 ttl=64 time=300 ms
> 64 bytes from 10.255.255.1: icmp_seq=19 ttl=64 time=707 ms
> 64 bytes from 10.255.255.1: icmp_seq=20 ttl=64 time=1427 ms
> 
> --- 10.255.255.1 ping statistics ---
> 20 packets transmitted, 20 received, 0% packet loss, time 19083ms
> rtt min/avg/max/mdev = 278.110/916.085/1619.525/404.685 ms, pipe 2
> root@OpenMPTCProuter:~#

Je viens de le refaire et le ping est beaucoup plus bas

Ysurac, 2020-08-31:
oui ça j'ai vu, mais je voulais dire par rapport à autre chose sur ton réseau. Voir aussi la connectique, les switchs, etc...
Le ping est toujours beaucoup trop haut.

kevin39, 2020-08-31:
Penser au conflit de mac aussi si des VM sont utilisées... (un clone de VM par exemple qui aurait conservé une mac)

AdrienDssy, 2020-08-31:
J'ai que du matos fait pour l'environnement des datacenters sauf le routeur en lui même

AdrienDssy, 2020-08-31:
> Penser au conflit de mac aussi si des VM sont utilisées... (un clone de VM par exemple qui aurait conservé une mac)

Pas de clone de VM fait sur le réseau

AdrienDssy, 2020-08-31:
Bizarrement tout est bon depuis 10 min (mais pour combien de temps)

kevin39, 2020-08-31:
Ca fait un moment que je suis cette discussion : 

Quelques idées en vrac : 
- La chaleur ? Tes equipements n'ont pas chaud ?
- Saturation : tes liens saturent pas en permanence ? (streaming, download, hosting ?). 
- Tes box sont connectées directes à ton routeur, c'est ça ? Sans switch ? Si tu isole ton routeur et tes box (donc sortir ton routeur du reste du réseau), tu as tjrs les soucis ?
- Change les câbles si tu as des câbles pas de gamme, on sait jamais

kevin39, 2020-08-31:
Et t'es sur de la fiabilité hardware de ton routeur ? Remplace le par une VM pour tester par exemple

Ysurac, 2020-08-31:
oui, même questions que kevin39

On peut tester les connexions, en faisant un ```ping -B -I eth4 8.8.8.8``` et  ```ping -B -I eth3 8.8.8.8``` il y a certainement des paquets perdus sur chaque.

AdrienDssy, 2020-08-31:
> Et t'es sur de la fiabilité hardware de ton routeur ? Remplace le par une VM pour tester par exemple ou un raspberry

Bah du ASUS c'est fiable après pdv réseau c'est fiable de ouf jamais de soucis en revanche avec un RPI c'est pas très pratique vu que ça faisais du conflict avec le lan et le wan donc faut acheter plusieurs adaptateurs ethernet

kevin39, 2020-08-31:
Autre chose : Je sais pas si le sujet a été abordé mais quid de la puissance CPU du routeur ? C'est pas lui qui sature ?

kevin39, 2020-08-31:
> donc faut acheter plusieurs adaptateurs ethernet
Mets un switch.

AdrienDssy, 2020-08-31:
> > donc faut acheter plusieurs adaptateurs ethernet
> 
> Mets un switch.

Justement c'est ce que je faisais mais ça fait un conflict entre le lan et le wan justement

AdrienDssy, 2020-08-31:
> Autre chose : Je sais pas si le sujet a été abordé mais quid de la puissance CPU du routeur ? C'est pas lui qui sature ?

Avec un LA de 0.31 ça m'étonnerais que le CPU sois la source du problème

kevin39, 2020-08-31:
> Justement c'est ce que je faisais mais ça fait un conflict entre le lan et le wan justement

Ce n'est pas normal. Ca montre bien que ton archi à un problème.
Cherche pas plus loin, tu as un soucis réseau chez toi. Il faut l'isolé. Je pense que @Ysurac va confirmer ?

AdrienDssy, 2020-08-31:
--- 8.8.8.8 ping statistics ---
246 packets transmitted, 197 received, 19.9187% packet loss, time 246191ms
rtt min/avg/max/mdev = 38.748/150.651/331.723/53.912 ms
root@OpenMPTCProuter:~#
ETH 3

--- 8.8.8.8 ping statistics ---
47 packets transmitted, 34 received, 27.6596% packet loss, time 46263ms
rtt min/avg/max/mdev = 43.561/108.486/164.852/30.744 ms
root@OpenMPTCProuter:~#
ETH4

[9 more replies]

closed 2020-08-31 by AdrienDssy