Ysurac/openmptcprouter #1187: How to configure WAN failback ?

NyKo24, 2020-08-29:
## Expected Behavior
So I would like to use openmptcprouter for failover and not aggregator.
The main connection is my 4G and if I lost network, use ADSL connection with transparence.

## Current Behavior
Now, I disable VPS, shadowsocks, I put my 4G WAN (wan2) on "master", but the traffic output on my ADSL wan (wan1). So I disabled interface wan1 to force traffic on my wan2.

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)
I wan two WAN connections : 
* ADSL 2mbps
* 4G+ LTE ~ 100mbps.

So aggregation is not powerful and I lose bandwidth via VPS etc.

## Specifications

  - OpenMPTCProuter version: 0.55.1
  - OpenMPTCProuter VPS version: 0.55.1
  - OpenMPTCProuter platform: RPI3

kevin39, 2020-08-29:
Hi,
Put your WAN2 as "backup" instead of "active" (network -> MTCP).

Be careful, "backup" mode is "copie de sauvegarde" (!!!) in french version.

nbordes-treezor, 2020-08-29:
That's all ? ;) 
* WAN 1 is my ADSL => backup
* WAN 2 is my 4G => active

kevin39, 2020-08-29:
If everything is green in the status page, that's all. Just disconnect 4G and it may works :)

NyKo24, 2020-08-29:
That's all ? ;)

WAN 1 is my ADSL => backup
WAN 2 is my 4G => active

NyKo24, 2020-08-29:
Okay thank's, I will try it.

NyKo24, 2020-08-29:
it's work ! Thank's

closed 2020-08-29 by NyKo24