Ysurac/openmptcprouter #1189: Can't access amazon.fr

TanamuR, 2020-08-29:
## Expected Behavior
access amazon.fr website

## Current Behavior
can't access 'établissement de la connection sécurisée'........
## Possible Solution
did add an entry in OMR bypass and reboot OMR. It did work for a few mins only.
ALso some flood in system-log : usr/bin/ss-redir[30943]: server recv: Operation timed out(30943 & 30944).
Please not that these numbers change at each reboot of omr 

## Steps to Reproduce the Problem

 ?

## Context (Environment)
OMR running on Virtualbox. Running fine since ages.
VPS on scaleway

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: <--x86_64)-->

Sat Aug 29 20:20:37 2020 daemon.err /usr/bin/ss-redir[30943]: server recv: Operation timed out
Sat Aug 29 20:20:39 2020 daemon.err /usr/bin/ss-redir[30943]: server recv: Operation timed out
Sat Aug 29 20:20:45 2020 daemon.err /usr/bin/ss-redir[30944]: server recv: Operation timed out
Sat Aug 29 20:20:45 2020 daemon.err /usr/bin/ss-redir[30943]: server recv: Operation timed out
Sat Aug 29 20:20:45 2020 daemon.err /usr/bin/ss-redir[30943]: server recv: Operation timed out
Sat Aug 29 20:20:45 2020 daemon.err /usr/bin/ss-redir[30943]: server recv: Operation timed out
Sat Aug 29 20:20:45 2020 daemon.err /usr/bin/ss-redir[30944]: server recv: Operation timed out
Sat Aug 29 20:20:45 2020 daemon.err /usr/bin/ss-redir[30944]: server recv: Operation timed out
Sat Aug 29 20:20:47 2020 daemon.err /usr/bin/ss-redir[30944]: server recv: Operation timed out
Sat Aug 29 20:20:47 2020 daemon.err /usr/bin/ss-redir[30944]: server recv: Operation timed out
Sat Aug 29 20:20:48 2020 daemon.err /usr/bin/ss-redir[30943]: server recv: Operation timed out
Sat Aug 29 20:20:51 2020 daemon.err /usr/bin/ss-redir[30944]: server recv: Operation timed out
Sat Aug 29 20:20:53 2020 daemon.err /usr/bin/ss-redir[30944]: server recv: Operation timed out
Sat Aug 29 20:20:58 2020 daemon.err /usr/bin/ss-redir[30944]: server recv: Operation timed out
Sat Aug 29 20:20:59 2020 daemon.err /usr/bin/ss-redir[30943]: server recv: Operation timed out
Sat Aug 29 20:20:59 2020 daemon.err /usr/bin/ss-redir[30943]: server recv: Operation timed out

Ysurac, 2020-08-29:
What are the version of OpenMTPCProuter and VPS ?
VPS was rebooted after last update ?

TanamuR, 2020-08-29:
oups   0.55.3 and 0.1017
Yes it was rebooted. I am doing it once again to be on the safe side.

TanamuR, 2020-08-29:
Yes it was (last command was reboot) Did it again, no change.
But there is something that i find strange. When loging TTY, welcome message is Welcome on Debian Stretch (GNU/Linux 5.4.52-mptcp x86_64 )
I did run a dist-upgrade a couple of days ago to upgrade to buster. Upgrade went successfull.
Dont know if there is any link.

Ysurac, 2020-08-29:
The script can only run on Debian 10, if it's a Debian 9 then it's upgraded by the script.
Can you check on the VPS if there is errors in ```/var/log/daemon.log``` ?

TanamuR, 2020-08-29:
root@scw-ab2db0:~# grep error /var/log/daemon.log
Aug 23 06:25:01 scw-ab2db0 systemd[15454]: gpgconf: running /usr/bin/gpg-agent failed (exitcode=2): General error
Aug 23 06:25:01 scw-ab2db0 systemd[15454]: gpgconf: fatal error (exit status 1)
Aug 24 06:25:01 scw-ab2db0 systemd[1796]: gpgconf: running /usr/bin/gpg-agent failed (exitcode=2): General error
Aug 24 06:25:01 scw-ab2db0 systemd[1796]: gpgconf: fatal error (exit status 1)
Aug 25 06:25:01 scw-ab2db0 systemd[22376]: gpgconf: running /usr/bin/gpg-agent failed (exitcode=2): General error
Aug 25 06:25:01 scw-ab2db0 systemd[22376]: gpgconf: fatal error (exit status 1)
Aug 25 17:53:39 scw-ab2db0 omr-admin.py[28190]:   File "/usr/local/lib/python3.7/dist-packages/starlette/middleware/errors.py", line 181, in __call__
Aug 25 17:53:39 scw-ab2db0 omr-admin.py[28190]:   File "/usr/local/lib/python3.7/dist-packages/starlette/middleware/errors.py", line 159, in __call__
Aug 25 17:54:18 scw-ab2db0 omr-admin.py[28190]:   File "/usr/local/lib/python3.7/dist-packages/starlette/middleware/errors.py", line 181, in __call__
Aug 25 17:54:18 scw-ab2db0 omr-admin.py[28190]:   File "/usr/local/lib/python3.7/dist-packages/starlette/middleware/errors.py", line 159, in __call__
Aug 25 17:55:31 scw-ab2db0 ovpn-tun0[677]: ERROR: Linux route delete command failed: external program exited with error status: 2
Aug 25 17:55:31 scw-ab2db0 ovpn-tun0[677]: Linux ip addr del failed: external program exited with error status: 2
Aug 26 06:25:02 scw-ab2db0 systemd[10133]: gpgconf: running /usr/bin/gpg-agent failed (exitcode=2): General error
Aug 26 06:25:02 scw-ab2db0 systemd[10133]: gpgconf: fatal error (exit status 1)
Aug 27 06:25:01 scw-ab2db0 systemd[5319]: gpgconf: running /usr/bin/gpg-agent failed (exitcode=2): General error
Aug 27 06:25:01 scw-ab2db0 systemd[5319]: gpgconf: fatal error (exit status 1)
Aug 28 06:25:02 scw-ab2db0 systemd[12266]: gpgconf: running /usr/bin/gpg-agent failed (exitcode=2): General error
Aug 28 06:25:02 scw-ab2db0 systemd[12266]: gpgconf: fatal error (exit status 1)
Aug 29 06:25:01 scw-ab2db0 systemd[1936]: gpgconf: running /usr/bin/gpg-agent failed (exitcode=2): General error
Aug 29 06:25:01 scw-ab2db0 systemd[1936]: gpgconf: fatal error (exit status 1)

TanamuR, 2020-08-29:
on ERROR
Aug 25 07:02:56 scw-ab2db0 ss-manager[702]:  2020-08-25 07:02:56 ERROR: bind: Address already in use
Aug 25 07:02:56 scw-ab2db0 ss-manager[702]:  2020-08-25 07:02:56 ERROR: Could not bind
Aug 25 07:02:56 scw-ab2db0 ss-manager[702]:  2020-08-25 07:02:56 ERROR: port is not available, please check.
Aug 25 16:42:09 scw-ab2db0 ss-manager[665]:  2020-08-25 16:42:09 ERROR: bind: Address already in use
Aug 25 16:42:09 scw-ab2db0 ss-manager[665]:  2020-08-25 16:42:09 ERROR: Could not bind
Aug 25 16:42:09 scw-ab2db0 ss-manager[665]:  2020-08-25 16:42:09 ERROR: port is not available, please check.
Aug 25 17:53:39 scw-ab2db0 omr-admin.py[28190]: ERROR:    Exception in ASGI application
Aug 25 17:54:18 scw-ab2db0 omr-admin.py[28190]: ERROR:    Exception in ASGI application
Aug 25 17:55:31 scw-ab2db0 ovpn-tun0[677]: ERROR: Linux route delete command failed: external program exited with error status: 2
Aug 29 18:40:38 scw-ab2db0 ss-manager[673]:  2020-08-29 18:40:38 ERROR: bind: Address already in use
Aug 29 18:40:38 scw-ab2db0 ss-manager[673]:  2020-08-29 18:40:38 ERROR: Could not bind
Aug 29 18:40:38 scw-ab2db0 ss-manager[673]:  2020-08-29 18:40:38 ERROR: port is not available, please check.

Ysurac, 2020-08-29:
What do you have in ```/etc/shadowsocks-libev/manager.json``` ? (replace keys by "xxx" or something like that)

TanamuR, 2020-08-29:
"server": [
        "[::0]",
        "0.0.0.0"
    ],
    "port_key": {
        "65101": "xxxxxxxxxxxxxxxxxxxx",
        "65101": "xxxxxxxxxxxxxxxxxxxx",
    },
    "local_port": 1081,
    "mode": "tcp_and_udp",
    "timeout": 600,
    "method": "aes-256-gcm",
    "verbose": 0,
    "ipv6_first": true,
    "prefer_ipv6": false,
    "fast_open": false,
    "no_delay": false,
    "reuse_port": true,
    "mptcp": true,
    "ebpf": 0,
    "acl": "/etc/shadowsocks-libev/local.acl"

Ysurac, 2020-08-29:
All good here.
What is the result of ```systemctl -a | grep shadowsocks``` and ```netstat -laputen``` on the VPS (remove entries where IP is not "0.0.0.0") ?
Do you have IPv6 enabled on the router ?

TanamuR, 2020-08-29:
shadowsocks-libev-manager@manager.service                                                                                                                    loaded    active   running   Shadowsocks-Libev Custom Manager Service for mana                                                                             ger
  system-shadowsocks\x2dlibev\x2dmanager.slice                                                                                                                 loaded    active   active    system-shadowsocks\x2dlibev\x2dmanager.slice   

****************************************************************************************
Active Internet connections (servers and established)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       User       Inode      PID/Program name
tcp        0      0 0.0.0.0:65301           0.0.0.0:*               LISTEN      0          6695       593/openvpn
tcp        0      0 127.0.0.1:25            0.0.0.0:*               LISTEN      0          10397      1176/exim4
tcp        0      0 0.0.0.0:65401           0.0.0.0:*               LISTEN      0          7781       608/dsvpn
tcp        0      0 0.0.0.0:65500           0.0.0.0:*               LISTEN      0          10127      1191/python3
tcp        0      0 0.0.0.0:65222           0.0.0.0:*               LISTEN      0          9029       937/sshd
tcp        0      0 0.0.0.0:65001           0.0.0.0:*               LISTEN      0          6019       548/glorytun-tcp
tcp        0      0 0.0.0.0:65101           0.0.0.0:*               LISTEN      0          7948       697/ss-server
tcp        0      0 10.16.52.31:21107       143.204.176.85:443      ESTABLISHED 0          152268     697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:45490   TIME_WAIT   0          0          -
tcp        0      0 10.16.52.31:21715       68.183.169.156:81       TIME_WAIT   0          0          -
tcp        0      0 10.16.52.31:65101       185.228.228.137:9236    ESTABLISHED 0          11785      697/ss-server
tcp        0      0 10.16.52.31:65101       37.170.213.62:23501     TIME_WAIT   0          0          -
tcp        0      0 10.16.52.31:65101       185.228.228.137:32076   TIME_WAIT   0          0          -
tcp        0      0 10.16.52.31:65101       185.228.228.137:9235    ESTABLISHED 0          151591     697/ss-server
tcp        0      0 10.16.52.31:27815       149.202.222.5:443       ESTABLISHED 0          149611     697/ss-server
tcp        0      0 10.16.52.31:48815       157.240.221.18:443      ESTABLISHED 0          152090     697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:37393   ESTABLISHED 0          11743      697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:25944   ESTABLISHED 0          149785     697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:4008    ESTABLISHED 0          147402     697/ss-server
tcp        0      0 10.16.52.31:65101       37.170.213.62:23579     ESTABLISHED 0          151396     697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:18381   ESTABLISHED 0          151047     697/ss-server
tcp        0      0 10.16.52.31:64853       172.217.18.195:443      ESTABLISHED 0          149696     697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:29796   ESTABLISHED 0          129435     697/ss-server
tcp        0      0 10.16.52.31:65101       37.170.213.62:23638     ESTABLISHED 0          148813     697/ss-server
tcp        0      0 10.16.52.31:65101       37.170.213.62:23939     ESTABLISHED 0          12470      697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:37392   ESTABLISHED 0          148809     697/ss-server
tcp        0      0 10.16.52.31:65101       37.170.213.62:23585     TIME_WAIT   0          0          -
tcp        0      0 10.16.52.31:65101       185.228.228.137:43615   ESTABLISHED 0          151571     697/ss-server
tcp        0      0 10.16.52.31:54343       108.128.110.172:443     ESTABLISHED 0          53779      697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:34460   ESTABLISHED 0          152101     697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:22630   ESTABLISHED 0          148943     697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:37395   ESTABLISHED 0          148813     697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:38863   ESTABLISHED 0          129886     697/ss-server
tcp        0      0 10.16.52.31:19627       3.123.245.194:1883      ESTABLISHED 0          149282     697/ss-server
tcp        0      0 10.16.52.31:25489       104.18.0.211:443        ESTABLISHED 0          22149      697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:57483   ESTABLISHED 0          152261     697/ss-server
tcp        0      0 10.16.52.31:65101       37.170.213.62:23702     ESTABLISHED 0          149787     697/ss-server
tcp        0      0 10.16.52.31:44367       198.11.132.39:443       ESTABLISHED 0          148769     697/ss-server
tcp        0      0 10.16.52.31:65101       37.170.213.62:23593     TIME_WAIT   0          0          -
tcp        0      0 10.16.52.31:65101       185.228.228.137:34456   ESTABLISHED 0          136697     697/ss-server
tcp        0      0 10.16.52.31:65101       185.228.228.137:2586    ESTABLISHED 0          22163      697/ss-server
[... cut ...]
udp        0      0 0.0.0.0:65301           0.0.0.0:*                           0          6727       604/openvpn
udp        0      0 0.0.0.0:67              0.0.0.0:*                           104        9111       319/systemd-network
udp        0      0 0.0.0.0:68              0.0.0.0:*                           0          4333       346/dhclient
udp6       0      0 ::1:323                 :::*                                0          6935       715/chronyd
udp6       0      0 :::65101                :::*                                0          7950       697/ss-server
root@scw-ab2db0:~#

TanamuR, 2020-08-29:
I don't see IPV6 enables in advanced wizard

When looking at network>Interfaces
On lan   eth0 an IPV6 address is displaid
Protocol: Static address
Uptime: 3h 13m 6s
MAC: 08:00:27:88:21:C3
RX: 697.57 MB (1080467 Pkts.)
TX: 703.06 MB (1099479 Pkts.)
IPv4: 192.168.100.1/24
IPv6: fded:43d0:1ba7::1/60

TanamuR, 2020-08-29:
Looking twice it is looking like a mask for ipv6 address. I let you confirm.
clicking edit on lan and dhcp server>ipv6 settings  all is disabled

Ysurac, 2020-08-30:
The IPv6 displayed is the local IPv6 address. If IPv6 is disabled then it's not related to the issue.

How many connections do you have and what are they ? DSL, LTE, fiber, IP over Avian Carriers, ... ?

TanamuR, 2020-08-30:
I have 2 * 4G+ modems. I was also having an adsl connection but did remove it from aggregation as it was decreasing download speed.
One one free on on SFR (NRJ)

Ysurac, 2020-08-30:
If you do a ```curl -4 -L https://www.amazon.fr/``` on the VPS and on the router, you have some output ?

TanamuR, 2020-08-30:
on vps
root@scw-ab2db0:~# curl -4 -L https://www.amazon.fr/
Warning: Binary output can mess up your terminal. Use "--output -" to tell
Warning: curl to output it to your terminal anyway, or consider "--output
Warning: <FILE>" to save to a file.
root@scw-ab2db0:~#

TanamuR, 2020-08-30:
root@scw-ab2db0:~# curl -4 -L https://www.amazon.fr/
Warning: Binary output can mess up your terminal. Use "--output -" to tell
Warning: curl to output it to your terminal anyway, or consider "--output
Warning: <FILE>" to save to a file.
root@scw-ab2db0:~#

TanamuR, 2020-08-30:
On OMR
root@OpenMPTCProuter:~# root@scw-ab2db0:~# curl -4 -L https://www.amazon.fr/
-ash: root@scw-ab2db0:~#: not found
root@OpenMPTCProuter:~# Warning: Binary output can mess up your terminal. Use "-
-output -" to tell
-ash: Warning:: not found
root@OpenMPTCProuter:~# Warning: curl to output it to your terminal anyway, or c
onsider "--output
> Warning: <FILE>" to save to a file.
-ash: Warning:: not found
root@OpenMPTCProuter:~# root@scw-ab2db0:~#
-ash: root@scw-ab2db0:~#: not found
root@OpenMPTCProuter:~#

TanamuR, 2020-08-30:
sorry   had problems with the cut & paste

on OMR
curl -4 -L https://www.amazon.fr/
curl: (35) ssl_handshake returned - mbedTLS: (-0x0050) NET - Connection was reset by peer
root@OpenMPTCProuter:~#

TanamuR, 2020-08-30:
On VPS .....

Last login: Sun Aug 30 19:45:33 2020 from 37.167.97.52
root@scw-ab2db0:~# curl -4 -L https://www.amazon.fr/
Warning: Binary output can mess up your terminal. Use "--output -" to tell
Warning: curl to output it to your terminal anyway, or consider "--output
Warning: <FILE>" to save to a file.
root@scw-ab2db0:~#
root@scw-ab2db0:~# curl -4 -L https://www.amazon.fr/
<!DOCTYPE html>

<html class="a-no-js" lang="fr"><head>
<meta http-equiv="content-type" content="text/html; charset=UTF-8">
<meta charset="utf-8">
<meta http-equiv="X-UA-Compatible" content="IE=edge,chrome=1">
<title dir="ltr">Amazon CAPTCHA</title>
<meta name="viewport" content="width=device-width">
<link rel="stylesheet" href="https://images-na.ssl-images-amazon.com/images/G/01/AUIClients/AmazonUI-3c913031596ca78a3768f4e934b1cc02ce238101.secure.min._V1_.css">
<script>

if (true === true) {
    var ue_t0 = (+ new Date()),
        ue_csm = window,
        ue = { t0: ue_t0, d: function() { return (+new Date() - ue_t0); } },
        ue_furl = "fls-eu.amazon.fr",
        ue_mid = "A13V1IB3VIYZZH",
        ue_sid = (document.cookie.match(/session-id=([0-9-]+)/) || [])[1],
        ue_sn = "opfcaptcha.amazon.fr",
        ue_id = 'F0QREWKSZNBBH4RVGW8E';
}
</script>
</head>
<body>

<div class="a-container a-padding-double-large" style="min-width:350px;padding:44px 0 !important">

    <div class="a-row a-spacing-double-large" style="width: 350px; margin: 0 auto">

        <div class="a-row a-spacing-medium a-text-center"><i class="a-icon a-logo"></i></div>

        <div class="a-box a-alert a-alert-info a-spacing-base">
            <div class="a-box-inner">
                <i class="a-icon a-icon-alert"></i>
                <h4>Saisissez les caractères que vous voyez ci-dessous</h4>
                <p class="a-last">Désolés, il faut que nous nous assurions que vous n'êtes pas un robot. Pour obtenir les meilleurs résultats, veuillez vous assurer que votre navigateur accepte les cookies.</p>
                </div>
            </div>

            <div class="a-section">

                <div class="a-box a-color-offset-background">
                    <div class="a-box-inner a-padding-extra-large">

                        <form method="get" action="/errors/validateCaptcha" name="">
                            <input type=hidden name="amzn" value="xGVmoIE3vXven006gEZvEg==" /><input type=hidden name="amzn-r" value="&#047;" />
                            <div class="a-row a-spacing-large">
                                <div class="a-box">
                                    <div class="a-box-inner">
                                        <h4>Saisissez les caractères que vous voyez dans cette image :</h4>
                                        
                                        <div class="a-row a-text-center">
                                            [image]
                                        </div>
                                        <div class="a-row a-spacing-base">
                                            <div class="a-row">
                                                <div class="a-column a-span6">
                                                </div>
                                                <div class="a-column a-span8 a-span-last a-text-right">
                                                    <a onclick="window.location.reload()">Essayez une autre image</a>
                                                </div>
[... 43 lines cut ...]
           };
          </script>
          <noscript>
            [image]
          </noscript>
        </div>
    </div>
    <script>
    if (true === true) {
        var head = document.getElementsByTagName('head')[0],
            prefix = "https://images-eu.ssl-images-amazon.com/images/G/01/csminstrumentation/",
            elem = document.createElement("script");
        elem.src = prefix + "csm-captcha-instrumentation.min.js";
        head.appendChild(elem);

        elem = document.createElement("script");
        elem.src = prefix + "rd-script-6d68177fa6061598e9509dc4b5bdd08d.js";
        head.appendChild(elem);
    }
    </script>
</body></html>
root@scw-ab2db0:~# curl -4 -L https://www.amazon.fr/
Warning: Binary output can mess up your terminal. Use "--output -" to tell
Warning: curl to output it to your terminal anyway, or consider "--output
Warning: <FILE>" to save to a file.

Ysurac, 2020-08-30:
The problem seems to be really related to shadowsocks...
You can try to disable TCP fast open in System->OpenMPTCProuter, advanced settings tab.
obfuscation is disabled ? (should be)

TanamuR, 2020-08-30:
obfuscation is set to 
plugin : v2ray
type  : http

ipv4 tcp fast open value is 3 and not disabled

TanamuR, 2020-08-30:
but enable shadowshocks obufuscating is not ticked

Ysurac, 2020-08-30:
ok, check "Disable TCP Fast Open"

TanamuR, 2020-08-30:
ok did it, reboot OMR and seems working :) thanks. 
do u know what was going on ?

TanamuR, 2020-08-30:
No I still have the issue.    As before  seems the reebot is fixing it only for a couple of minutes only

Ysurac, 2020-08-31:
You can try to lower MTU on each WAN. In Network->Interface, edit interface and advanced parameters tab.

TanamuR, 2020-08-31:
did decrease eth0 wan1 wan2 to
1432
1400
1350
I dont see any improvement

TanamuR, 2020-08-31:
I did switch off 4g modems on wan 1 & 2 (free & sfr) and add my ADSL (free) on wan3
it seems to work.
I will swith on one of the 4G+ modem to see what is happening having bypass on wan3 for amazon and keep trying various combination

TanamuR, 2020-09-05:
I was not able find a way to solve it.
I did a fress install on router side (move from debian to win10, new virtaulbox, new install of OMV)
Looks better except that VPS cant connect to shadowshock.
I will probable scratch as well the scaleway install and create a new one.
I close the ticket as it is probably an issue coming from the various upgrade on both omp & vps and impossible to reproduce.
Thanks @Ysurac for the help

closed 2020-09-05 by TanamuR