Ysurac/openmptcprouter #1190: Problème shadowsocks et vpn

Anoratsu, 2020-08-30:
Bonjour,
Besoin d'aide car je n'y comprends plus rien !
J'ai donc mon rasp et mon vps qui sont à jour. Ma compagne est en télétravail, pour utiliser son PC, elle ouvre une session qui lance une VM. Dans cette VM, elle se connecte au VPN de l'entreprise (pulse secure). Une fois connectée, elle a accès à l'ensembles des logiciels pour travailler.

Depuis le confinement, cela a toujours fonctionné. Depuis la dernière mise à jour du VPS, on rencontre des soucis de connection. Problème de proxy...

En configuration shadowsocks + Glorytun UDP activés cela bug, c'était la config utilisée jusqu'à présent.
Après plusieurs tests j'en arrive à la conclusion suivante :
- Shadowsocks désactivé + aucun VPN : fonctionne
- Shadowsocks désactivé + Dead simple VPN : fonctionne
- Dans les autres situations : cela ne fonctionne pas

J'ai du mal à comprendre pourquoi ça ne fonctionne plus. Conflit de proxy ? bug de maj ?
Est ce que shadowsocks OFF + dead simple ON permet de profiter de l'agrégation ? Pas de manière optimale j'imagine.
Un moyen de laisser tout activer et de contourner le système pour son PC ? Sachant qu'omr bypass n'y fait rien...

Merci d'avance pour l'aide.

Ysurac, 2020-08-30:
En utilisant le template cela aiderait, j'ai besoin des versions et de l'architecture de chaque chose.

Anoratsu, 2020-08-30:
## Current Behavior
 Ma compagne est en télétravail, pour utiliser son PC, elle ouvre une session qui lance une VM. Dans cette VM, elle se connecte au VPN de l'entreprise (pulse secure). Une fois connectée, elle a accès à l'ensembles des logiciels pour travailler.

Depuis le confinement, cela a toujours fonctionné. Depuis la dernière mise à jour du VPS, on rencontre des soucis de connection. Problème de proxy...

## Possible Solution
Cela fonctionnait normalement avant la dernière maj

## Steps to Reproduce the Problem

Shadowsocks désactivé + aucun VPN : fonctionne
Shadowsocks désactivé + Dead simple VPN : fonctionne
Dans les autres situations : cela ne fonctionne pas

## Context (Environment)
J'ai du mal à comprendre pourquoi ça ne fonctionne plus. Conflit de proxy ? bug de maj ?
Est ce que shadowsocks OFF + dead simple ON permet de profiter de l'agrégation ? Pas de manière optimale j'imagine.
Un moyen de laisser tout activer et de contourner le système pour son PC ? Sachant qu'omr bypass n'y fait rien...

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: 	Raspberry Pi 4 Model B Rev 1.1

En espérant ne rien oublier

Ysurac, 2020-08-30:
J'ai supprimé le commentaire en double.

Dans le cas par défaut, shadowsocks on et Glorytun TCP en VPN, le résultat est quoi ? Tout marche sauf le VPN pour le télétravail ?
C'est quoi le VPN utilisé ? Si il s'appuie sur UDP ça peut être un soucis de MTU.

Anoratsu, 2020-08-30:
Oui tout marche sauf le télétravail.
Le VPN utilisé par son entreprise est pulse secure

Ysurac, 2020-08-30:
Visiblement c'est bien un VPN UDP, donc en théorie que shadowsocks soit activé ou pas devrait rien changer, car le traffic UDP passe uniquement par le VPN d'OMR.
Par contre pas possible de régler le MTU sur pulse secure qui semble être calculé selon le MTU de l'interface réseau du client...

Le VPN affiche quelque chose quand il y a un soucis ?

Anoratsu, 2020-08-30:
Quand shadowsocks est activé, le vpn arrive à se connecter mais son environnement de travail (mails outlook, diverses pages internet) ne fonctionnent pas, cela met une erreur proxy.
Si je désactive shadowsocks, plus de problème

Ysurac, 2020-08-30:
Et donc même soucis en bypassant l'IP/domaine d'arriver du VPN ?
L'IPv6 est activé ?

Anoratsu, 2020-08-30:
En mettant l'adresse du proxy dans omr bypass ça ne fonctionne pas.
L'ipv6 n'est pas activée (vps chez milkywan)

Ysurac, 2020-08-30:
Ca devrait. L'adresse est une IP ou un domaine ? Si c'est un domaine il faut être certain que le DNS sur le poste client soit bien l'IP d'OpenMPTCProuter et peut être relancer le poste client, il peut avoir un cache DNS et pas faire de nouvelle requête.

Visiblement Pulse Secure a 2 modes de fonctionnement : 
* ESP qui utilise UDP (et le mode SSL en même temps ?)
* SSL qui utilise TCP sur le port 443

Donc effectivement il doit bien utilisé shadowsocks.
Possible dans Système->OpenMPTCProuter onglet paramètres avancés de "Désactiver TCP Fast Open".

Sinon il faudrait voir dans Etat->journal système si il y a quelque chose lié à "ss-redir".

Anoratsu, 2020-08-30:
Ne pouvant tester que pendant ses horaires de travail, je fais la manip "tcp fast open" et je te tiens au courant ici, merci.

Anoratsu, 2020-08-31:
Bonjour,
Le proxy est un domaine. TCP Fast open ne change rien, quand shadow est activé ça ne fonctionne plus
Pour le DNS, la VM n'a pas l'ip d'OpenMPTCProuter par contre le PC possède plusieurs dns dont OpenMPTCProuter . Le soucis viendrait de là j'imagine ? PC d'entreprise bloqué de partout je sais pas si je vais réussir à modifier ça...
Concernant ss-redir : 
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8550]: enable TCP no-delay
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8550]: resolving hostname to IPv6 address first
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8550]: initializing ciphers... none
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8550]: listening at 0.0.0.0:1100
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8550]: tcp port reuse enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8550]: UDP relay enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8550]: udp port reuse enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8550]: running from root user
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8551]: enable TCP no-delay
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8551]: initializing ciphers... none
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8551]: listening at 0.0.0.0:1100
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8551]: tcp port reuse enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8551]: UDP relay enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8551]: udp port reuse enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8551]: running from root user
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-local[8548]: enable TCP no-delay
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-local[8548]: initializing ciphers... none
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-local[8548]: listening at 127.0.0.1:1111
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-local[8548]: tcp port reuse enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-local[8548]: udprelay enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-local[8548]: udp port reuse enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-local[8548]: running from root user
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8549]: enable TCP no-delay
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8549]: resolving hostname to IPv6 address first
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8549]: initializing ciphers... none
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8552]: enable TCP no-delay
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8552]: initializing ciphers... none
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8552]: listening at 0.0.0.0:1100
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8549]: listening at 0.0.0.0:1100
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8552]: tcp port reuse enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8552]: UDP relay enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8549]: tcp port reuse enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8549]: UDP relay enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8552]: udp port reuse enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8552]: running from root user
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8549]: udp port reuse enabled
Mon Aug 31 15:29:41 2020 daemon.info /usr/bin/ss-redir[8549]: running from root user

Ysurac, 2020-08-31:
Oui pour que le bypass sur un domaine fonctionne il faut effectivement que l'IP d'OpenMPTCProuter soit la seule IP dispo.
Mais rien empêche de mettre l'ip correspondant au domaine, avec un peu de chance il n'y a qu'une seule IP associée, et un ```pin <ledomaine>``` donnera l'IP associée à ajouter dans le bypass.
Je vois que le chiffrement a été désactivé, normalement ça ne pose pas de soucis mais possible de tester avec.

Anoratsu, 2020-08-31:
Ok je vois, je ne connais pas cette commande "pin" je l'utilise où ?
Je vais activer le chiffrement en utilisant chacha20

Ysurac, 2020-08-31:
ping (le g est resté bloqué dans le clavier) depuis n'importe où tu peux le faire. C'est juste pour avoir l'IP.

Anoratsu, 2020-09-01:
Bonjour,
Une seule IP pour le proxy donc parfait. Un soucis pour une de ses pages web, pareil une seule IP.
Actuellement tout fonctionne normalement avec shadowsocks activé, chiffrement chacha20, Glorytun TCP en vpn.
2 IP bypass (une sur l'adsl, une sur master). TCP Fast Open désactivé.

Merci pour l'aide car je n'aurais vraiment pas réussi

closed 2020-09-02 by Anoratsu