Ysurac/openmptcprouter #1191: Wizard reverts to Static IP for interfaces configured as Other

cr3ative, 2020-08-30:
## Expected Behavior

When using the wizard to set up two WAN interfaces, you can select "Other" to configure them in OpenWRT further - for example, PPPoE connections. I would expect the wizard to remember that "Other" has been selected.

## Current Behavior

At the moment, after applying the configuration, the wizard screen reverts to "Static IP". If you were to reapply the settings, they would be incorrect.

## Possible Solution

Remember "other" state when wizard applies settings.

## Steps to Reproduce the Problem

  1. Set up two PPPoE WAN connections (you will need at least one USB Ethernet device to separate PADI/PADO packets physically)
  2. Run the setup wizard, using "Other" and configuring PPPoE in OpenWRT's Interfaces screen
  3. The connections will work! The setup is successful.
  4. Reload the wizard page. You will see that rather than "Other", the wizard has dropped the two interfaces back to unconfigured "Static IP". This is a problem, as if you were to press Save, it would overwrite the working connection information which we previously set up in OpenWRT's interfaces screen.

## Specifications

  - OpenMPTCProuter version:  v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version: Debian 10 Buster, 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform:  RPI4B using image from site

Ysurac, 2020-08-30:
I'm not able to reproduce the issue.
When in Network->Interface, I select something else then Static or DHCP, I can see in wizard that Other is selected.

cr3ative, 2020-08-30:
Strange! Maybe my configuration got confused. In the end, I reflashed my SD card to stock and set it up with PPPoE handled elsewhere, so I don't see this problem now. Thank you for looking, anyway!

closed 2020-08-30 by cr3ative