Ysurac/openmptcprouter #1192: Can't access twitter.com

xillibit, 2020-08-30:
## Expected Behavior
twitter.com should be open

## Current Behavior
In firefox i try to open twitter.com and then fails :

> Le délai d’attente est dépassé
> Le serveur à l’adresse twitter.com met trop de temps à répondre.

## Possible Solution

## Steps to Reproduce the Problem

## Context (Environment)
OMR is running in a VM on proxmox 6.2-11
The VPS is on scaleway

[image]

In "Journal système" i have many lines like that : 

> Sun Aug 30 13:25:44 2020 daemon.err odhcpd[4760]: Failed to send to ff02::1%LAN2@eth1 (Permission denied)
Sun Aug 30 13:26:00 2020 daemon.warn odhcpd[4760]: A default route is present but there is no public prefix on LAN2 thus we don't announce a default route!
Sun Aug 30 13:26:00 2020 daemon.err odhcpd[4760]: Failed to send to ff02::1%LAN2@eth1 (Permission denied)
Sun Aug 30 13:26:16 2020 daemon.warn odhcpd[4760]: A default route is present but there is no public prefix on LAN2 thus we don't announce a default route!
Sun Aug 30 13:26:16 2020 daemon.err odhcpd[4760]: Failed to send to ff02::1%LAN2@eth1 (Permission denied)
Sun Aug 30 13:26:32 2020 daemon.warn odhcpd[4760]: A default route is present but there is no public prefix on LAN2 thus we don't announce a default route!
Sun Aug 30 13:26:32 2020 daemon.err odhcpd[4760]: Failed to send to ff02::1%LAN2@eth1 (Permission denied)
Sun Aug 30 13:26:48 2020 daemon.warn odhcpd[4760]: A default route is present but there is no public prefix on LAN2 thus we don't announce a default route!
Sun Aug 30 13:26:48 2020 daemon.err odhcpd[4760]: Failed to send to ff02::1%LAN2@eth1 (Permission denied)
Sun Aug 30 13:27:04 2020 daemon.warn odhcpd[4760]: A default route is present but there is no public prefix on LAN2 thus we don't announce a default route!
Sun Aug 30 13:27:04 2020 daemon.err odhcpd[4760]: Failed to send to ff02::1%LAN2@eth1 (Permission denied)
Sun Aug 30 13:27:20 2020 daemon.warn odhcpd[4760]: A default route is present but there is no public prefix on LAN2 thus we don't announce a default route!
Sun Aug 30 13:27:20 2020 daemon.err odhcpd[4760]: Failed to send to ff02::1%LAN2@eth1 (Permission denied)

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: 0.1017 5.4.52-mptcp
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-08-30:
If you have IPv6 enabled, check from the router and the VPS, the result of ```curl -6 -L https://www.twitter.com``` and in all case, check on router and VPS ```curl -4 -L https://www.twitter.com```.

For the LAN2 error, it's because you don't check "Always announce default router".

kevin39, 2020-08-30:
About Twitter, There is currently a major outage with Level3. Scaleway is impacted as many isp. Twitter is unavailable from Scaleway.

Kevin LABECOT

xillibit, 2020-08-30:
> For the LAN2 error, it's because you don't check "Always announce default router".

It's already checked on LAN2 : 

[image]

> About Twitter, There is currently a major outage with Level3. Scaleway is impacted as many isp. Twitter is unavailable from Scaleway.

Thanks for the info, i just tried I'am able now to access twitter.com

Ysurac, 2020-08-30:
What is IPv6 of LAN2,  I see "ff02::1" and this is a multicast IPv6 address ?

xillibit, 2020-08-30:
I had set that but i was not sure because i don't know to much IPv6 : 

[image]

Ysurac, 2020-08-30:
This can work.
Do you have any router/firewall/... on LAN2 ?
Check also firewall rules of LAN2, possible that IPv6 RA are blocked.

xillibit, 2020-08-30:
On LAN2 i have my computer

The firewall rules are the following : 

[image]

Adorfer, 2020-08-31:
> Le délai d’attente est dépassé
> Le serveur à l’adresse twitter.com met trop de temps à répondre.

please provide a traceroute or (better) mtr (-4) for twitter.com. 

```
opkg update; opkg install mtr
mtr -4 www.twitter.com
```

xillibit, 2020-08-31:
There was a datacenter failure in France so it has blocked a part of internet yesterday : https://blog.cloudflare.com/analysis-of-todays-centurylink-level-3-outage/

Now i still have issues with IPv6

Adorfer, 2020-08-31:
> Now i still have issues with IPv6

twitter.com has afaik no IPv6.

xillibit, 2020-08-31:
I mean issue with IPv6 globally not only with twitter.com but i have already this other issue open : https://github.com/Ysurac/openmptcprouter/issues/1147 so i close this one

closed 2020-08-31 by xillibit