Ysurac/openmptcprouter #1193: wildcard domain do not pass through OMR-Bypass

hambe76, 2020-08-30:
## Expected Behavior
 

subdomains from a recorded domain pass through OMR-Bypass

## Current Behavior

ASN and Domain not working

[image]

## Possible Solution

Adding possibility to use * as wildcard before domain name or use regex (ex: ^([A-Za-z0-9.-]*\.)?example\.com\.?/ or ^([A-Za-z0-9.-]*\.)?ne?t?fli?x(img|ext|video)?\.(com|net)/ )

Or explain me how to have it working 😄 

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:  openmptcprouter v0.55.3 r0+14198-da98603597
  - OpenMPTCProuter VPS version:  OpenMPTCProuter VPS 0.1017
  - OpenMPTCProuter platform:  RPI4

Ysurac, 2020-08-31:
It's working but OpenMPTCProuter IP must be used as DNS on the clients.

hambe76, 2020-08-31:
OK used router's IP instead of unbound server in pi-hole

closed 2020-08-31 by hambe76