Ysurac/openmptcprouter #1195: Can't contact server Admin script (no result - no api script answer)

domo5177, 2020-09-01:
- OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform: 

VPS chez onenetsolutions debian10 from scratch
OpenMPTCProuter from scratch sur rpi4B

> Tue Sep  1 10:35:23 2020 daemon.info omr-tracker-ss: Shadowsocks is up (can contact via http 1.1.1.1)
> Tue Sep  1 10:35:25 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, server API answer)
> Tue Sep  1 10:35:25 2020 user.notice post-tracking-post-tracking: Set ip route replace default scope global  nexthop via 192.168.10.1 dev wan1 weight 2 nexthop via 192.168.11.1 dev wan2 weight 1
> Tue Sep  1 10:35:46 2020 user.notice OMR-VPS: Can't get token, try later
> Tue Sep  1 10:35:46 2020 user.notice OMR-VPS: Can't get token, try later
> Tue Sep  1 10:35:52 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Tue Sep  1 10:35:59 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Tue Sep  1 10:36:05 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)
> Tue Sep  1 10:36:12 2020 user.notice OMR-VPS: Can't get token, try later (can ping server, no server API answer)

Ysurac, 2020-09-01:
Le serveur a bien été redémarré ?
La clef et l'IP indiquées dans le wizard sont bon ?
Il est toujours possible de revalider le wizard.

domo5177, 2020-09-01:
Serveur redémarré. les clés sont OK. A part cette erreur tous semble fonctionner.

Ysurac, 2020-09-01:
Sur le VPS, dans ```/var/log/daemon.log``` si l'API a planté il devrait y avoir l'info.

domo5177, 2020-09-01:
Oui j'ai ça dans daemon.log
Sep  1 08:54:39 cloud-7373750 sh[422]: PRC: Lease failed to satisfy.
Sep  1 08:54:39 cloud-7373750 dhclient[530]: XMT: Solicit on eth0, interval 128080ms.
Sep  1 08:54:39 cloud-7373750 dhclient[530]: RCV: Advertise message on eth0 from fe80::250:56ff:feb6:f99d.
Sep  1 08:54:48 cloud-7373750 systemd[1]: omr-admin.service: State 'stop-sigterm' timed out. Killing.
Sep  1 08:54:48 cloud-7373750 systemd[1]: omr-admin.service: Killing process 4364 (python3) with signal SIGKILL.
Sep  1 08:54:48 cloud-7373750 systemd[1]: omr-admin.service: Killing process 4538 (sh) with signal SIGKILL.
Sep  1 08:54:48 cloud-7373750 systemd[1]: omr-admin.service: Killing process 4539 (wget) with signal SIGKILL.
Sep  1 08:54:48 cloud-7373750 systemd[1]: omr-admin.service: Main process exited, code=killed, status=9/KILL
Sep  1 08:54:48 cloud-7373750 systemd[1]: omr-admin.service: Failed with result 'timeout'.
Sep  1 08:54:48 cloud-7373750 systemd[1]: Stopped OMR-Admin.
Sep  1 08:54:48 cloud-7373750 systemd[1]: Started OMR-Admin.
Sep  1 08:55:01 cloud-7373750 systemd[1]: Stopping OMR-Admin...
Sep  1 08:56:31 cloud-7373750 systemd[1]: omr-admin.service: State 'stop-sigterm' timed out. Killing.
Sep  1 08:56:31 cloud-7373750 systemd[1]: omr-admin.service: Killing process 4540 (python3) with signal SIGKILL.
Sep  1 08:56:31 cloud-7373750 systemd[1]: omr-admin.service: Killing process 4714 (sh) with signal SIGKILL.
Sep  1 08:56:31 cloud-7373750 systemd[1]: omr-admin.service: Killing process 4715 (wget) with signal SIGKILL.
Sep  1 08:56:31 cloud-7373750 systemd[1]: omr-admin.service: Main process exited, code=killed, status=9/KILL
Sep  1 08:56:31 cloud-7373750 systemd[1]: omr-admin.service: Failed with result 'timeout'.
Sep  1 08:56:31 cloud-7373750 systemd[1]: Stopped OMR-Admin.
Sep  1 08:56:31 cloud-7373750 systemd[1]: Started OMR-Admin.
Sep  1 08:56:44 cloud-7373750 systemd[1]: Stopping OMR-Admin...
Sep  1 08:56:47 cloud-7373750 sh[422]: XMT: Forming Solicit, 655350 ms elapsed.
Sep  1 08:56:47 cloud-7373750 sh[422]: XMT:  X-- IA_NA 56:00:f0:e9
Sep  1 08:56:47 cloud-7373750 sh[422]: XMT:  | X-- Request renew in  +3600
Sep  1 08:56:47 cloud-7373750 sh[422]: XMT:  | X-- Request rebind in +5400
Sep  1 08:56:47 cloud-7373750 sh[422]: XMT: Solicit on eth0, interval 113450ms.
Sep  1 08:56:47 cloud-7373750 dhclient[530]: XMT: Solicit on eth0, interval 113450ms.
Sep  1 08:56:47 cloud-7373750 dhclient[530]: RCV: Advertise message on eth0 from fe80::250:56ff:feb6:f99d.
Sep  1 08:56:47 cloud-7373750 sh[422]: RCV: Advertise message on eth0 from fe80::250:56ff:feb6:f99d.
Sep  1 08:56:47 cloud-7373750 sh[422]: RCV:  X-- IA_NA 56:00:f0:e9
Sep  1 08:56:47 cloud-7373750 sh[422]: RCV:  | X-- starts 1598950607
Sep  1 08:56:47 cloud-7373750 sh[422]: RCV:  | X-- t1 - renew  +0
Sep  1 08:56:47 cloud-7373750 sh[422]: RCV:  | X-- t2 - rebind +0
Sep  1 08:56:47 cloud-7373750 sh[422]: RCV:  | X-- [Options]
Sep  1 08:56:47 cloud-7373750 sh[422]: RCV:  | !-- Status code of no addrs, IA_NA discarded.
Sep  1 08:56:47 cloud-7373750 sh[422]: RCV:  X-- Server ID: 00:01:00:01:24:a8:96:7c:00:50:56:b6:f9:9d
Sep  1 08:56:47 cloud-7373750 sh[422]: PRC: Lease failed to satisfy.

domo5177, 2020-09-02:
Ok après désactivation de l'IPV6 sur le VPS.

closed 2020-09-02 by domo5177