Ysurac/openmptcprouter #1196: Gateway down error

Corrado56, 2020-09-01:
## Expected Behavior
Gateway Up

## Current Behavior
Look at this situation.
[image]

I am trying to aggregate three LTE connection from different providers.
the modem type of connection "Ogilink_voda" and "Iliad" are the same (Huawei B535) with identical configuration except for the IP range
For some reason the connection "Iliad" is "Gateway down".
If I connect directly to the modem with WI-FI connection, I am connected to internet and navigate.
I have tried to reboot the Huawei Modem, the openMPTCP router, the VPS, but nothing happens
What are the possible causes for this error ? How could I indagate?

I am actually aggregating only two connections

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: 0.1017
  - OpenMPTCProuter platform: Linksys 3200

Ysurac, 2020-09-01:
The gateway down is only a warning? Some ISP/dongle doesn't answer to ping on gateway so it's not a problem.
But in your case if it's the same modem as Ogilink then you should get same info with operator, lte power,...
How do you check that only two connections are aggregated ?

Corrado56, 2020-09-01:
Speedtest using MPTCP router
[image]

Connecting the modem via Wi Fi
[image]

The iliad SIM is working, but I have no traffic on Iliad interface on MPTCP Router

Ysurac, 2020-09-01:
Can you try, for about 2 minutes, to test via SSH on the router ```omr-test-speed``` and ```omr-test-speed wan1``` where wan1 is a real interface (visible via ```ip link```) on each wan interface ?

Corrado56, 2020-09-01:
[image]

[image]

Ysurac, 2020-09-01:
What is the interface corresponding to lan4 ? Iliad ?

Corrado56, 2020-09-01:
Hi Yannik,
FOUND !
I try to create a new interface WAN3 connected to phisical port LAN3
And this is the result:
[image]
What ? the same MAC on 2 interfaces ?
And the mac is equal to:
[image]
So the culprit is: **THE WIZARD**
[image]
When You create a new interface of type Static IP (other cases not tested), the wizard by default suggest the MAC of the Physical Port WAN (instead of a casual one), and If You don't change the suggestion You have a not working interface. 
After changing the MAC on Iliad interface:
[image]

I like happy ends

closed 2020-09-04 by Corrado56