Ysurac/openmptcprouter #1197: WRT3200ACM instable Wifi

cl3f3, 2020-09-02:
## Expected Behavior
Stable Wifi, LED indication working

## Current Behavior
WiFi whatever is the way you configure it (AC, BGN or both) crashes once a day (SSID disappear, or the connection become incapable to serve anything) and if you try to configure LED activity (wlan LES) Wifi doesn't work anymore (no SSID, no connection).

## Possible Solution
No idea

## Steps to Reproduce the Problem

  1. Configure wifi netxork with WPA or WPA2
  2. wait
  3. crash
or
1. Configure Wifi
2. Configure WLAN LED activity
3. Crash

## Context (Environment)
1 ADSL gateway on lan 1
2 4G gateway on lan 2 et 3

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter platform:

Ysurac, 2020-09-02:
Please complete the template with versions.

Check in State->System log and kernel log what you have when there is the crash.

cl3f3, 2020-09-03:
syslog : 

`Thu Sep  3 17:12:56 2020 daemon.info dnsmasq-dhcp[15512]: DHCPREQUEST(br-lan) 192.168.100.162 a4:4e:31:43:e8:3c
Thu Sep  3 17:12:56 2020 daemon.info dnsmasq-dhcp[15512]: DHCPACK(br-lan) 192.168.100.162 a4:4e:31:43:e8:3c altertube-ThinkPad-X230
Thu Sep  3 17:13:04 2020 daemon.info dnsmasq-dhcp[15512]: DHCPREQUEST(br-lan) 192.168.100.162 a4:4e:31:43:e8:3c
Thu Sep  3 17:13:04 2020 daemon.info dnsmasq-dhcp[15512]: DHCPACK(br-lan) 192.168.100.162 a4:4e:31:43:e8:3c altertube-ThinkPad-X230
Thu Sep  3 17:13:17 2020 daemon.info dnsmasq-dhcp[15512]: DHCPREQUEST(br-lan) 192.168.100.162 a4:4e:31:43:e8:3c
Thu Sep  3 17:13:17 2020 daemon.info dnsmasq-dhcp[15512]: DHCPACK(br-lan) 192.168.100.162 a4:4e:31:43:e8:3c altertube-ThinkPad-X230
Thu Sep  3 17:13:28 2020 daemon.info dnsmasq-dhcp[15512]: DHCPREQUEST(br-lan) 192.168.100.162 a4:4e:31:43:e8:3c
Thu Sep  3 17:13:28 2020 daemon.info dnsmasq-dhcp[15512]: DHCPACK(br-lan) 192.168.100.162 a4:4e:31:43:e8:3c altertube-ThinkPad-X230
Thu Sep  3 17:13:54 2020 daemon.err /usr/bin/ss-redir[14128]: server recv: Operation timed out
Thu Sep  3 17:14:00 2020 daemon.info hostapd: wlan1: STA 28:02:d8:be:4a:94 IEEE 802.11: associated (aid 4)
Thu Sep  3 17:14:00 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 28:02:d8:be:4a:94
Thu Sep  3 17:14:00 2020 daemon.info hostapd: wlan1: STA 28:02:d8:be:4a:94 WPA: pairwise key handshake completed (WPA)
Thu Sep  3 17:14:00 2020 daemon.info hostapd: wlan1: STA 28:02:d8:be:4a:94 WPA: group key handshake completed (WPA)
Thu Sep  3 17:14:00 2020 daemon.info dnsmasq-dhcp[15512]: DHCPDISCOVER(br-lan) 28:02:d8:be:4a:94
Thu Sep  3 17:14:00 2020 daemon.info dnsmasq-dhcp[15512]: DHCPOFFER(br-lan) 192.168.100.148 28:02:d8:be:4a:94
Thu Sep  3 17:14:00 2020 daemon.info dnsmasq-dhcp[15512]: DHCPREQUEST(br-lan) 192.168.100.148 28:02:d8:be:4a:94
Thu Sep  3 17:14:00 2020 daemon.info dnsmasq-dhcp[15512]: DHCPACK(br-lan) 192.168.100.148 28:02:d8:be:4a:94 Galaxy-Tab-A-80-2019
Thu Sep  3 17:14:01 2020 daemon.info hostapd: wlan1: STA 28:02:d8:be:4a:94 IEEE 802.11: authenticated
Thu Sep  3 17:14:52 2020 daemon.err /usr/bin/ss-redir[14127]: server recv: Operation timed out
Thu Sep  3 17:18:05 2020 daemon.err /usr/bin/ss-redir[14127]: server recv: Operation timed out
Thu Sep  3 17:18:36 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED a4:4e:31:43:e8:3c
Thu Sep  3 17:18:36 2020 daemon.info hostapd: wlan1: STA a4:4e:31:43:e8:3c IEEE 802.11: disassociated due to inactivity
Thu Sep  3 17:18:36 2020 kern.debug kernel: [111888.819477] ieee80211 phy1: staid 1 deleted
Thu Sep  3 17:18:36 2020 daemon.info hostapd: wlan1: STA a4:4e:31:43:e8:3c IEEE 802.11: associated (aid 2)
Thu Sep  3 17:18:37 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED a4:4e:31:43:e8:3c
Thu Sep  3 17:18:37 2020 daemon.info hostapd: wlan1: STA a4:4e:31:43:e8:3c WPA: pairwise key handshake completed (WPA)
Thu Sep  3 17:18:37 2020 daemon.info hostapd: wlan1: STA a4:4e:31:43:e8:3c WPA: group key handshake completed (WPA)
Thu Sep  3 17:18:40 2020 daemon.info hostapd: wlan1: STA a4:4e:31:43:e8:3c IEEE 802.11: authenticated
Thu Sep  3 17:19:25 2020 daemon.notice hostapd: wlan1: AP-STA-DISCONNECTED 28:02:d8:be:4a:94
Thu Sep  3 17:19:25 2020 daemon.info hostapd: wlan1: STA 28:02:d8:be:4a:94 IEEE 802.11: disassociated due to inactivity
Thu Sep  3 17:19:26 2020 daemon.info hostapd: wlan1: STA 28:02:d8:be:4a:94 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Thu Sep  3 17:19:26 2020 kern.debug kernel: [111938.488733] ieee80211 phy1: staid 4 deleted
Thu Sep  3 17:20:23 2020 daemon.err /usr/bin/ss-redir[14127]: server recv: Operation timed out
Thu Sep  3 17:21:18 2020 daemon.err /usr/bin/ss-redir[14127]: server recv: Operation timed out
Thu Sep  3 17:21:51 2020 daemon.err /usr/bin/ss-redir[14127]: server recv: Operation timed out
Thu Sep  3 17:22:16 2020 daemon.err /usr/bin/ss-redir[14127]: server recv: Operation timed out
Thu Sep  3 17:22:22 2020 daemon.err /usr/bin/ss-redir[14128]: server recv: Operation timed out
Thu Sep  3 17:23:01 2020 daemon.info hostapd: wlan1: STA 28:02:d8:be:4a:94 IEEE 802.11: associated (aid 4)
Thu Sep  3 17:23:01 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 28:02:d8:be:4a:94
Thu Sep  3 17:23:01 2020 daemon.info hostapd: wlan1: STA 28:02:d8:be:4a:94 WPA: pairwise key handshake completed (WPA)
Thu Sep  3 17:23:01 2020 daemon.info hostapd: wlan1: STA 28:02:d8:be:4a:94 WPA: group key handshake completed (WPA)
Thu Sep  3 17:23:01 2020 daemon.info dnsmasq-dhcp[15512]: DHCPDISCOVER(br-lan) 28:02:d8:be:4a:94
Thu Sep  3 17:23:01 2020 daemon.info dnsmasq-dhcp[15512]: DHCPOFFER(br-lan) 192.168.100.148 28:02:d8:be:4a:94
Thu Sep  3 17:23:01 2020 daemon.info dnsmasq-dhcp[15512]: DHCPREQUEST(br-lan) 192.168.100.148 28:02:d8:be:4a:94
Thu Sep  3 17:23:01 2020 daemon.info dnsmasq-dhcp[15512]: DHCPACK(br-lan) 192.168.100.148 28:02:d8:be:4a:94 Galaxy-Tab-A-80-2019
Thu Sep  3 17:23:02 2020 daemon.info hostapd: wlan1: STA 28:02:d8:be:4a:94 IEEE 802.11: authenticated
Thu Sep  3 17:24:32 2020 daemon.err /usr/bin/ss-redir[14127]: server recv: Operation timed out
Thu Sep  3 17:25:18 2020 daemon.err /usr/bin/ss-redir[14127]: server recv: Operation timed out
Thu Sep  3 17:25:20 2020 daemon.err /usr/bin/ss-redir[14127]: server recv: Operation timed out
Thu Sep  3 17:27:24 2020 daemon.err /usr/bin/ss-redir[14127]: server recv: Operation timed out
Thu Sep  3 17:27:34 2020 daemon.err /usr/bin/ss-redir[14127]: server recv: Operation timed out
[... cut ...]
Thu Sep  3 21:22:40 2020 daemon.info dnsmasq-dhcp[15512]: DHCPREQUEST(br-lan) 192.168.100.221 74:38:b7:94:a9:dd
Thu Sep  3 21:22:40 2020 daemon.info dnsmasq-dhcp[15512]: DHCPACK(br-lan) 192.168.100.221 74:38:b7:94:a9:dd
Thu Sep  3 21:22:45 2020 daemon.info hostapd: wlan1: STA 74:38:b7:94:a9:dd IEEE 802.11: authenticated
Thu Sep  3 21:23:16 2020 daemon.err uhttpd[4114]: luci: accepted login on / for root from 192.168.100.134
Thu Sep  3 21:23:23 2020 daemon.err /usr/bin/ss-redir[14128]: server recv: Operation timed out`

cl3f3, 2020-09-03:
Kernel log : 

> [    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.58 (circleci@default-b75ebe4d-e054-48db-8d48-d7f15d0c9456) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r0+14198-da98603597)) #0 SMP Tue Aug 18 16:14:54 2020
[    0.000000] CPU: ARMv7 Processor [414fc091] revision 1 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Linksys WRT3200ACM
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 131072
[    0.000000]   Normal zone: 1152 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 131072 pages, LIFO batch:31
[    0.000000] percpu: Embedded 11 pages/cpu s15500 r8192 d21364 u45056
[    0.000000] pcpu-alloc: s15500 r8192 d21364 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129920
[    0.000000] Kernel command line: console=ttyS0,115200 root=/dev/mtdblock6 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,128K(u_env),256K(s_env),256K@8064K(devinfo),1920K@8320K(sysdiag),80m@10m(kernel),74m@16m(rootfs),80m@90m(alt_kernel),74m@96m(alt_rootfs),160m@10m(ubifs),-@170m(syscfg)
[    0.000000] Bootloader command line (ignored): console=ttyS0,115200 root=/dev/mtdblock6 ro rootdelay=1 rootfstype=jffs2 earlyprintk mtdparts=armada-nand:2048K(uboot)ro,128K(u_env),256K(s_env),256K@8064K(devinfo),1920K@8320K(sysdiag),80m@10m(kernel),74m@16m(rootfs),80m@90m(alt_kernel),74m@96m(alt_rootfs),160m@10m(ubifs),-@170m(syscfg)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 508380K/524288K available (7168K kernel code, 231K rwdata, 1696K rodata, 1024K init, 238K bss, 15908K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 D prefetch enabled, offset 1 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 Coherent cache controller enabled, 16 ways, 1024 kB
[    0.000000] L2C-310 Coherent: CACHE_ID 0x410054c9, AUX_CTRL 0x56070001
[    0.000000] random: get_random_bytes called from start_kernel+0x3b4/0x564 with crng_init=0
[    0.000004] sched_clock: 64 bits at 933MHz, resolution 1ns, wraps every 4398046511103ns
[    0.000013] clocksource: arm_global_timer: mask: 0xffffffffffffffff max_cycles: 0x1ae5b571769, max_idle_ns: 881590513431 ns
[    0.000024] Switching to timer-based delay loop, resolution 1ns
[    0.000129] Ignoring duplicate/late registration of read_current_timer delay
[    0.000134] clocksource: armada_370_xp_clocksource: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 76450417870 ns
[    0.000241] Calibrating delay loop (skipped), value calculated using timer frequency.. 1866.00 BogoMIPS (lpj=9330000)
[    0.000246] pid_max: default: 32768 minimum: 301
[    0.000306] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000311] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.000598] CPU: Testing write buffer coherency: ok
[    0.000608] CPU0: Spectre v2: using BPIALL workaround
[    0.000701] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.000850] Setting up static identity map for 0x100000 - 0x100060
[    0.000911] mvebu-soc-id: MVEBU SoC ID=0x6820, Rev=0x4
[    0.000973] mvebu-pmsu: Initializing Power Management Service Unit
[    0.001017] rcu: Hierarchical SRCU implementation.
[    0.001107] smp: Bringing up secondary CPUs ...
[    0.001220] Booting CPU 1
[    0.001357] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.001360] CPU1: Spectre v2: using BPIALL workaround
[    0.001406] smp: Brought up 1 node, 2 CPUs
[    0.001409] SMP: Total of 2 processors activated (3732.00 BogoMIPS).
[    0.001412] CPU: All CPU(s) started in SVC mode.
[    0.001578] devtmpfs: initialized
[    0.002914] VFP support v0.3: implementor 41 architecture 3 part 30 variant 9 rev 4
[    0.002967] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.002974] futex hash table entries: 512 (order: 3, 32768 bytes, linear)
[    0.003039] pinctrl core: initialized pinctrl subsystem
[    0.003125] reg-dummy reg-dummy: no of_node; not parsing pinctrl DT
[    0.003507] NET: Registered protocol family 16
[    0.003929] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.004398] cpuidle: using governor ladder
[    0.004535] mvebu-pmsu: CPU hotplug support is currently broken on Armada 38x: disabling
[    0.004540] mvebu-pmsu: CPU idle is currently broken on Armada 38x: disabling
[    0.011183] cryptd: max_cpu_qlen set to 1000
[... 1229 lines cut ...]
[... cut ...]
[123918.034444] ieee80211 phy1: staid 1 deleted
[124234.272919] ieee80211 phy1: staid 1 deleted
[124541.491697] ieee80211 phy1: staid 1 deleted
[124673.873261] ieee80211 phy1: staid 4 deleted
[124987.765326] ieee80211 phy1: staid 1 deleted
[125310.024271] ieee80211 phy1: staid 1 deleted
[125526.349468] ieee80211 phy1: staid 4 deleted
[125617.241853] ieee80211 phy1: staid 1 deleted
[125921.477055] ieee80211 phy1: staid 1 deleted
[126298.802638] ieee80211 phy1: staid 1 deleted
[126713.563373] ieee80211 phy1: staid 4 deleted

Ysurac, 2020-09-04:
In Network->Wifi, edit and in Interface configuration part, "advanced settings" tab, check "Disable Inactivity Polling"

cl3f3, 2020-09-10:
checking this item did not solve wifi instability.

Ysurac, 2020-09-10:
Did you try to change channel, mode, transmit power,... ?

cl3f3, 2020-10-08:
I gave up using the Wi-Fi part of the router. I deactivated the wifi and I connected two small mesh boxes on it.

closed 2021-01-12 by github-actions[bot]