Ysurac/openmptcprouter #1198: WRT3200ACM WIFI LED

cl3f3, 2020-09-02:
## Expected Behavior
Wifi LED activity

## Current Behavior
If you try to add LED activity for Wifi (WLAN), wifi stop working

## Possible Solution
Disable LED activity

## Steps to Reproduce the Problem

  1. Configure Wifi network (WPA or WPA2)
  2. Configure LED indicator (WLAN LED)
  3. crash

## Context (Environment)

OMR with 3 gatways
ADSL on LAN1
4G Bouygues (routeur) on LAN2
4G Orange (routeur) on LAN 3

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.55.3 r0+14198-da98603597 / LuCI Master git-20.219.57394-35e2735 
  - OpenMPTCProuter VPS version: 0.1017 5.4.52 
  - OpenMPTCProuter platform:  Linksys WRT3200ACM

Ysurac, 2020-09-04:
Check in State->System log and kernel log if there is something related to this issue.

cl3f3, 2020-09-10:
system log : 

Thu Sep 10 08:10:11 2020 kern.err kernel: [69618.718194] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:11 2020 kern.err kernel: [69618.777042] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:11 2020 kern.err kernel: [69618.823750] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:11 2020 kern.err kernel: [69618.873755] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:11 2020 kern.err kernel: [69618.933767] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69618.973784] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.043828] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.083835] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.143861] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.183882] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.243889] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.283907] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.343920] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.383921] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.453942] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.493972] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.563984] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.613998] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.674015] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.724032] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.794057] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.828505] ieee80211 phy1: adapter does not exist
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.833317] ieee80211 phy1: adapter does not exist
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.838145] ieee80211 phy1: adapter does not exist
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.842955] ieee80211 phy1: adapter does not exist
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.844073] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.847772] ieee80211 phy1: adapter does not exist
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.858173] ieee80211 phy1: adapter does not exist
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.914110] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:12 2020 kern.err kernel: [69619.954112] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.024134] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.064175] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.124174] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.174187] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.234204] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.274218] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.284284] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.334238] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.384257] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.454280] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.494293] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.564315] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.604328] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.674352] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.724369] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.794394] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.834406] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.914432] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:13 2020 kern.err kernel: [69620.954469] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:14 2020 kern.err kernel: [69621.014486] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:14 2020 kern.err kernel: [69621.054503] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:14 2020 kern.err kernel: [69621.114530] ieee80211 phy1: buffer is NULL for tx done ring
[... cut ...]
Thu Sep 10 08:10:22 2020 kern.err kernel: [69629.097184] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:22 2020 kern.err kernel: [69629.147202] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:22 2020 kern.err kernel: [69629.207222] ieee80211 phy0: buffer is NULL for tx done ring
Thu Sep 10 08:10:22 2020 kern.err kernel: [69629.257238] ieee80211 phy1: buffer is NULL for tx done ring
Thu Sep 10 08:10:22 2020 kern.err kernel: [69629.317270] ieee80211 phy1: buffer is NULL for tx done ring

cl3f3, 2020-09-10:
kernel log 👍 [69634.348976] ieee80211 phy0: buffer is NULL for tx done ring
[69634.388986] ieee80211 phy1: buffer is NULL for tx done ring
[69634.449020] ieee80211 phy1: buffer is NULL for tx done ring
[69634.489025] ieee80211 phy0: buffer is NULL for tx done ring
[69634.549045] ieee80211 phy0: buffer is NULL for tx done ring
[69634.589053] ieee80211 phy1: buffer is NULL for tx done ring
[69634.634576] ieee80211 phy1: buffer is NULL for tx done ring
[69634.649084] ieee80211 phy1: buffer is NULL for tx done ring
[69634.699080] ieee80211 phy0: buffer is NULL for tx done ring
[69634.759090] ieee80211 phy0: buffer is NULL for tx done ring
[69634.799102] ieee80211 phy1: buffer is NULL for tx done ring
[69634.869133] ieee80211 phy1: buffer is NULL for tx done ring
[69634.909139] ieee80211 phy0: buffer is NULL for tx done ring
[69634.969158] ieee80211 phy0: buffer is NULL for tx done ring
[69635.009174] ieee80211 phy1: buffer is NULL for tx done ring
[69635.069191] ieee80211 phy1: buffer is NULL for tx done ring
[69635.119211] ieee80211 phy0: buffer is NULL for tx done ring
[69635.179229] ieee80211 phy0: buffer is NULL for tx done ring
[69635.219241] ieee80211 phy1: buffer is NULL for tx done ring
[69635.279265] ieee80211 phy1: buffer is NULL for tx done ring
[69635.319281] ieee80211 phy0: buffer is NULL for tx done ring
[69635.389301] ieee80211 phy0: buffer is NULL for tx done ring
[69635.439317] ieee80211 phy1: buffer is NULL for tx done ring
[69635.514487] ieee80211 phy1: buffer is NULL for tx done ring
[69635.559361] ieee80211 phy0: buffer is NULL for tx done ring
[69635.629380] ieee80211 phy0: buffer is NULL for tx done ring
[69635.679412] ieee80211 phy1: buffer is NULL for tx done ring
[69635.749423] ieee80211 phy1: buffer is NULL for tx done ring
[69635.789435] ieee80211 phy0: buffer is NULL for tx done ring
[69635.859461] ieee80211 phy0: buffer is NULL for tx done ring
[69635.899472] ieee80211 phy1: buffer is NULL for tx done ring
[69635.959491] ieee80211 phy1: buffer is NULL for tx done ring
[69635.999506] ieee80211 phy0: buffer is NULL for tx done ring
[69636.069528] ieee80211 phy0: buffer is NULL for tx done ring
[69636.129548] ieee80211 phy1: buffer is NULL for tx done ring
[69636.189569] ieee80211 phy1: buffer is NULL for tx done ring
[69636.239585] ieee80211 phy0: buffer is NULL for tx done ring
[69636.299604] ieee80211 phy0: buffer is NULL for tx done ring
[69636.339619] ieee80211 phy1: buffer is NULL for tx done ring
[69636.345268] ieee80211 phy1: buffer is NULL for tx done ring
[69636.419645] ieee80211 phy1: buffer is NULL for tx done ring
[69636.459671] ieee80211 phy0: buffer is NULL for tx done ring
[69636.529730] ieee80211 phy0: buffer is NULL for tx done ring
[69636.579703] ieee80211 phy1: buffer is NULL for tx done ring
[69636.649724] ieee80211 phy1: buffer is NULL for tx done ring
[69636.699750] ieee80211 phy0: buffer is NULL for tx done ring
[69636.759777] ieee80211 phy0: buffer is NULL for tx done ring
[69636.799800] ieee80211 phy1: buffer is NULL for tx done ring
[69636.859820] ieee80211 phy1: buffer is NULL for tx done ring
[69636.899831] ieee80211 phy0: buffer is NULL for tx done ring
[69636.959845] ieee80211 phy0: buffer is NULL for tx done ring
[69636.999859] ieee80211 phy1: buffer is NULL for tx done ring
[69637.069869] ieee80211 phy1: buffer is NULL for tx done ring
[69637.119883] ieee80211 phy0: buffer is NULL for tx done ring
[69637.179901] ieee80211 phy0: buffer is NULL for tx done ring
[69637.239921] ieee80211 phy1: buffer is NULL for tx done ring
[69637.309945] ieee80211 phy1: buffer is NULL for tx done ring
[69637.359961] ieee80211 phy0: buffer is NULL for tx done ring
[69637.420044] ieee80211 phy0: buffer is NULL for tx done ring
[69637.460014] ieee80211 phy1: buffer is NULL for tx done ring
[69637.540026] ieee80211 phy1: buffer is NULL for tx done ring
[69637.590043] ieee80211 phy0: buffer is NULL for tx done ring
[69637.660064] ieee80211 phy0: buffer is NULL for tx done ring
[69637.715250] ieee80211 phy1: buffer is NULL for tx done ring
[69637.790114] ieee80211 phy0: buffer is NULL for tx done ring
[69637.840125] ieee80211 phy1: buffer is NULL for tx done ring
[69637.910147] ieee80211 phy1: buffer is NULL for tx done ring
[69637.930272] ieee80211 phy1: adapter does not exist
[69637.935084] ieee80211 phy1: adapter does not exist
[69637.939896] ieee80211 phy1: adapter does not exist
[... 967 lines cut ...]
[69687.706915] ieee80211 phy1: buffer is NULL for tx done ring
[69687.746922] ieee80211 phy0: buffer is NULL for tx done ring
[69687.806949] ieee80211 phy0: buffer is NULL for tx done ring
[69687.846967] ieee80211 phy1: buffer is NULL for tx done ring
[69687.916985] ieee80211 phy1: buffer is NULL for tx done ring
[69687.957002] ieee80211 phy0: buffer is NULL for tx done ring
[69688.017010] ieee80211 phy0: buffer is NULL for tx done ring
[69688.057033] ieee80211 phy1: buffer is NULL for tx done ring
[69688.117049] ieee80211 phy1: buffer is NULL for tx done ring
[69688.157071] ieee80211 phy0: buffer is NULL for tx done ring
[69688.217086] ieee80211 phy0: buffer is NULL for tx done ring
[69688.257107] ieee80211 phy1: buffer is NULL for tx done ring
[69688.327125] ieee80211 phy1: buffer is NULL for tx done ring
[69688.377136] ieee80211 phy0: buffer is NULL for tx done ring
[69688.437146] ieee80211 phy0: buffer is NULL for tx done ring
[69688.477160] ieee80211 phy1: buffer is NULL for tx done ring
[69688.547178] ieee80211 phy1: buffer is NULL for tx done ring
[69688.587190] ieee80211 phy0: buffer is NULL for tx done ring
[69688.657211] ieee80211 phy0: buffer is NULL for tx done ring
[69688.697226] ieee80211 phy1: buffer is NULL for tx done ring
[69688.718607] ieee80211 phy1: buffer is NULL for tx done ring
[69688.757246] ieee80211 phy1: buffer is NULL for tx done ring
[69688.807262] ieee80211 phy0: buffer is NULL for tx done ring
[69688.877285] ieee80211 phy0: buffer is NULL for tx done ring
[69688.922470] ieee80211 phy1: buffer is NULL for tx done ring

Ysurac, 2020-09-10:
This seems to be an old OpenWrt issue not yet solved.

closed 2021-01-04 by github-actions[bot]