Ysurac/openmptcprouter #1199: Port Forwarding Issue: data being sent from wan's instead of bonding VPS

BlueWaterCrystal, 2020-09-03:
## Expected Behavior
TCP/UDP data:
sending:  client > VPS > bonding > openmptcprouter > server
receiving: server  > openmptcprouter > bonding > VPS > client

sending mail:
email client > mail server > openmptcprouter > bonding > VPS > another mail server
receiving mail:
another mail server > VPS > bonding > openmptcprouter > mail server > email client

## Current Behavior
TCP/UDP data:
sending:  client > VPS > bonding > openmptcprouter > server
receiving: server  > wan > client

sending mail:
email client > mail server > wan > another mail server (SPAM folder because not VPS public ip)
receiving mail: (works correctly)
another mail server > VPS > bonding > openmptcprouter > mail server > email client

## Possible Solution
when there is only one wan it seems to send email via bonding.
I remember one recent update where it fixed this I think.

## Steps to Reproduce the Problem
  0.  more than one wan interface
  1.  iRedMail
  2.  Port forward:
       Protocol: TCP
       Source Zone: VPN
       External Port: *port number here*
       Destination zone: LAN
       Internal IP address: pointed at mail server
       Internal port: *same as External Port*

## Context (Environment)
when the mail server sends mail to another server it lands up sending through the wan interfaces instead of the bonding.
I have blocked all traffic to each interface on the mail server, so its not possible for the server to communicate with any wan.

this was confirmed looking at gmail data and failing SPF because of different IP but when 1 out of 3 wan's enabled it sends through bonding and doesn't fail SPF being the VPS address.
I have also tried NAT Rules with SNAT but hasn't done anything.

any other traffic port forwarded (ECHO TEST):
when a client sending data to the VPS Public IP it gets forwarded to the right destination but the destination reply's using a wan instead of bonding. 

I remember a recent update that fixed this issue I think but it seems to be back I guess?

tldr: incoming correct,   outcoming not correct sending from wan's instead of bonding VPS 

## Specifications

  - OpenMPTCProuter version: 0.55.3
  - OpenMPTCProuter VPS version: latest
  - OpenMPTCProuter platform: x86_64

Ysurac, 2020-09-03:
All is green on status page ? When you try a ```curl ifconfig.co``` from the mail server you get VPS IP ?

BlueWaterCrystal, 2020-09-03:
Hey

I did the [curl ifconfig.co](url) on the mail server and it was the VPS IP

and everything seems to be green
[image]

Ysurac, 2020-09-03:
What is the result, on the VPS of a ```tcpdump -p -vvv port 25``` when you send a mail (```apt-get install tcpdump``` if not already installed on the VPS) ?

BlueWaterCrystal, 2020-09-04:
When 3 wans were connected: 
when sending email to gmail nothing appeared 
when sending gmail to email packets appeared

when 1 wan were connected:
appeared on both and email went through

I also did a test sending email on a different network with 1 wan and 3 wans
and it had the same effect

It seems to be issue between mail server and openmptcprouter

I think I'm going to try roll back the OS and see if it will work again

BlueWaterCrystal, 2020-09-04:
update:
I installed v0.55.2 r0+14198-da98603597 and port forwarding now works properly for mail server. 

however,
I did notice this when port forwarding web sockets and trying to connect to it via VPS public IP
[image]

BlueWaterCrystal, 2020-09-13:
Update:
If you upgrade x86_64 image it seems to break the port forwarding and settings.

closed 2020-09-13 by BlueWaterCrystal

Adorfer, 2020-09-14:
> If you upgrade x86_64 image it seems to break the port forwarding and settings.

So then: why close it? 

this sounds like a bug and perhaps just editing the title towards "upgrading from 55.2 to 55.3 breaks port forwarding" would be useful.